coolsnowwolf/lede #11032: 多次尝试编译都出现同样错误

justformg, 2023-03-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：多次编译出现同样错误

### (2) 路由器型号和固件版本

A：XY-C5

### (3) 详细日志

A：
make[3]: *** [Makefile:67: /home/chan/lede/build_dir/target-mipsel_24kc_musl/frp-0.47.0/.built] Error 1
make[3]: Leaving directory '/home/chan/lede/feeds/packages/net/frp'
time: package/feeds/packages/frp/compile#0.20#2.76#257.99
    ERROR: package/feeds/packages/frp failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/frp/compile] Error 1
make[2]: Leaving directory '/home/chan/lede'
make[1]: *** [package/Makefile:110: /home/chan/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/chan/lede'
make: *** [/home/chan/lede/include/toplevel.mk:231: world] Error 2

bzlzm, 2023-03-26:
我从昨天到今天一直也是这个231 错误，云编译基本在快到2小时的时候出错，我恢复了以前成功的配置文件得以编译成功，应该是软件冲突造成的吧。我的是新添加了阿里云webdav和alist造成的吧

cheneylaw, 2023-03-27:
看样子可能是frp造成的，去掉这个再试试看呢

justformg, 2023-03-28:
估计就是插件冲突造成的

closed 2023-03-28 by justformg