coolsnowwolf/lede #11037: 新版本ssrp 使用xraycore 1.8.0 路由器本地代理正常 客户端无法代理  user.err ssr-rules Start failed!

ilafz, 2023-03-27:
x86路由 内核5.10.46 升级了最新的5.15内核也是同样的情况
已经排除节点和使用的协议影响 
2023-03-28 07:00:52: -----------end------------
2023-03-28 07:00:50: Main node:Xray 1.8.0 (Xray, Penetrates Everything.) OpenWrt (go1.19.6 linux/amd64) Started!
2023-03-28 07:00:50: ----------start------------
2023-03-28 07:00:36: -----------end------------
2023-03-28 07:00:34: Main node:Xray 1.8.0 (Xray, Penetrates Everything.) OpenWrt (go1.19.6 linux/amd64) Started!
2023-03-28 07:00:34: ----------start------------
2023-03-28 06:47:59: -----------end------------
查看ssr日志正常

系统日志
Tue Mar 28 06:42:33 2023 user.err ssr-rules[32420]: Start failed!
Tue Mar 28 06:42:33 2023 daemon.info dnsmasq[32672]: Connected to system UBus
Tue Mar 28 06:43:59 2023 authpriv.info dropbear[25474]: Exit (root) from <192.168.99.131:62988>: Exited normally
Tue Mar 28 06:44:24 2023 daemon.info dnsmasq[909]: Connected to system UBus
Tue Mar 28 06:44:25 2023 user.err ssr-rules[1067]: Start failed!
Tue Mar 28 06:44:25 2023 daemon.info dnsmasq[1356]: Connected to system UBus
Tue Mar 28 06:44:51 2023 daemon.info dnsmasq[1759]: Connected to system UBus
Tue Mar 28 06:44:52 2023 user.err ssr-rules[1919]: Start failed!
Tue Mar 28 06:44:52 2023 daemon.info dnsmasq[2176]: Connected to system UBus
Tue Mar 28 06:47:29 2023 daemon.info dnsmasq[2682]: Connected to system UBus
Tue Mar 28 06:47:30 2023 user.err ssr-rules[2839]: Start failed!
Tue Mar 28 06:47:30 2023 daemon.info dnsmasq[3076]: Connected to system UBus
Tue Mar 28 06:47:58 2023 daemon.info dnsmasq[3552]: Connected to system UBus
Tue Mar 28 06:47:59 2023 user.err ssr-rules[3707]: Start failed!
Tue Mar 28 06:47:59 2023 daemon.info dnsmasq[3960]: Connected to system UBus
Tue Mar 28 06:59:57 2023 daemon.info dnsmasq[5605]: Connected to system UBus
Tue Mar 28 07:00:14 2023 daemon.err uhttpd[3100]: Command failed: Not found
Tue Mar 28 07:00:35 2023 user.err ssr-rules[6187]: Start failed!
Tue Mar 28 07:00:35 2023 daemon.err uhttpd[3100]: ssr-rules[6187]: Start failed!

路由器本机测试国内国外都正常

TCP 透明代理运行中 
游戏模式 UDP 中继运行中 
SOCKS5 代理服务端（全局）未运行 
本机服务端未运行 
KcpTun 版本号20200701-1 KcpTun未运行 
【谷歌】连通性检查 连接正常 
【百度】连通性检查 连接正常 
【GFW 列表】数据库 0 条记录 
【中国大陆 IP 段】数据库 0 条记录 
【服务器端口】检查
[us] OK.
 

问题：所有的客户端都不走代理 使用全局模式也是同样的情况

ilafz, 2023-03-27:
passwall可以正常使用 但是passwall不走mosdns dnsmasq的ipset geo分流 删除chinaip规则就全局代理了
ssrp 旧版本使用xray core1.7.5 xtls direct协议 是完全正常的 升级了1.8.0之后 出现了这样

ilafz, 2023-03-28:
暂时换回了passwall  把chinadns指向mosdns 删除规则文件 分流正常了

felixxiao99, 2023-03-31:
我之前发过issure，发了几次，没过多久就被关闭，问题也没修复，不知道和你的情况是不是相同的原因，

我遇到的问题是
最新编译的R2S固件作为旁路由使用，使用ShadowSocksR Plus+不能科学上网，只能访问国内网站
年前编译的固件在相同旁路由设置下是可以科学上网的

另外：个人感觉问题应该是出在ShadowSocksR Plus+插件上，除了自编译固件有这个问题，我也下载了骷髅头大神的R2S固件，早期的版本(1月17号)设置旁路由能科学上网，用最新的固件(3月23号 )也不能科学上网

ilafz, 2023-03-31:
> 

现在用passwall2 问题都解决了 自带geo分流 不用ssrp了

closed 2023-04-02 by aiamadeus