coolsnowwolf/lede #11038: Openwrt使用p3terx编译失败 超过6小时，但只改了一行参数，请帮忙看看

carlliuxx, 2023-03-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用https://github.com/P3TERX/Actions-OpenWrt编译的，之前都好好的，3个小时多点就完成了，但是这次增加了网易云音乐，就显示超过6小时被取消，请大家帮忙看看什么原因？尝试了多次，都不行

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：The job running on runner GitHub Actions 2 has exceeded the maximum execution time of 360 minutes.

[image]
[image]
[image]

EnnawYang, 2023-03-28:
node太耗时了

carlliuxx, 2023-03-28:
> node太耗时了

所以只能本地编译了么？:(

Cathgao, 2023-03-28:
本地只编译插件

WYC-2020, 2023-03-28:
我也编译了node 也就耗时一共也就耗时4个多小时
[image]
你是编译了多少插件，我都算编译的比较多了，也就四个多小时 哈哈

jjm2473, 2023-03-28:
node编译要半天

carlliuxx, 2023-03-28:
> 本地只编译插件

请问怎么只编译插件呀？小白~求指教~

carlliuxx, 2023-03-28:
> > 本地只编译插件
> 
> 请问怎么只编译插件呀？小白~ 求指教~

查到了，这样？对么？
#在 luci -> application 选中插件，开始编译
    make package/luci-app-unblockneteasemusic/compile V=s

closed 2023-04-02 by aiamadeus