coolsnowwolf/lede #11039: BPI-R3 sfp接口问题

dd-bpir3, 2023-03-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [* ] 没有类似的issue

## 2. 详细叙述
sfp接口能发现模块，但是获取不到IP地址
### (1) 具体问题

A：使用TL-SM410U得2.5G电口模块，向下兼容1G。spf2接口绑定到lan。连通后电脑无法获取IP地址。

### (2) 路由器型号和固件版本

A：主机型号	Bananapi BPI-R3
      固件版本	OpenWrt R23.3.3 / LuCI Master (git-23.082.27304-adda8b2)

### (3) 详细日志

A：root@OpenWrt:/# [27280.089145] sfp sfp1: module transmit fault indicated
[27280.094224] sfp sfp1: module removed
[27292.798714] sfp sfp2: module TP-LINK          TL-SM410U        rev 2.0  sn 122C304000986    dc 221228  

root@OpenWrt:/# ifconfig -a
br-lan    Link encap:Ethernet  HWaddr EA:FB:AE:D1:3E:73  
          inet addr:192.168.50.1  Bcast:192.168.50.255  Mask:255.255.255.0
          inet6 addr: fe80::e8fb:aeff:fed1:3e73/64 Scope:Link
          inet6 addr: fd7a:178d:efe5::1/60 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1488078 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2254358 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:152447256 (145.3 MiB)  TX bytes:5454282891 (5.0 GiB)

br-wan    Link encap:Ethernet  HWaddr EA:FB:AE:D1:3E:73  
          inet addr:192.168.1.199  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::e8fb:aeff:fed1:3e73/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2467252 errors:0 dropped:17832 overruns:0 frame:0
          TX packets:2093042 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3553429517 (3.3 GiB)  TX bytes:213739764 (203.8 MiB)

eth0      Link encap:Ethernet  HWaddr EA:FB:AE:D1:3E:73  
          inet6 addr: fe80::e8fb:aeff:fed1:3e73/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1504  Metric:1
          RX packets:4982377 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2976247 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:6297929683 (5.8 GiB)  TX bytes:968302978 (923.4 MiB)
          Interrupt:132 

eth1      Link encap:Ethernet  HWaddr EA:FB:AE:D1:3E:74  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:132 

lan1      Link encap:Ethernet  HWaddr EA:FB:AE:D1:3E:73  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lan2      Link encap:Ethernet  HWaddr EA:FB:AE:D1:3E:73  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:1209 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1340 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:371769 (363.0 KiB)  TX bytes:434944 (424.7 KiB)

lan3      Link encap:Ethernet  HWaddr EA:FB:AE:D1:3E:73  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lan4      Link encap:Ethernet  HWaddr EA:FB:AE:D1:3E:73  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:246803 errors:0 dropped:0 overruns:0 frame:0
          TX packets:439795 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:29254345 (27.8 MiB)  TX bytes:677500475 (646.1 MiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:49750 errors:0 dropped:0 overruns:0 frame:0
          TX packets:49750 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3976314 (3.7 MiB)  TX bytes:3976314 (3.7 MiB)

sfp2      Link encap:Ethernet  HWaddr EA:FB:AE:D1:3E:73  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sit0      Link encap:IPv6-in-IPv4  
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tunl0     Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wan       Link encap:Ethernet  HWaddr EA:FB:AE:D1:3E:74  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:4734365 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2286426 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:6178620783 (5.7 GiB)  TX bytes:235790781 (224.8 MiB)

wlan0     Link encap:Ethernet  HWaddr 00:0C:43:26:60:00  
          inet6 addr: fe80::20c:43ff:fe26:6000/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:15045 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:5266886 (5.0 MiB)

wlan1     Link encap:Ethernet  HWaddr 82:0C:43:26:60:00  
          inet6 addr: fe80::800c:43ff:fe26:6000/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1441642 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4212652 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:162767533 (155.2 MiB)  TX bytes:5480910225 (5.1 GiB)

qjiang888, 2024-04-19:
楼主解决了吗？

dd-bpir3, 2024-05-31:
> 楼主解决了吗？
没

coolsnowwolf, 2026-09-05:
核对了当前 LEDE 与 OpenWrt 主线的 BPI-R3 DTS：SFP2 已正确声明为 2500base-x/in-band-status，主线还补充了 SFP1 的 netdev 命名；未发现针对 TL-SM410U「模块检测到但无 IP」的可移植软件修复。原始日志同时显示 module transmit fault、module removed 以及 eth1 无 carrier，指向模块/链路协商或硬件兼容性，而非 DHCP 配置。该报告基于 R23.3.3，已过时且无法在当前代码复现，先关闭；若仍存在，请使用当前固件并附 ethtool、dmesg 和模块 EEPROM 信息重新提交。

closed 2026-09-05 by coolsnowwolf