coolsnowwolf/lede #11040: 极路由4增强版编译出错

wjc0325, 2023-03-28:
使用GitHub云编译极路由4增强版编译出错具体代码如下有没有知道的大佬试了好几次都是这样
  mipsel-openwrt-linux-musl-gcc -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -o /workdir/openwrt/build_dir/target-mipsel_24kc_musl/node-v12.22.1/out/Release/obj.target/openssl/deps/openssl/openssl/ssl/bio_ssl.o ../deps/openssl/openssl/ssl/bio_ssl.c '-DV8_DEPRECATION_WARNINGS' '-DV8_IMMINENT_DEPRECATION_WARNINGS' '-D__STDC_FORMAT_MACROS' '-DOPENSSL_NO_PINSHARED' '-DOPENSSL_THREADS' '-DOPENSSL_NO_HW' '-DNDEBUG' '-DOPENSSL_USE_NODELETE' '-DL_ENDIAN' '-DOPENSSL_PIC' '-DOPENSSL_CPUID_OBJ' '-DOPENSSL_IA32_SSE2' '-DOPENSSL_BN_ASM_MONT' '-DOPENSSL_BN_ASM_MONT5' '-DOPENSSL_BN_ASM_GF2m' '-DSHA1_ASM' '-DSHA256_ASM' '-DSHA512_ASM' '-DKECCAK1600_ASM' '-DRC4_ASM' '-DMD5_ASM' '-DAESNI_ASM' '-DVPAES_ASM' '-DGHASH_ASM' '-DECP_NISTZ256_ASM' '-DX25519_ASM' '-DPOLY1305_ASM' '-DOPENSSLDIR="/etc/ssl"' '-DENGINESDIR="/dev/null"' '-DTERMIOS' -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include -I../deps/openssl/openssl -I../deps/openssl/openssl/include -I../deps/openssl/openssl/crypto -I../deps/openssl/openssl/crypto/include -I../deps/openssl/openssl/crypto/modes -I../deps/openssl/openssl/crypto/ec/curve448 -I../deps/openssl/openssl/crypto/ec/curve448/arch_32 -I../deps/openssl/config -I../deps/openssl/config/archs/linux-x86_64/asm -I../deps/openssl/config/archs/linux-x86_64/asm/include -I../deps/openssl/config/archs/linux-x86_64/asm/crypto -I../deps/openssl/config/archs/linux-x86_64/asm/crypto/include/internal  -pthread -Wall -Wextra -Wno-unused-parameter -Wa,--noexecstack -Wall -O3 -pthread -m64 -Wall -O3 -Wno-missing-field-initializers -Wno-old-style-declaration -O3 -fno-omit-frame-pointer  -MMD -MF /workdir/openwrt/build_dir/target-mipsel_24kc_musl/node-v12.22.1/out/Release/.deps//workdir/openwrt/build_dir/target-mipsel_24kc_musl/node-v12.22.1/out/Release/obj.target/openssl/deps/openssl/openssl/ssl/bio_ssl.o.d.raw  -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/node-v12.22.1=node-v12.22.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include  -c
mipsel-openwrt-linux-musl-gcc: error: unrecognized command line option '-m64'
make[5]: *** [deps/openssl/openssl.target.mk:867: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/node-v12.22.1/out/Release/obj.target/openssl/deps/openssl/openssl/ssl/bio_ssl.o] Error 1
make[4]: *** [Makefile:104: node] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/node-v12.22.1'
make[3]: *** [Makefile:235: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/node-v12.22.1/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#0.56#0.14#0.69
    ERROR: package/feeds/packages/node failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/node/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

FeiGenb, 2023-03-28:
我也是这个问题，问题出在'-m64'这个命令，正常是不需要的，但是不知道改哪里的才可以

coolsnowwolf, 2023-03-30:
[image]
看起来 nodejs 官方已经彻底放弃 mips 的支持了，我尽力了

beenbin, 2023-04-01:
也是云编译，三个月前编译成功，现在也跳出这个问题了，有办法解决么

FeiGenb, 2023-04-02:
临时办法：
我参考了ImmortalWrt的node.js的Makefile，昨天尝试自己修改了一下node.js的Makefile，将里68行代码改为了：
DEPENDS:=@(HAS_FPU||KERNEL_MIPS_FPU_EMULATOR) @!arc @!armeb @!powerpc \
然后编译成功了，不过没了那个SOFT_FLOAT，会造成什么后果我也不知道，因为太晚了没刷机继续试验；
应该是正确的办法，我还没试过：
不改代码，尝试在编译配置页面，languages-node.js里面，找到选择node.js版本的选项，选择node.js 10.X，不要选择12.X，然后编译，还是由于太晚，我还没试过：
[image]

wjc0325, 2023-04-02:
> 临时办法： 我参考了ImmortalWrt的node.js的Makefile，昨天尝试自己修改了一下node.js的Makefile，将里68行代码改为了： DEPENDS:=@(HAS_FPU||KERNEL_MIPS_FPU_EMULATOR) @!arc @!armeb @!powerpc 然后编译成功了，不过没了那个SOFT_FLOAT，会造成什么后果我也不知道，暂时来说，没发现问题，拿来做单臂路由的核心功能openclash、adguardhome都可以正常使用

兄弟 ，languages-node.js里面，找到选择node.js版本的选项，选择node.js 10.X修改完这个可以成功编译吗？

FeiGenb, 2023-04-02:
> 

不行，暂时可行的办法还是上面说到的临时办法，node.js的Makefile在lede/build_dir/target-mipsel_24kc_musl里面，node-v12.*文件夹的根目录

FeiGenb, 2023-04-02:
> 兄弟 ，languages-node.js里面，找到选择node.js版本的选项，选择node.js 10.X修改完这个可以成功编译吗？

不行，我修改了上面的答复，参考上面答复可以临时解决问题，但是会不会造成别的BUG可能需要lean大帮忙看看了。如果你和我编译的目标硬件也是HC5962，node.js的Makefile在lede/build_dir/target-mipsel_24kc_musl/node-v12.22.1根目录

xiangfeigao, 2023-04-19:
这个路由器看来要完蛋了

xiangfeigao, 2023-04-19:
有什么好办法解决node.js
ma

Cathgao, 2023-04-19:
mips直接不用nodejs得了，这玩意编译久占用高，除了网易云解锁还有别的地方用得到吗。
MIPS都已经是历史的弃子了，别再折磨它了

spin200, 2023-04-21:
😩我也遇到了这个问题，就选择了架构和型号，其他啥也没改，保持默认，云编译不通过，一直错误，大佬们有没有办法解决，或者是不是说换其他源码？

xiangfeigao, 2023-05-19:
可以正常囊编译了

xiangfeigao, 2023-10-16:
还有人折腾啊

closed 2026-09-05 by coolsnowwolf