coolsnowwolf/lede #11043: Turbo ACC 流量分载 如何关闭，没有选项

sunyink, 2023-03-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
[image]

### (1) 具体问题

A：图里的Turbo ACC，流量分载默认就开了个QCA-NSS-ECM，没法关闭；我甚至去etc/Turboacc 里把前面几个参数都改成0重启都无效。现在内网SMB被限制带10mb以下了....原来和刚重启好那会儿，可以跑80mb-100mb。

### (2) 路由器型号和固件版本

A：Netgear Nighthawk X4S R7800 OpenWrt R23.2.14 / LuCI Master (git-23.061.31098-da92698)

### (3) 详细日志

A：

Her0R0cket, 2023-04-03:
防火墙里检查下有没有这个设置

Travel2Here, 2023-04-14:
楼主找到解决没？我没编译turbo acc，但在启动日志里也发现有关于offload的信息，且也发现samba在10M以下。

sunyink, 2023-05-01:
> 楼主找到解决没？我没编译turbo acc，但在启动日志里也发现有关于offload的信息，且也发现samba在10M以下。

换了个官方源。但NSS对高通路由很重要，不然外网转发只有300M。

closed 2023-05-21 by sunyink