coolsnowwolf/lede #11044: 本地编译出错

pangfeng161, 2023-03-31:
本地编译X86到这步出错,具体代码如下有没有知道的大佬能解决
Applying ./patches/010-gcc.patch using plaintext: 
patching file build/gen.py
touch /home/pangzhao/lede/build_dir/hostpkg/gn-2023-02-17-b25a2f8c/.prepared40f9c769f93e797072cb45413b7224c5_6664517399ebbbc92a37c5bb081b5c53
python3 /home/pangzhao/lede/build_dir/hostpkg/gn-2023-02-17-b25a2f8c/build/gen.py --no-last-commit-position
touch /home/pangzhao/lede/build_dir/hostpkg/gn-2023-02-17-b25a2f8c/.configured
ninja -C /home/pangzhao/lede/build_dir/hostpkg/gn-2023-02-17-b25a2f8c/out
ninja: Entering directory `/home/pangzhao/lede/build_dir/hostpkg/gn-2023-02-17-b25a2f8c/out'
[1/287] CXX src/base/environment.o
FAILED: src/base/environment.o 
g++ -MMD -MF src/base/environment.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++17 -c ../src/base/environment.cc -o src/base/environment.o
g++: error: unrecognized command line option '-Wextra-semi'; did you mean '-Wextra'?
[2/287] CXX src/base/command_line.o
FAILED: src/base/command_line.o 
g++ -MMD -MF src/base/command_line.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++17 -c ../src/base/command_line.cc -o src/base/command_line.o
g++: error: unrecognized command line option '-Wextra-semi'; did you mean '-Wextra'?
[3/287] CXX src/base/files/file_enumerator.o
FAILED: src/base/files/file_enumerator.o 
g++ -MMD -MF src/base/files/file_enumerator.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++17 -c ../src/base/files/file_enumerator.cc -o src/base/files/file_enumerator.o
g++: error: unrecognized command line option '-Wextra-semi'; did you mean '-Wextra'?
[4/287] CXX src/base/files/file.o
FAILED: src/base/files/file.o 
g++ -MMD -MF src/base/files/file.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++17 -c ../src/base/files/file.cc -o src/base/files/file.o
g++: error: unrecognized command line option '-Wextra-semi'; did you mean '-Wextra'?
[5/287] CXX src/base/files/file_path.o
FAILED: src/base/files/file_path.o 
g++ -MMD -MF src/base/files/file_path.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++17 -c ../src/base/files/file_path.cc -o src/base/files/file_path.o
g++: error: unrecognized command line option '-Wextra-semi'; did you mean '-Wextra'?
[6/287] CXX src/base/files/file_path_constants.o
FAILED: src/base/files/file_path_constants.o 
g++ -MMD -MF src/base/files/file_path_constants.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++17 -c ../src/base/files/file_path_constants.cc -o src/base/files/file_path_constants.o
g++: error: unrecognized command line option '-Wextra-semi'; did you mean '-Wextra'?
ninja: build stopped: subcommand failed.
Makefile:60: recipe for target '/home/pangzhao/lede/build_dir/hostpkg/gn-2023-02-17-b25a2f8c/.built' failed
make[3]: *** [/home/pangzhao/lede/build_dir/hostpkg/gn-2023-02-17-b25a2f8c/.built] Error 1
make[3]: Leaving directory '/home/pangzhao/lede/feeds/helloworld/gn'
time: package/feeds/helloworld/gn/host-compile#7.66#5.61#14.73
    ERROR: package/feeds/helloworld/gn [host] failed to build.
package/Makefile:114: recipe for target 'package/feeds/helloworld/gn/host/compile' failed
make[2]: *** [package/feeds/helloworld/gn/host/compile] Error 1
make[2]: Leaving directory '/home/pangzhao/lede'
package/Makefile:110: recipe for target '/home/pangzhao/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/pangzhao/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/pangzhao/lede'
/home/pangzhao/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2

WYC-2020, 2023-03-31:
那是ssr那边的gn 有问题，有两个方法，一个是你直接用xiaorouji仓库的naiveproxy 直接覆盖，缺陷就是不是最新，如果非要用ssr的 那就 更新下那个补丁
[image]
但是低版本的gcc不知道编译又没问题，因为我是gcc12 低版本的你自己测试一下了

WYC-2020, 2023-04-01:
现在xiaorouji那边也被人更新了，并且并没有解决这个问题，要用就只能更新patch，我测试了最低gcc 8 都没问题，其他低于8的自测

pangfeng161, 2023-04-01:
这么明显bug没人修复

kentooo, 2023-04-02:
> 这么明显bug没人修复

这个bug也不算明显，可能编译平台不同，会有不同的问题吧，
我就是xiaorouji那边发现这个bug的人，目前我两台linux，archlinux上编译是通过的，opensuse上编译有问题，
我用的openwrt 官方22.03.03 sdk，这个不确定是不是系统的问题

WYC-2020, 2023-04-03:
> > 这么明显bug没人修复
> 
> 这个bug也不算明显，可能编译平台不同，会有不同的问题吧， 我就是xiaorouji那边发现这个bug的人，目前我两台linux，archlinux上编译是通过的，opensuse上编译有问题， 我用的openwrt 官方22.03.03 sdk，这个不确定是不是系统的问题

那是因为 系统自带的g++ 版本不同导致的 只要本机自带g++ 大于7 就没问题，8以下的任何版本都会出这个问题，按我那个方法 去掉那个选项就行了，反正这玩意也是一个工具链而已，我试过 去掉后 兼容所有g++

WYC-2020, 2023-04-03:
要么就自己升级本机自带的g++ 版本

WYC-2020, 2023-04-03:
别问为啥我知道，问就是因为我验证过 哈哈

pangfeng161, 2023-04-05:
> 别问为啥我知道，问就是因为我验证过 哈哈

具体解决方法公布出来，对小白还是一道坎

sclsyin, 2023-04-06:
本地已升级GCC++到8，问题解决。但云编译不知道该怎么修改。

pangfeng161, 2023-04-06:
> 本地已升级GCC++到8，问题解决。但云编译不知道该怎么修改。

刚好相反，云编译没什么问题

andyhang1980, 2023-04-11:
> ssr那边的gn 有问题

补丁在哪儿？

closed 2023-04-14 by aiamadeus