coolsnowwolf/lede #11047: 设置无法保存，x86软路由

KnowledegAndAction, 2023-03-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：3865U+256GSSD配置的软路由，使用2021年2月编译自用的固件一直稳定运行到现在。因为想升级一些插件所以在2023年3月重新编译了固件。新固件问题表现为安装（dd写入ssd）成功后首次开机一切正常，但是好像在关机（poweroff）之后再开就会出现修改设置无法保存，luci界面右上角一直提示，而且重启后设置修改确实丢失。已搜索到的问题基本都是硬路由空间满的原因，但我这个空间足够。sda3是把ssd盘剩余空间用f2fs格式创建的分区。
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root              2061160    765792   1278984  37% /rom
tmpfs                  4040132     18232   4021900   0% /tmp
/dev/sda2              2061160    765792   1278984  37% /
/dev/sda1                64468      5036     58124   8% /boot
/dev/sda1                64468      5036     58124   8% /boot
tmpfs                      512         0       512   0% /dev
cgroup                 4040132         0   4040132   0% /sys/fs/cgroup
/dev/sda3            242032824   2721996 239310828   1% /mnt/sda3
/dev/sda1                64468      5036     58124   8% /mnt/sda1

### (2) 路由器型号和固件版本

A：OpenWrt R23.3.3 / LuCI Master (git-23.080.41392-a06e7b6)

### (3) 详细日志

A：Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.255305] cryptodev: driver 1.12 loaded.
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.260958] i2c_dev: i2c /dev entries driver
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.268631] fuse: init (API version 7.34)
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.274406] 8139too: 8139too Fast Ethernet driver 0.9.28
Fri Mar 31 22:57:04 2023 kern.notice kernel: [    9.296312] Bridge firewalling registered
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.302006] e1000e: Intel(R) PRO/1000 Network Driver
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.307364] e1000e: Copyright(c) 1999 - 2015 Intel Corporation.
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.316800] i40e: Intel(R) Ethernet Connection XL710 Network Driver
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.323358] i40e: Copyright (c) 2013 - 2019 Intel Corporation.
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.342311] i915 0000:00:02.0: [drm] Incompatible option enable_guc=3 - GuC submission is N/A
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.355107] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.631039] i915 0000:00:02.0: [drm] failed to retrieve link info, disabling eDP
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.668005] i915 0000:00:02.0: [drm] GuC firmware i915/kbl_guc_62.0.0.bin version 62.0 submission:disabled
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.678248] i915 0000:00:02.0: [drm] GuC SLPC: disabled
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.683816] i915 0000:00:02.0: [drm] HuC firmware i915/kbl_huc_4.0.0.bin version 4.0 authenticated:yes
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.695129] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.704762] ACPI: video: Video Device [GFX0] (multi-head: yes  rom: no  post: no)
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.713067] input: Video Bus as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input3
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.728165] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.737642] iavf: Intel(R) Ethernet Adaptive Virtual Function Network Driver
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.741263] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.746690] Copyright (c) 2013 - 2018 Intel Corporation.
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.761145] Intel(R) 2.5G Ethernet Linux Driver
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.766742] Copyright(c) 2018 Intel Corporation.
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.791979] usbcore: registered new interface driver r8152
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.802785] usbcore: registered new interface driver rtl8150
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.814176] usbcore: registered new interface driver ums-alauda
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.821282] usbcore: registered new interface driver ums-cypress
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.828438] usbcore: registered new interface driver ums-datafab
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.835618] usbcore: registered new interface driver ums-freecom
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.842809] usbcore: registered new interface driver ums-isd200
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.849884] usbcore: registered new interface driver ums-jumpshot
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.857132] usbcore: registered new interface driver ums-karma
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.864215] usbcore: registered new interface driver ums-sddr09
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.871300] usbcore: registered new interface driver ums-sddr55
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.878389] usbcore: registered new interface driver ums-usbat
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.889668] xt_time: kernel timezone is -0000
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.920743] [drm] amdgpu kernel modesetting enabled.
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.927739] usbcore: registered new interface driver aqc111
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.934755] usbcore: registered new interface driver ax88179_178a
Fri Mar 31 22:57:04 2023 kern.info kernel: [    9.953876] PPP generic driver version 2.4.2
[... 429 lines cut ...]
Fri Mar 31 22:57:31 2023 daemon.err smbd[6615]:   Failed to open /etc/samba/secrets.tdb
Fri Mar 31 22:57:31 2023 daemon.err smbd[6615]: [2023/03/31 22:57:31.660195,  0] ../../lib/util/become_daemon.c:121(exit_daemon)
Fri Mar 31 22:57:31 2023 daemon.err smbd[6615]:   exit_daemon: daemon failed to start: smbd can not open secrets.tdb, error code 13
Fri Mar 31 22:57:35 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)
Fri Mar 31 22:57:36 2023 daemon.err smbd[6965]: [2023/03/31 22:57:36.704791,  0] ../../source3/passdb/secrets.c:66(secrets_init_path)
Fri Mar 31 22:57:36 2023 daemon.err smbd[6965]:   Failed to open /etc/samba/secrets.tdb
Fri Mar 31 22:57:36 2023 daemon.err smbd[6965]: [2023/03/31 22:57:36.705016,  0] ../../source3/passdb/secrets.c:66(secrets_init_path)
Fri Mar 31 22:57:36 2023 daemon.err smbd[6965]:   Failed to open /etc/samba/secrets.tdb
Fri Mar 31 22:57:36 2023 daemon.err smbd[6965]: [2023/03/31 22:57:36.705051,  0] ../../lib/util/become_daemon.c:121(exit_daemon)
Fri Mar 31 22:57:36 2023 daemon.err smbd[6965]:   exit_daemon: daemon failed to start: smbd can not open secrets.tdb, error code 13
Fri Mar 31 22:57:37 2023 authpriv.info dropbear[6967]: Child connection from 192.168.1.221:57261
Fri Mar 31 22:57:37 2023 authpriv.notice dropbear[6967]: Password auth succeeded for 'root' from 192.168.1.221:57261
Fri Mar 31 22:57:37 2023 authpriv.info dropbear[6977]: Child connection from 192.168.1.221:57262
Fri Mar 31 22:57:37 2023 authpriv.notice dropbear[6977]: Password auth succeeded for 'root' from 192.168.1.221:57262
Fri Mar 31 22:57:41 2023 daemon.err smbd[6982]: [2023/03/31 22:57:41.752403,  0] ../../source3/passdb/secrets.c:66(secrets_init_path)
Fri Mar 31 22:57:41 2023 daemon.err smbd[6982]:   Failed to open /etc/samba/secrets.tdb
Fri Mar 31 22:57:41 2023 daemon.err smbd[6982]: [2023/03/31 22:57:41.752627,  0] ../../source3/passdb/secrets.c:66(secrets_init_path)
Fri Mar 31 22:57:41 2023 daemon.err smbd[6982]:   Failed to open /etc/samba/secrets.tdb
Fri Mar 31 22:57:41 2023 daemon.err smbd[6982]: [2023/03/31 22:57:41.752660,  0] ../../lib/util/become_daemon.c:121(exit_daemon)
Fri Mar 31 22:57:41 2023 daemon.err smbd[6982]:   exit_daemon: daemon failed to start: smbd can not open secrets.tdb, error code 13
Fri Mar 31 22:57:41 2023 daemon.info procd: Instance samba4::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Fri Mar 31 22:57:43 2023 daemon.err nmbd[5530]: [2023/03/31 22:57:43.745099,  0] ../../source3/nmbd/nmbd_become_lmb.c:398(become_local_master_stage2)
Fri Mar 31 22:57:43 2023 daemon.err nmbd[5530]:   *****
Fri Mar 31 22:57:43 2023 daemon.err nmbd[5530]:
Fri Mar 31 22:57:43 2023 daemon.err nmbd[5530]:   Samba name server X86OPENWRT is now a local master browser for workgroup WORKGROUP on subnet 192.168.1.1

helllkz, 2023-04-02:
一模一样的问题，我也是才刷了更新到最新版本，通过在线升级方式出现这问题后，重新全新编译后通过写盘方式，也还是有这个问题，而且部分配置提交了也没生效，比如helloworld，我是EXT4,Efi的固件

helllkz, 2023-04-02:
貌似是文件权限问题，因为wan口配置没法修改保存生效，我就想看看实际文件是不是没修改成功，通过ssh上去看确实是没修改起，通过手动方式修改，结果提示文件是只读的
[image]
[image]

mspzyzh, 2023-04-02:
我编译的X86 EXT4固件也是配置无法保存，要手动挂载为读写然后生成配置

WYC-2020, 2023-04-02:
有吗？我也是86没遇到，并且没有合并最近那个断电重新挂载那个提交一样可行

alfredfenglq, 2023-04-02:
是直接在OpenWrt里面dd的吗？我发现直接dd不行，得把盘挂到其他系统下dd才正常

helllkz, 2023-04-02:
> 是直接在OpenWrt里面dd的吗？我发现直接dd不行，得把盘挂到其他系统下dd才正常

我是windows里面用balenaEtcher写盘的，之前也是这样的，看了下日志，确实也出现了readonly
[    1.130373] VFS: Mounted root (ext4 filesystem) readonly on device 8:2.

helllkz, 2023-04-02:
基本复现了这个问题了，目前的几个情况如下：

1. 文件系统只读会出现在第一次重启以后出现
2. 之前的流程里面，我把硬盘在win上面写盘后，会在软路由上测试启动，没问题了就断电放机柜里面去
3. 刚刚写盘后，直接没测试放机柜，把配置好，一切都正常，各种修改都没问题
4. 然后再重启，就发现又出现只读了

[image]

@coolsnowwolf，L大看看有没有这个问题呢

mspzyzh, 2023-04-02:
我虚拟机安装的，ext4的包会出现这种问题，重启后重新挂载为读写，再生成配置就好了

helllkz, 2023-04-02:
> 我虚拟机安装的，ext4的包会出现这种问题，重启后重新挂载为读写，再生成配置就好了

怎么修改呢，把default改为rw吗？需要其他选项不，只改根节点，不用改boot吧
[image]

mspzyzh, 2023-04-02:
我是使用命令挂载的 mount -o remount rw /

helllkz, 2023-04-02:
> 我是使用命令挂载的 mount -o remount rw /

执行了这个，然后又重新生成了挂载配置，再重启目前没出问题了

KnowledegAndAction, 2023-04-02:
> 基本复现了这个问题了，目前的几个情况如下：
> 
> 1. 文件系统只读会出现在第一次重启以后出现
> 2. 之前的流程里面，我把硬盘在win上面写盘后，会在软路由上测试启动，没问题了就断电放机柜里面去
> 3. 刚刚写盘后，直接没测试放机柜，把配置好，一切都正常，各种修改都没问题
> 4. 然后再重启，就发现又出现只读了
> 
> [image]
> 
> @coolsnowwolf，L大看看有没有这个问题呢

好像不是只有第一次会重启出现。我这里是每次重启都会出现。

KnowledegAndAction, 2023-04-02:
> > 我是使用命令挂载的 mount -o remount rw /
> 
> 执行了这个，然后又重新生成了挂载配置，再重启目前没出问题了

重新rw挂载后可以正常保存设置，但是只要重启，不管第几次重启，都会又变成只读，又不能保存设置。

KnowledegAndAction, 2023-04-02:
> 有吗？我也是86没遇到，并且没有合并最近那个断电重新挂载那个提交一样可行

断电重新挂载的提交，听起来就很像是这个问题的原因，每次重启就是断电，提交里有bug，你没合并所以反而不会出现这个问题。

wushenghn, 2023-04-02:
一样的情况overlay好像挂载有问题
[    9.693701] loop0: detected capacity change from 0 to 4194304
[    9.731620] loop0: detected capacity change from 4194304 to 3902720
[    9.735154] block: overlay filesystem in /dev/loop0 has not been formatted yet
[    9.739182] block: attempting to load /etc/config/fstab
[    9.740022] block: unable to load configuration (fstab: Entry not found)
[    9.740871] block: no usable configuration
[    9.741830] loop0: detected capacity change from 0 to 4194304
[    9.781659] loop0: detected capacity change from 4194304 to 3902720
[    9.782750] mount_root: overlay filesystem in /dev/loop0 has not been formatted yet
[    9.784358] mount_root: no usable overlay filesystem found, using tmpfs overlay
[    9.786632] overlayfs: null uuid detected in lower fs '/', falling back to xino=off,index=off,nfs_export=off.

littoy, 2023-04-03:
l大新的提交修复了

helllkz, 2023-04-03:
> 

我是执行了那个命令，然后自动挂载里面重新生成了配置，试了2次系统里面重启，目前是没出现只读，但是直接断电的方式还没试，不太清楚

mspzyzh, 2023-04-03:
目前我的解决方法是重启后使用命令 mount -o remount rw / 重新挂载，然后在挂载页面生成配置，再重启就不会是只读了

KnowledegAndAction, 2023-04-03:
> 目前我的解决方法是重启后使用命令 mount -o remount rw / 重新挂载，然后在挂载页面生成配置，再重启就不会是只读了

你说的”生成配置“是指什么？重挂载后，修改配置保存和应用确实可以，但是只要重启，又会恢复成只读状态，不能继续修改配置。

KnowledegAndAction, 2023-04-04:
重新拉取编译后问题已解决。

closed 2023-04-04 by KnowledegAndAction

kxdn3, 2024-02-25:
额，又出现这个问题了

mspzyzh, 2024-02-25:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。