coolsnowwolf/lede #11054: K2P编译生成的固件网口映射异常

jjm2473, 2023-04-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：挺长时间没编译K2P的固件了，今天拉代码编译了下（[提交ID](https://github.com/coolsnowwolf/lede/commit/bad53b1207fa0889a5f3037223eaae0e91c4d503)），发现物理网口全是逻辑LAN口。物理WAN口接了上一级的路由，DNSMASQ却认为LAN口DHCP服务冲突而不启动。然后电脑手动配置了1.x网段的IP，发现不管插K2P的哪个物理网口都能访问1.1的后台。
系统里面只有eth0和eth1。
`swconfig list`没有任何输出

目前这种情况到底咋回事？是不是我编译时少选了什么内核模块？

### (2) 路由器型号和固件版本

A：K2P，主线master分支

### (3) 详细日志

A：
<details><summary>生成的默认配置/etc/config/network</summary>
<p>

config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option packet_steering '1'
        option ula_prefix 'fdcb:e565:5ca3::/48'

config interface 'lan'
        option type 'bridge'
        option ifname 'eth0'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'wan'
        option ifname 'eth1'
        option proto 'dhcp'

config interface 'wan6'
        option ifname 'eth1'
        option proto 'dhcpv6'

config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option ports '0 1 2 3 6'

config switch_vlan
        option device 'switch0'
        option vlan '2'
        option ports '4 5'

</p>
</details> 
[... 766 lines cut ...]
[   98.839262] EventExtEventHandler: Unknown Ext Event(7)
[   98.939316] EventExtEventHandler: Unknown Ext Event(7)
[   98.954294] [DfsCacNormalStart] Normal start. Enable MAC TX
[   98.965885] wifi_sys_linkup(), wdev idx = 4
[   98.974241] wtc_acquire_groupkey_wcid: Found a non-occupied wtbl_idx:126 for WDEV_TYPE:1
[   98.974241]  LinkToOmacIdx = 14, LinkToWdevType = 1
[   99.000248] bssUpdateBmcMngRate (BSS_INFO_BROADCAST_INFO), CmdBssInfoBmcRate.u2BcTransmit= 8192, CmdBssInfoBmcRate.u2McTransmit = 8196
[   99.025110] MtCmdSetDbdcCtrl:(ret = 0)
[   99.033266] UpdateBeaconHandler, BCN_UPDATE_INIT, OmacIdx = 14
[   99.045053] APStartUpForMbss: BssIndex = 1 channel = 36
[   99.055693] MtCmdTxPowerDropCtrl: ucPowerDrop: 100, BandIdx: 1
[   99.069618] apidx 4 for WscUUIDInit
[   99.076633] Generate UUID for apidx(4)
[   99.177714] br-lan: port 2(rax0) entered blocking state
[   99.188173] br-lan: port 2(rax0) entered disabled state
[   99.199061] device rax0 entered promiscuous mode
[   99.208533] br-lan: port 2(rax0) entered blocking state
[   99.219011] br-lan: port 2(rax0) entered forwarding state
[   99.317057] br-lan: port 3(ra0) entered blocking state
[   99.327357] br-lan: port 3(ra0) entered disabled state
[   99.337975] device ra0 entered promiscuous mode
[   99.347161] br-lan: port 3(ra0) entered blocking state
[   99.357400] br-lan: port 3(ra0) entered forwarding state
</p>
</details>

aiamadeus, 2023-04-02:
换回 5.4，别上 5.10 了。

jjm2473, 2023-04-02:
谢谢，我试试5.4

jjm2473, 2023-04-02:
切换到测试内核，交换机识别正常了。

closed 2023-04-02 by jjm2473

SeanLi-OI, 2023-04-25:
同样的问题，安装5.10版本就会将所有K2P的5个口都映射到eth0，5.4版本没有ipv6 hwnat的patch，5.10版本又没法链接WAN。

SeanLi-OI, 2023-04-25:
对比了一下内核日志，应该缺少了mt7530的driver，但才疏学浅，不太知道怎么补充，大佬能不能重开下这个issue，我认为这个问题没有解决。

5.4 内核启动日志：
``
[    1.560080] mt753x 1e110000.gsw: LAN/WAN VLAN setting=llllw 
[    1.571350] mt753x 1e110000.gsw: Switch is MediaTek MT7530 rev 1
``

5.10中没有

jjm2473, 2023-04-27:
切换到5.4内核就没问题了