coolsnowwolf/lede #11056: 新编译的CR660X 5.10内核无法拨号

yuzhe1129, 2023-04-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
刚又git pull了 重新编译了一次5.4 和5.10版本，5.10的如果没插光猫的可以获取IP登录界面，如果插了光猫的线就获取不了，手机连上WIFI获取的是1段光猫的IP，换了5.4的版本固件一切正常，附上TTL信息，

### (2) 路由器型号和固件版本

CR660X

### (3) 详细日志

5.10内核的，

PandoraBox-Boot Version 2.1
Build:Jun 13 2021-17:26:48
SoC:MT7621AT
Memory-Testing...262144K OK
MT7621:CPU clock set to 880Mhz
PCIe:reset controller
PB-Boot running at 0x8FF9C000!
NAND: F59L1G81MA 128MiB
Using internal default config
Board:Xiaomi CR660X
peripheral:device reset
PandoraBox GPIO subsystem init
PandoraBox MT762x GPIO driver initialized.
GPIO_Mode_REG: 0x000AD028
PandoraBox LED subsystem init
LED [Power] registered
PandoraBox LED test
PandoraBox Button subsystem init
Button [Reset] registered
Enable USB#1 Power
MT7530 GSW initialized
MT7530 set LAN/WAN LLLLW
Booting from Flash...
Booting image from bc200000 ...
   Image Name:   MIPS OpenWrt Linux-5.10.176
   Image Type:   MIPS Linux Kernel Image (uncompressed)
   Data Size:    2750935 Bytes =  2.6 MB
   Load Address: 80001000
   Entry Point:  80001000
   Verifying Checksum ... OK
   Loading Kernel Image ... OK
   Flushing cache from 0x80001000 size 2750935 ... OK

OpenWrt kernel loader for MIPS based SoC
Copyright (C) 2011 Gabor Juhos <juhosg@openwrt.org>
Decompressing kernel... done!
Starting kernel at 80001000...

[    0.000000] Linux version 5.10.176 (cc@ubuntu) (mipsel-openwrt-linux-musl-gcc (        OpenWrt GCC 8.4.0 r5865-72efda15b) 8.4.0, GNU ld (GNU Binutils) 2.37) #0 SMP Sat A        pr 1 07:17:55 2023
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router CR660x
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[... 293 lines cut ...]
[   42.903950] device eth0 entered promiscuous mode
[   42.914165] br-lan: port 1(eth0) entered blocking state
[   42.919528] br-lan: port 1(eth0) entered forwarding state
[   42.956128] mtk_soc_eth 1e100000.ethernet eth1: configuring for fixed/rgmii lin        k mode
[   42.964785] mtk_soc_eth 1e100000.ethernet eth1: Link is Up - 1Gbps/Full - flow         control rx/tx
[   42.973887] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   43.971981] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   46.864222] br-lan: port 2(wlan0) entered blocking state
[   46.869658] br-lan: port 2(wlan0) entered disabled state
[   46.875613] device wlan0 entered promiscuous mode
[   46.880931] br-lan: port 2(wlan0) entered blocking state
[   46.886362] br-lan: port 2(wlan0) entered forwarding state
[   47.028583] br-lan: port 2(wlan0) entered disabled state
[   47.051828] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   47.058701] br-lan: port 2(wlan0) entered blocking state
[   47.064061] br-lan: port 2(wlan0) entered forwarding state
[   47.816343] br-lan: port 3(wlan1) entered blocking state
[   47.821669] br-lan: port 3(wlan1) entered disabled state
[   47.828006] device wlan1 entered promiscuous mode
[   47.833060] br-lan: port 3(wlan1) entered blocking state
[   47.838915] br-lan: port 3(wlan1) entered forwarding state
[   47.906216] br-lan: port 3(wlan1) entered disabled state
[   48.315033] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   48.321855] br-lan: port 3(wlan1) entered blocking state
[   48.327450] br-lan: port 3(wlan1) entered forwarding state

yuzhe1129, 2023-04-01:
5.4内核的信息
PandoraBox-Boot Version 2.1
Build:Jun 13 2021-17:26:48
SoC:MT7621AT
Memory-Testing...262144K OK
MT7621:CPU clock set to 880Mhz
PCIe:reset controller
PB-Boot running at 0x8FF9C000!
NAND: F59L1G81MA 128MiB
Using internal default config
Board:Xiaomi CR660X
peripheral:device reset
PandoraBox GPIO subsystem init
PandoraBox MT762x GPIO driver initialized.
GPIO_Mode_REG: 0x000AD028
PandoraBox LED subsystem init
LED [Power] registered
PandoraBox LED test
PandoraBox Button subsystem init
Button [Reset] registered
Enable USB#1 Power
MT7530 GSW initialized
MT7530 set LAN/WAN LLLLW                                                      0
Booting from Flash...
Booting image from bc200000 ...
   Image Name:   MIPS OpenWrt Linux-5.4.238
   Image Type:   MIPS Linux Kernel Image (uncompressed)
   Data Size:    2548702 Bytes =  2.4 MB
   Load Address: 80001000
   Entry Point:  80001000
   Verifying Checksum ... OK
   Loading Kernel Image ... OK
   Flushing cache from 0x80001000 size 2548702 ... OK

OpenWrt kernel loader for MIPS based SoC
Copyright (C) 2011 Gabor Juhos <juhosg@openwrt.org>
Decompressing kernel... done!
Starting kernel at 80001000...

[    0.000000] Linux version 5.4.238 (cc@ubuntu) (gcc version 8.4.0 (OpenWrt GCC   8.4.0 r5865-72efda15b)) #0 SMP Sat Apr 1 07:17:55 2023
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router CR660x
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 byt  es
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] percpu: Embedded 14 pages/cpu s26640 r8192 d22512 u57344
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,j  ffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, l  inear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, lin  ear)
[    0.000000] Writing ErrCtl register=00000390
[    0.000000] Readback ErrCtl register=00000390
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 250088K/262144K available (6190K kernel code, 273K rwdata  , 1300K rodata, 1248K init, 235K bss, 12056K reserved, 0K cma-reserved, 0K highm  em)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jif  fies.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 880MHz
[... 1249 lines cut ...]
1Gbps/Full - flow control rx/tx
ash: missing ]
root@OpenWrt:/# [   41.617183] br-lan: port 1(eth0) entered blocking state
/bin/ash: syntax error: unexpected "("
root@OpenWrt:/# [   41.622558] br-lan: port 1(eth0) entered disabled state
/bin/ash: syntax error: unexpected "("
root@OpenWrt:/# [   41.628536] device eth0 entered promiscuous mode
ash: missing ]
root@OpenWrt:/# [   41.638983] br-lan: port 1(eth0) entered blocking state
/bin/ash: syntax error: unexpected "("
root@OpenWrt:/# [   41.644264] br-lan: port 1(eth0) entered forwarding state
/bin/ash: syntax error: unexpected "("
root@OpenWrt:/# [   41.682001] mtk_soc_eth 1e100000.ethernet eth1: configuring f
or fixed/rgmii link mode
ash: missing ]
root@OpenWrt:/# [   41.690539] mtk_soc_eth 1e100000.ethernet eth1: Link is Up -
1Gbps/Full - flow control rx/tx
ash: missing ]
root@OpenWrt:/# [   41.700257] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes
 ready
/bin/ash: syntax error: unexpected "("
root@OpenWrt:/# [   43.023879] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becom
es ready
/bin/ash: syntax error: unexpected "("
root@OpenWrt:/# [   43.034204] pppoe-wan: renamed from ppp0

yuzhe1129, 2023-04-01:
刚才发现5.4内核的，居然无线没有，降回23.3.3了，新的怎么感觉都是BUG

qjiang888, 2023-04-01:
@yuzhe1129 请教大佬怎么降级回23.3.3

yuzhe1129, 2023-04-01:
> @yuzhe1129请教大佬怎么降级回23.3.3

之前编译的固件啊，刷回来啊，要不然呢

Kuron88, 2023-04-03:
我也是遇到了知音的问题，然后我把5.10跟5.4的内核跳转下，还是5.4的内核是正常的

Kuron88, 2023-04-04:
这个问题现在修复了没？

lwb1978, 2023-04-04:
> 这个问题现在修复了没？

截止今天并没有。

Kuron88, 2023-04-05:
好像恢复到5.4的内核去了

lwb1978, 2023-04-05:
> 好像恢复到5.4的内核去了

4月5日恢复到了5.4内核，重新编译后正常拨号了

closed 2023-05-15 by aiamadeus