coolsnowwolf/lede #11060: 更新源码后mt7620 编译出错

mitsukileung, 2023-04-01:
mt7620/yl-l1

/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.10.176/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_hiwifi_r33.dtb /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_hiwifi_r33.dtb.tmp
../dts/mt7620a.dtsi:596.22-604.4: ERROR (phandle_references): /wmac@10180000: Reference to non-existent node or label "pa_gpio_pins"

  also defined at ../dts/mt7620a_hiwifi_r33.dts:213.7-220.3
ERROR: Input tree has errors, aborting (use -f to force output)
make[5]: *** [Makefile:217: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_hiwifi_r33.dtb] Error 2
make[5]: Leaving directory '/workdir/openwrt/target/linux/ramips/image'
make[4]: *** [Makefile:23: install] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/ramips'
make[3]: Leaving directory '/workdir/openwrt/target/linux'
make[3]: *** [Makefile:11: install] Error 2
time: target/linux/install#87.63#14.67#98.89
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:24: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

zfdx123, 2023-04-04:
下次记得带上型号，不然都注意不到，我会排查一下

zfdx123, 2023-04-04:
找到原因了我会尽快pr

wsklkl, 2023-04-04:
/home/jw/2/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.10.176/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /home/jw/2/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_hiwifi_r33.dtb /home/jw/2/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_hiwifi_r33.dtb.tmp
../dts/mt7620a.dtsi:596.22-604.4: ERROR (phandle_references): /wmac@10180000: Reference to non-existent node or label "pa_gpio_pins"

  also defined at ../dts/mt7620a_hiwifi_r33.dts:213.7-220.3
ERROR: Input tree has errors, aborting (use -f to force output)
make[5]: *** [Makefile:217: /home/jw/2/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_hiwifi_r33.dtb] Error 2
make[5]: Leaving directory '/home/jw/2/lede/target/linux/ramips/image'
make[4]: *** [Makefile:23: install] Error 2
make[4]: Leaving directory '/home/jw/2/lede/target/linux/ramips'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/jw/2/lede/target/linux'
time: target/linux/install#51.98#8.83#58.65
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/jw/2/lede'
make[1]: *** [target/Makefile:24: /home/jw/2/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/jw/2/lede'
make: *** [/home/jw/2/lede/include/toplevel.mk:231：world] 错误 2

HC5861 型号

zfdx123, 2023-04-04:
> HC5861 型号

这是极路由3你报错信息是r33是3pro的，这个错误刚刚已经提交修复了，请不要编译错设备

zfdx123, 2023-04-04:
编译测试已通过，可以更新代码，请重新编译测试，[pr](https://github.com/coolsnowwolf/lede/pull/11076/files)
@mitsukileung @wsklkl

mitsukileung, 2023-04-04:
> 编译测试已通过，可以更新代码，请重新编译测试，[pr](https://github.com/coolsnowwolf/lede/pull/11076/files) @mitsukileung @wsklkl

谢谢大佬！我编译试试
另外大佬能不能看看 YK-L1 的软重启问题？在BREED下软重启卡死，断电重启正常
尝试把打上 https://github.com/LGA1150/openwrt/commit/74516f4357d281f093f0daac01c4c5c239acc443 这个 PR，但编译出来的固件还是不能软重启

zfdx123, 2023-04-04:
YK-L1我没有相关设备，看起来只是修改了设备树，编译请确定环境是否干净，或者重新clone修改然后编译测试

mitsukileung, 2023-04-04:
> YK-L1我没有相关设备，看起来只是修改了设备树，编译请确定环境是否干净，或者重新clone修改然后编译测试

这个固件是云编译的，编译环境应该是干净的吧

zfdx123, 2023-04-04:
> > YK-L1我没有相关设备，看起来只是修改了设备树，编译请确定环境是否干净，或者重新clone修改然后编译测试
> 
> 这个固件是云编译的，编译环境应该是干净的吧

个别云编译存在缓存，如果你是云编译，请把你修改的库的地址给我，谢谢

mitsukileung, 2023-04-04:
[image]

这个应该是没有启用缓存吧？

zfdx123, 2023-04-04:
> [image]
> 
> 这个应该是没有启用缓存吧？

跟这个没关系，算了，看看代码库吧

closed 2023-04-11 by mitsukileung