coolsnowwolf/lede #11061: 最新源码 mt7621 Newifi3 编译的固件似乎有些问题，未知是否个例

aceslup, 2023-04-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
编译出的固件有以下问题：
A：有线无法自动获取IP地址，有线可以
B：无法拨号

### (2) 路由器型号和固件版本
Newifi3，mt7621
使用以下插件：
accesscontrol
webdav
arpbind
autoreboot
easymesh
filetransfer
guest-wifi
mwan3
mwan3helper
netdata
ramfree
serverchan
smartdns
rplus
syncdial
ttyd
turboacc
unblockneteasemusic-go
upnp
usb3disable
vlmcsd
watchcat
webadmin
wol
wrtbwmon(-nlbwmon)
zerotier

### (3) 详细日志
Sat Apr  1 23:07:45 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sat Apr  1 23:07:54 2023 daemon.warn pppd[10333]: Timeout waiting for PADO packets
Sat Apr  1 23:07:54 2023 daemon.err pppd[10333]: Unable to complete PPPoE Discovery
Sat Apr  1 23:07:54 2023 daemon.info pppd[10333]: Exit.
Sat Apr  1 23:07:54 2023 daemon.notice netifd: Interface 'wan' is now down
Sat Apr  1 23:07:54 2023 daemon.notice ttyd[9893]: [2023/04/01 23:07:54:2194] N: rops_handle_POLLIN_netlink: DELADDR
Sat Apr  1 23:07:54 2023 daemon.notice netifd: Interface 'wan' is disabled
Sat Apr  1 23:07:54 2023 daemon.notice netifd: Interface 'wan' is enabled
Sat Apr  1 23:07:54 2023 daemon.notice netifd: Interface 'wan' is setting up now
Sat Apr  1 23:07:54 2023 daemon.info pppd[11166]: Plugin pppoe.so loaded.
Sat Apr  1 23:07:54 2023 daemon.info pppd[11166]: PPPoE plugin from pppd 2.4.9
Sat Apr  1 23:07:54 2023 daemon.notice pppd[11166]: pppd 2.4.9 started by root, uid 0
Sat Apr  1 23:07:55 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sat Apr  1 23:08:05 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sat Apr  1 23:08:09 2023 cron.err crond[11773]: crond (busybox 1.36.0) started, log level 8
Sat Apr  1 23:08:09 2023 daemon.warn pppd[11166]: Timeout waiting for PADO packets
Sat Apr  1 23:08:09 2023 daemon.err pppd[11166]: Unable to complete PPPoE Discovery
Sat Apr  1 23:08:09 2023 daemon.info pppd[11166]: Exit.
Sat Apr  1 23:08:09 2023 daemon.notice netifd: Interface 'wan' is now down
Sat Apr  1 23:08:09 2023 daemon.notice ttyd[9893]: [2023/04/01 23:08:09:9862] N: rops_handle_POLLIN_netlink: DELADDR
Sat Apr  1 23:08:10 2023 daemon.notice netifd: Interface 'wan' is disabled
Sat Apr  1 23:08:10 2023 daemon.notice netifd: Interface 'wan' is enabled
Sat Apr  1 23:08:10 2023 daemon.notice netifd: Interface 'wan' is setting up now
Sat Apr  1 23:08:10 2023 daemon.info pppd[11930]: Plugin pppoe.so loaded.
Sat Apr  1 23:08:10 2023 daemon.info pppd[11930]: PPPoE plugin from pppd 2.4.9
Sat Apr  1 23:08:10 2023 daemon.notice pppd[11930]: pppd 2.4.9 started by root, uid 0
Sat Apr  1 23:08:12 2023 cron.err crond[12394]: crond (busybox 1.36.0) started, log level 8
Sat Apr  1 23:08:15 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sat Apr  1 23:08:25 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sat Apr  1 23:08:25 2023 daemon.warn pppd[11930]: Timeout waiting for PADO packets
Sat Apr  1 23:08:25 2023 daemon.err pppd[11930]: Unable to complete PPPoE Discovery
Sat Apr  1 23:08:25 2023 daemon.info pppd[11930]: Exit.
Sat Apr  1 23:08:25 2023 daemon.notice netifd: Interface 'wan' is now down
Sat Apr  1 23:08:25 2023 daemon.notice ttyd[9893]: [2023/04/01 23:08:25:7507] N: rops_handle_POLLIN_netlink: DELADDR
Sat Apr  1 23:08:25 2023 daemon.notice netifd: Interface 'wan' is disabled
Sat Apr  1 23:08:25 2023 daemon.notice netifd: Interface 'wan' is enabled
Sat Apr  1 23:08:25 2023 daemon.notice netifd: Interface 'wan' is setting up now
Sat Apr  1 23:08:26 2023 daemon.info pppd[12792]: Plugin pppoe.so loaded.
Sat Apr  1 23:08:26 2023 daemon.info pppd[12792]: PPPoE plugin from pppd 2.4.9
Sat Apr  1 23:08:26 2023 daemon.notice pppd[12792]: pppd 2.4.9 started by root, uid 0
Sat Apr  1 23:08:35 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sat Apr  1 23:08:41 2023 daemon.warn pppd[12792]: Timeout waiting for PADO packets
Sat Apr  1 23:08:41 2023 daemon.err pppd[12792]: Unable to complete PPPoE Discovery
Sat Apr  1 23:08:41 2023 daemon.info pppd[12792]: Exit.
Sat Apr  1 23:08:41 2023 daemon.notice netifd: Interface 'wan' is now down
Sat Apr  1 23:08:41 2023 daemon.notice ttyd[9893]: [2023/04/01 23:08:41:5573] N: rops_handle_POLLIN_netlink: DELADDR
Sat Apr  1 23:08:41 2023 daemon.notice netifd: Interface 'wan' is disabled
Sat Apr  1 23:08:41 2023 daemon.notice netifd: Interface 'wan' is enabled
Sat Apr  1 23:08:41 2023 daemon.notice netifd: Interface 'wan' is setting up now
Sat Apr  1 23:08:42 2023 daemon.info pppd[13155]: Plugin pppoe.so loaded.
Sat Apr  1 23:08:42 2023 daemon.info pppd[13155]: PPPoE plugin from pppd 2.4.9
Sat Apr  1 23:08:42 2023 daemon.info watchcat[9556]: Could not reach 218.2.135.1 for 60. Rebooting after reaching 120
Sat Apr  1 23:08:42 2023 daemon.notice pppd[13155]: pppd 2.4.9 started by root, uid 0

aceslup, 2023-04-01:
本地编译，云编译，固件都有以上问题。未知是否个例

yuzhe1129, 2023-04-02:
都是有问题吧？前天我编译的cr660x 5.10内核也是拨号不了，5.4内核的没无线

potoo0, 2023-04-02:
我也有这个问题，pppoe 拨号失败

syslog:
```
Sun Apr  2 10:29:49 2023 daemon.warn pppd[6425]: Timeout waiting for PADO packets
Sun Apr  2 10:29:49 2023 daemon.err pppd[6425]: Unable to complete PPPoE Discovery
Sun Apr  2 10:29:49 2023 daemon.info pppd[6425]: Exit.
Sun Apr  2 10:29:49 2023 daemon.notice netifd: Interface 'wan' is now down
Sun Apr  2 10:29:49 2023 daemon.notice netifd: Interface 'wan' is disabled
Sun Apr  2 10:29:49 2023 daemon.info : 09[KNL] interface eth2 deactivated
Sun Apr  2 10:29:49 2023 daemon.info : 10[KNL] interface eth2 activated
Sun Apr  2 10:29:49 2023 daemon.notice netifd: Interface 'wan' is enabled
Sun Apr  2 10:29:49 2023 daemon.notice netifd: Interface 'wan' is setting up now
Sun Apr  2 10:29:49 2023 daemon.notice netifd: Network device 'eth2' link is down
Sun Apr  2 10:29:49 2023 daemon.notice netifd: Interface 'wan' has link connectivity loss
Sun Apr  2 10:29:49 2023 daemon.notice netifd: wan (7919): uci: Entry not found
Sun Apr  2 10:29:49 2023 daemon.notice netifd: wan (7919): sh: out of range
Sun Apr  2 10:29:49 2023 daemon.info pppd[7926]: Plugin pppoe.so loaded.
Sun Apr  2 10:29:49 2023 daemon.info pppd[7926]: PPPoE plugin from pppd 2.4.9
Sun Apr  2 10:29:49 2023 daemon.notice pppd[7926]: pppd 2.4.9 started by root, uid 0
```

dmesg:
```
[    7.170824] PPP generic driver version 2.4.2
[    7.171210] PPP MPPE Compression module registered
[    7.171584] NET: Registered PF_PPPOX protocol family
[    7.172184] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[    7.172483] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[    7.173247] kmodloader: done loading kernel modules from /etc/modules.d/*
[    9.059136] vmxnet3 0000:0b:00.0 eth0: intr type 3, mode 0, 3 vectors allocated
[    9.059976] vmxnet3 0000:0b:00.0 eth0: NIC Link is Up 10000 Mbps
[    9.060403] 8021q: adding VLAN 0 to HW filter on device eth0
[    9.060766] br-lan: port 1(eth0) entered blocking state
[    9.060971] br-lan: port 1(eth0) entered disabled state
[    9.061308] device eth0 entered promiscuous mode
[    9.061804] br-lan: port 1(eth0) entered blocking state
[    9.062008] br-lan: port 1(eth0) entered forwarding state
[    9.116439] br-lan: port 2(eth1) entered blocking state
[    9.116667] br-lan: port 2(eth1) entered disabled state
[    9.116932] device eth1 entered promiscuous mode
[   10.063182] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
```

micro-stone, 2023-04-02:
昨天编译的x86，无线无法通过DHCP获取IPv4地址，有线直通正常。

aceslup, 2023-04-02:
> 都是有问题吧？前天我编译的cr660x 5.10内核也是拨号不了，5.4内核的没无线

很长时间我的固件都是刷好了，启动好，还要重启一波无线才能用，哎，加了重启网卡的定时任务才得

aceslup, 2023-04-02:
> 昨天编译的x86，无线无法通过DHCP获取IPv4地址，有线直通正常。

看来问题是普遍的

reply2future, 2023-04-02:
我也遇到一樣的問題
1. commit sha: d2a4fb1c89b36e4947ca931f23b96e7f862c6d8d
2. 紅米 ac2100
3. 關鍵日志：
```txt
Sun Apr  2 16:02:53 2023 daemon.info pppd[28592]: Plugin pppoe.so loaded.
Sun Apr  2 16:02:53 2023 daemon.info pppd[28592]: PPPoE plugin from pppd 2.4.9
Sun Apr  2 16:02:54 2023 daemon.notice pppd[28592]: pppd 2.4.9 started by root, uid 0
Sun Apr  2 16:02:54 2023 daemon.debug pppd[28592]: Send PPPOE Discovery V1T1 PADI session 0x0 length 12
Sun Apr  2 16:02:54 2023 daemon.debug pppd[28592]:  dst ff:ff:ff:ff:ff:ff  src 54:48:e6:a1:a0:18
Sun Apr  2 16:02:54 2023 daemon.debug pppd[28592]:  [service-name] [host-uniq  b0 6f 00 00]
Sun Apr  2 16:02:59 2023 daemon.debug pppd[28592]: Send PPPOE Discovery V1T1 PADI session 0x0 length 12
Sun Apr  2 16:02:59 2023 daemon.debug pppd[28592]:  dst ff:ff:ff:ff:ff:ff  src 54:48:e6:a1:a0:18
Sun Apr  2 16:02:59 2023 daemon.debug pppd[28592]:  [service-name] [host-uniq  b0 6f 00 00]
Sun Apr  2 16:03:04 2023 daemon.debug pppd[28592]: Send PPPOE Discovery V1T1 PADI session 0x0 length 12
Sun Apr  2 16:03:04 2023 daemon.debug pppd[28592]:  dst ff:ff:ff:ff:ff:ff  src 54:48:e6:a1:a0:18
Sun Apr  2 16:03:04 2023 daemon.debug pppd[28592]:  [service-name] [host-uniq  b0 6f 00 00]
Sun Apr  2 16:03:09 2023 daemon.warn pppd[28592]: Timeout waiting for PADO packets
Sun Apr  2 16:03:09 2023 daemon.err pppd[28592]: Unable to complete PPPoE Discovery
Sun Apr  2 16:03:09 2023 daemon.info pppd[28592]: Exit.
```

lwb1978, 2023-04-03:
小娱c5同样是pppoe无法拨号，pppd错误

Star-ZER0, 2023-04-04:
红米ac2100最新编译有同样问题，日志如下
`Mon Apr  3 21:37:50 2023 daemon.notice netifd: Interface 'wan' is enabled
Mon Apr  3 21:37:50 2023 daemon.notice netifd: Interface 'wan' is setting up now
Mon Apr  3 21:37:50 2023 daemon.notice netifd: wan (9349): uci: Entry not found
Mon Apr  3 21:37:50 2023 daemon.notice netifd: wan (9349): sh: out of range
Mon Apr  3 21:37:51 2023 daemon.info pppd[9379]: Plugin pppoe.so loaded.
Mon Apr  3 21:37:51 2023 daemon.info pppd[9379]: PPPoE plugin from pppd 2.4.9
Mon Apr  3 21:37:51 2023 daemon.notice pppd[9379]: pppd 2.4.9 started by root, uid 0
Mon Apr  3 21:37:55 2023 daemon.err pppd[9379]: select (waitForPADO): Interrupted system call
Mon Apr  3 21:37:55 2023 daemon.warn pppd[9379]: Timeout waiting for PADO packets
Mon Apr  3 21:37:55 2023 daemon.err pppd[9379]: Unable to complete PPPoE Discovery
Mon Apr  3 21:37:55 2023 daemon.info pppd[9379]: Exit.
Mon Apr  3 21:37:55 2023 daemon.notice netifd: Interface 'wan' is now down
Mon Apr  3 21:37:55 2023 daemon.notice ttyd[5871]: [2023/04/03 21:37:55:5511] N: rops_handle_POLLIN_netlink: DELADDR
Mon Apr  3 21:37:55 2023 daemon.notice netifd: Interface 'wan' is disabled`

lwb1978, 2023-04-04:
看来7621等修复时间就长了，ER-X eth0口不能用的问题半年多了都没修复

0769559, 2023-04-04:
> 看来7621等修复时间就长了，ER-X eth0口不能用的问题半年多了都没修复

哥们。我昨天才收了个ER-X想刷opwrt，请问你这刷了后能不能拨号？还有是仅eth0不能用吗？

reply2future, 2023-04-04:
> 红米ac2100最新编译有同样问题，日志如下 Mon Apr 3 21:37:50 2023 daemon.notice netifd: Interface 'wan' is enabled Mon Apr 3 21:37:50 2023 daemon.notice netifd: Interface 'wan' is setting up now Mon Apr 3 21:37:50 [...]

我使用了openwrt 22.03的固件就没有这个问题了，能够正常使用wifi，pppoe，自动获取ip。

lwb1978, 2023-04-04:
> > 看来7621等修复时间就长了，ER-X eth0口不能用的问题半年多了都没修复
> 
> 哥们。我昨天才收了个ER-X想刷opwrt，请问你这刷了后能不能拨号？还有是仅eth0不能用吗？

er-x我吃灰3个月了，之前因eth0不能用所以弃用了，现在估计也有无法拨号的情况，毕竟现在反馈这个无法拨号bug的都是7621用户

lwb1978, 2023-04-05:
> > 小娱c5同样是pppoe无法拨号，pppd错误
> 
> 我的 pppoe 错误解决了，其实是接口顺序标错误了，我的三个接口，两个直通给 openwrt，以前一直用的最外边的 eth2，然而这次版本编译后在 openwrt 里给我标成了 eth1，看起来是把 eth1 和 eth2 搞反了

你的是x86的机子，跟7621不是一回事，7621的机子无法拨号是pppd报错，看日志