coolsnowwolf/lede #11074: NanoPi R2C 无法使用WAN口

Gabrielxzx, 2023-04-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue
* 存在相关的issue，但已经是大约1年前的issue，已经不适用了

## 2. 详细叙述

### (1) 具体问题

A：NanoPi R2C 无法使用WAN口，WAN口已经接入光猫或者上级路由，但是openwrt无法连接使用，且网口指示灯LED无法点亮。
编译采用的P3TERX的云编译方案，下面附件是config文件。
[NanoPi-R2C.config.txt](https://github.com/coolsnowwolf/lede/files/11138743/NanoPi-R2C.config.txt)

之前大概见过类似的issue，是OrangePi R1 Plus LTS提出的issue，大概也是YT的网卡无法驱动。
https://github.com/coolsnowwolf/lede/issues/10980

### (2) 路由器型号和固件版本

A：FriendlyElec NanoPi R2C，内核版本为5.15.104，R23.4.1

### (3) 详细日志

A：详细的日志见附件。

[系统日志.txt](https://github.com/coolsnowwolf/lede/files/11138716/default.txt)

[内核日志.txt](https://github.com/coolsnowwolf/lede/files/11138723/default.txt)

XiaoBinin, 2023-04-04:
参考一下天灵大佬的patch把这三行加上应该就可以使用了（香橙派lts也是一样）
motorcomm,clk-out-frequency-hz = <125000000>;
motorcomm,keep-pll-enabled;
motorcomm,auto-sleep-disabled;
https://github.com/immortalwrt/immortalwrt/blob/1b080245fb1ed056a531022bfc2bb86165037ae7/package/boot/uboot-rockchip/src/arch/arm/dts/rk3328-nanopi-r2c.dts

XiaoBinin, 2023-04-06:
> > 参考一下天灵大佬的patch把这三行加上应该就可以使用了（香橙派lts也是一样） motorcomm,clk-out-frequency-hz = <125000000>; motorcomm,keep-pll-enabled; motorcomm,auto-sleep-disabled;  [https://github.com/immortalwrt/immortalwrt/blob/1b080245fb1ed056a531022bfc2bb86165037ae7/package/boot/uboot-rockchip/src/arch/arm/dts/rk3328-nanopi-r2c.dts ](https://github.com/immortalwrt/immortalwrt/blob/1b080245fb1ed056a531022bfc2bb86165037ae7/package/boot/uboot-rockchip/src/arch/arm/dts/rk3328-nanopi-r2c.dts)
> 
> 你好，问下是在什么文件里修改？Lean的仓库里好像没有rk3328-nanopi-r2c.dts这个文件

 当前默认的是5.15内核，你就改这个文件
lede/target/linux/rockchip/patches-5.15/204-rockchip-rk3328-Add-support-for-FriendlyARM-NanoPi-R.patch

Gabrielxzx, 2023-04-07:
> 当前默认的是5.15内核，你就改这个文件 lede/target/linux/rockchip/patches-5.15/204-rockchip-rk3328-Add-support-for-FriendlyARM-NanoPi-R.patch

大佬牛逼，修改了patch文件后可以用了，测试了下可以使用了，就是WAN口的指示灯好像不太正常，只有一个灯在亮了，修改的patch文件已经附上了，不知道是不是改的不对。
[204-rockchip-rk3328-Add-support-for-FriendlyARM-NanoPi-R.patch](https://github.com/coolsnowwolf/lede/files/11180229/204-rockchip-rk3328-Add-support-for-FriendlyARM-NanoPi-R.patch)

XiaoBinin, 2023-04-07:
> > > > 参考一下天灵大佬的patch把这三行加上应该就可以使用了（香橙派lts也是一样） motorcomm,clk-out-frequency-hz = <125000000>; motorcomm,keep-pll-enabled; motorcomm,auto-sleep-disabled;  [https://github.com/immortalwrt/immortalwrt/blob/1b080245fb1ed056a531022bfc2bb86165037ae7/package/boot/uboot-rockchip/src/arch/arm/dts/rk3328-nanopi-r2c.dts ](https://github.com/immortalwrt/immortalwrt/blob/1b080245fb1ed056a531022bfc2bb86165037ae7/package/boot/uboot-rockchip/src/arch/arm/dts/rk3328-nanopi-r2c.dts)
> > > 
> > > 
> > > 你好，问下是在什么文件里修改？Lean的仓库里好像没有rk3328-nanopi-r2c.dts这个文件
> > 
> > 
> > 当前默认的是5.15内核，你就改这个文件 lede/target/linux/rockchip/patches-5.15/204-rockchip-rk3328-Add-support-for-FriendlyARM-NanoPi-R.patch
> 
> 大佬牛逼，修改了patch文件后可以用了，测试了下可以使用了，就是WAN口的指示灯好像不太正常，只有一个灯在闪烁了，修改的patch文件已经附上了，不知道是不是改的不对。 [204-rockchip-rk3328-Add-support-for-FriendlyARM-NanoPi-R.patch](https://github.com/coolsnowwolf/lede/files/11180229/204-rockchip-rk3328-Add-support-for-FriendlyARM-NanoPi-R.patch)

指示灯不正常是因为驱动的问题，这里已经解决了。

XiaoBinin, 2023-04-08:
> 大佬牛逼，修改了patch文件后可以用了，测试了下可以使用了，就是WAN口的指示灯好像不太正常，只有一个灯在闪烁了，修改的patch文件已经附上了，不知道是不是改的不对。 [...]

正常已经可以用了，如果还是想添加灯的支持可以参考的自己加的patch
https://github.com/XiaoBinin/Actions-lede/blob/main/patchs/lede/target/9998-add-missing-parameters-for-LTS.patch
279~309行
drivers/net/phy/motorcomm.c

commit 0d498429 (2023-04-08): rockchip: fix wan port for nanopi r2c (#11091)

Fixes: #11074
closed 2023-04-08 by aiamadeus