coolsnowwolf/lede #11075: 为什么勾选了NodeJS Version就出错？

yuzhe1129, 2023-04-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
编译选网易云解锁这个插件
单单选Golang就能编译成功，如果勾选上NodeJS Version就是百分百出错，这是什么情况？以前编译时候都还是可以的，为什么现在都出错了，
### (2) 路由器型号和固件版本

A：

### (3) 详细日志
time: package/kernel/cryptodev-linux/compile#0.18#0.10#0.31
make[3]: Entering directory '/home/cc/lede/feeds/luci/libs/rpcd-mod-rrdns'
mkdir -p /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/cc/lede/tmp/.root-copy.flock -c 'cp -fpR /home/cc/lede/build_dir/target-mipsel_24kc_musl/dnsmasq-full/dnsmasq-2.88/.pkgdir/dnsmasq-full/. /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
mkdir -p /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
touch /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.dnsmasq-full_installed
echo '12' | cmp -s - /home/cc/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/libxtables.version || { echo '12' > /home/cc/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/libxtables.version;  } 
SHELL= flock /home/cc/lede/tmp/.root-copy.flock -c 'cp -fpR /home/cc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/iptables-1.8.7/.pkgdir/libxtables/. /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.libxtables_installed
echo "dnsmasq-full" >> /home/cc/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/dnsmasq.full.install
make[3]: Leaving directory '/home/cc/lede/package/network/services/dnsmasq'
time: package/network/services/dnsmasq/full/compile#0.18#0.12#0.32
echo "libxtables" >> /home/cc/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/iptables.default.install
make[3]: Entering directory '/home/cc/lede/feeds/luci/applications/luci-app-turboacc'
mkdir -p /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
echo '2' | cmp -s - /home/cc/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/libip4tc.version || { echo '2' > /home/cc/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/libip4tc.version;  } 
SHELL= flock /home/cc/lede/tmp/.root-copy.flock -c 'cp -fpR /home/cc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/iptables-1.8.7/.pkgdir/libip4tc/. /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.libip4tc_installed
echo "libip4tc" >> /home/cc/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/iptables.default.install
mkdir -p /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
echo '2' | cmp -s - /home/cc/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/libip6tc.version || { echo '2' > /home/cc/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/libip6tc.version;  } 
SHELL= flock /home/cc/lede/tmp/.root-copy.flock -c 'cp -fpR /home/cc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/iptables-1.8.7/.pkgdir/libip6tc/. /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.libip6tc_installed
echo "libip6tc" >> /home/cc/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/iptables.default.install
mkdir -p /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
  mipsel-openwrt-linux-musl-gcc -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -o /home/cc/lede/build_dir/target-mipsel_24kc_musl/node-v12.22.1/out/Release/obj.target/openssl/deps/openssl/openssl/ssl/bio_ssl.o ../deps/openssl/openssl/ssl/bio_ssl.c '-DV8_DEPRECATION_WARNINGS' '-DV8_IMMINENT_DEPRECATION_WARNINGS' '-D__STDC_FORMAT_MACROS' '-DOPENSSL_NO_PINSHARED' '-DOPENSSL_THREADS' '-DOPENSSL_NO_HW' '-DNDEBUG' '-DOPENSSL_USE_NODELETE' '-DL_ENDIAN' '-DOPENSSL_PIC' '-DOPENSSL_CPUID_OBJ' '-DOPENSSL_IA32_SSE2' '-DOPENSSL_BN_ASM_MONT' '-DOPENSSL_BN_ASM_MONT5' '-DOPENSSL_BN_ASM_GF2m' '-DSHA1_ASM' '-DSHA256_ASM' '-DSHA512_ASM' '-DKECCAK1600_ASM' '-DRC4_ASM' '-DMD5_ASM' '-DAESNI_ASM' '-DVPAES_ASM' '-DGHASH_ASM' '-DECP_NISTZ256_ASM' '-DX25519_ASM' '-DPOLY1305_ASM' '-DOPENSSLDIR="/etc/ssl"' '-DENGINESDIR="/dev/null"' '-DTERMIOS' -I/home/cc/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I../deps/openssl/openssl -I../deps/openssl/openssl/include -I../deps/openssl/openssl/crypto -I../deps/openssl/openssl/crypto/include -I../deps/openssl/openssl/crypto/modes -I../deps/openssl/openssl/crypto/ec/curve448 -I../deps/openssl/openssl/crypto/ec/curve448/arch_32 -I../deps/openssl/config -I../deps/openssl/config/archs/linux-x86_64/asm -I../deps/openssl/config/archs/linux-x86_64/asm/include -I../deps/openssl/config/archs/linux-x86_64/asm/crypto -I../deps/openssl/config/archs/linux-x86_64/asm/crypto/include/internal  -pthread -Wall -Wextra -Wno-unused-parameter -Wa,--noexecstack -Wall -O3 -pthread -m64 -Wall -O3 -Wno-missing-field-initializers -Wno-old-style-declaration -O3 -fno-omit-frame-pointer  -MMD -MF /home/cc/lede/build_dir/target-mipsel_24kc_musl/node-v12.22.1/out/Release/.deps//home/cc/lede/build_dir/target-mipsel_24kc_musl/node-v12.22.1/out/Release/obj.target/openssl/deps/openssl/openssl/ssl/bio_ssl.o.d.raw  -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/home/cc/lede/build_dir/target-mipsel_24kc_musl/node-v12.22.1=node-v12.22.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/cc/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/home/cc/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/home/cc/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include  -c
SHELL= flock /home/cc/lede/tmp/.root-copy.flock -c 'cp -fpR /home/cc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/iptables-1.8.7/.pkgdir/iptables/. /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.iptables_installed
echo "iptables" >> /home/cc/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/iptables.default.install
mipsel-openwrt-linux-musl-gcc: error: unrecognized command line option '-m64'
make[5]: *** [deps/openssl/openssl.target.mk:867: /home/cc/lede/build_dir/target-mipsel_24kc_musl/node-v12.22.1/out/Release/obj.target/openssl/deps/openssl/openssl/ssl/bio_ssl.o] Error 1
make[4]: *** [Makefile:104: node] Error 2
mkdir -p /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
make[4]: Leaving directory '/home/cc/lede/build_dir/target-mipsel_24kc_musl/node-v12.22.1'
make[3]: *** [Makefile:235: /home/cc/lede/build_dir/target-mipsel_24kc_musl/node-v12.22.1/.built] Error 2
make[3]: Leaving directory '/home/cc/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#0.46#0.63#4.41
SHELL= flock /home/cc/lede/tmp/.root-copy.flock -c 'cp -fpR /home/cc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/iptables-1.8.7/.pkgdir/iptables-mod-extra/. /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
    ERROR: package/feeds/packages/node failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/node/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
touch /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.iptables-mod-extra_installed
echo "iptables-mod-extra" >> /home/cc/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/iptables.default.install
mkdir -p /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/cc/lede/tmp/.root-copy.flock -c 'cp -fpR /home/cc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/iptables-1.8.7/.pkgdir/iptables-mod-tproxy/. /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.iptables-mod-tproxy_installed
echo "iptables-mod-tproxy" >> /home/cc/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/iptables.default.install
make[3]: Leaving directory '/home/cc/lede/package/network/utils/iptables'
time: package/network/utils/iptables/compile#0.23#0.26#0.53
mkdir -p /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/cc/lede/tmp/.root-copy.flock -c 'cp -fpR /home/cc/lede/build_dir/target-mipsel_24kc_musl/rpcd-mod-rrdns-20170710/.pkgdir/rpcd-mod-rrdns/. /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/cc/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.rpcd-mod-rrdns_installed
[... cut ...]
echo "luci-i18n-turboacc-zh-cn" >> /home/cc/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-app-turboacc.default.install
make[3]: Leaving directory '/home/cc/lede/feeds/luci/applications/luci-app-turboacc'
time: package/feeds/luci/luci-app-turboacc/compile#0.26#0.24#0.52
make[2]: Leaving directory '/home/cc/lede'
make[1]: *** [package/Makefile:110: /home/cc/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/cc/lede'
make: *** [/home/cc/lede/include/toplevel.mk:231：world] 错误 2

mitsukileung, 2023-04-03:
超时吧，github 编译超过6小时就会强制停止
NodeJS编译时间太久了很容易超时

hyy-666, 2023-04-03:
几个线程编译？大概率是内存不够，现在编译node要很多内存

hzjnet, 2023-04-03:
mipsel-openwrt-linux-musl-gcc: error: unrecognized command line option '-m64'   明显是gcc版本啊，先设置成11版本试试吧

yuzhe1129, 2023-04-03:
> mipsel-openwrt-linux-musl-gcc: error: unrecognized command line option '-m64' 明显是gcc版本啊，先设置成11版本试试吧

11版本是什么意思？

WYC-2020, 2023-04-04:
可真牛，mipsel这玩意还想配套node 确定即使编译过了能打包成功？这种小容量的就别想了

Cathgao, 2023-04-04:
MIPS就别凑nodeJS的热闹了，老实用仅golang

felixxiao99, 2023-04-04:
@coolsnowwolf lean大大晚上好，我之前提了一个issue，没看到有任何回复就连续被close了3次，但问题还是存在，能否帮我看下是什么原因，感谢！

下面是之前issue的部分内容：
1) 具体问题

A：最新编译的R2S固件作为旁路由使用，使用ShadowSocksR Plus+不能科学上网，只能访问国内网站
年前编译的固件在相同旁路由设置下是可以科学上网的

另外：个人感觉问题应该是出在ShadowSocksR Plus+插件上，除了自编译固件有这个问题，我也下载了骷髅头大神的R2S固件，早期的版本(1月17号)设置旁路由能科学上网，用最新的固件(3月23号 )也不能科学上网

(2) 路由器型号和固件版本

A：
路由器型号：NanoPi R2S
固件版本：OpenWrt R23.3.3 Compiled by xiaofei / LuCI Master (git-23.082.27304-adda8b2)
内核版本：5.15.104

yuzhe1129, 2023-04-04:
> 可真牛，mipsel这个玩意还想配套node 确定即使编译过能打包成功吗？这种小容量的就别想了

试过啊，23.3.3以前都是可以成功的啊

felixxiao99, 2023-04-04:
> 重新加入了官方所不支持的 nodejs 的 MIPS 支持。 我发誓真的是最后一次了。。。。

@coolsnowwolf lean大大晚上好，我之前提了一个issue，没看到有任何回复就连续被close了3次，但问题还是存在，能否帮我看下是什么原因，感谢！

下面是之前issue的部分内容：
1) 具体问题

A：最新编译的R2S固件作为旁路由使用，使用ShadowSocksR Plus+不能科学上网，只能访问国内网站
年前编译的固件在相同旁路由设置下是可以科学上网的

另外：个人感觉问题应该是出在ShadowSocksR Plus+插件上，除了自编译固件有这个问题，我也下载了骷髅头大神的R2S固件，早期的版本(1月17号)设置旁路由能科学上网，用最新的固件(3月23号 )也不能科学上网

(2) 路由器型号和固件版本

A：
路由器型号：NanoPi R2S
固件版本：OpenWrt R23.3.3 Compiled by xiaofei / LuCI Master (git-23.082.27304-adda8b2)
内核版本：5.15.104

WYC-2020, 2023-04-04:
> > 可真牛，mipsel这个玩意还想配套node 确定即使编译过能打包成功吗？这种小容量的就别想了
> 
> 试过啊，23.3.3以前都是可以成功的啊

肯定被改了默认编译的node版本呗，没匹配好，mips还编译node真是空间多哈哈