coolsnowwolf/lede #11080: 网口标反

potoo0, 2023-04-04:
设备信息：
- cpu: N5105
- 网卡: I225-V
- 虚拟: ESXi-7.0U3 网卡硬件直通

实际网卡顺序是对应物理口从左到右为 eth0, eth1, eth2, 对应 mac 是 ad:44, ad:45, ad:46。
但是新固件编译后实际表现把 eth1 和 eth2 搞反了，所以我前天用了后 pppoe 拨号失败，刚刚看 mac 地址确认搞反后 wan 口调换后就正常使用了。

openwrt `ifconfig`:
```
eth1      Link encap:Ethernet  HWaddr ........AD:46

eth2      Link encap:Ethernet  HWaddr ........AD:45
```

esxi `esxcfg-nics -l`:
```
0000:01:00.0 ... ...ad:44    #eth1
0000:02:00.0 ... ...ad:45    #eth2
0000:03:00.0 ... ...ad:46    #eth3
```

potoo0, 2023-04-04:
issue 里一大堆 pppoe 无法拨号的，如果是多网口的话那有可能就是这个问题导致的

lwb1978, 2023-04-05:
> issue 里一大堆 pppoe 无法拨号的，如果是多网口的话那有可能就是这个问题导致的

不是一回事，7621的机子无法拨号是pppd报错

xiechangan123, 2023-04-05:
openwrt老祖宗就是eth1 和 eth2 对调的，最好别自己改网口顺序。下次格式化新装又变回来了，不利于远程在线升级。。。。

potoo0, 2023-04-05:
> openwrt老祖宗就是eth1 和 eth2 对调的，最好别自己改网口顺序。下次格式化新装又变回来了，不利于远程在线升级。。。。

我 2021 年末和 2022 年中编译的版本都没有反，时隔十个月的现在的版本反了。

closed 2023-04-05 by potoo0

WYC-2020, 2023-04-05:
> > openwrt老祖宗就是eth1 和 eth2 对调的，最好别自己改网口顺序。下次格式化新装又变回来了，不利于远程在线升级。。。。
> 
> 我 2021 年末和 2022 年中编译的版本都没有反，时隔十个月的现在的版本反了。

这问题我也留意到了，我vmare 虚拟机 配置了四个网卡，并且他们mac 是顺序递增的，我测试刷入后 直接eth0 和eth1 的mac 反了 eth2 3倒是遵循顺序的，我怀疑是那个patch 导致的，这个我抽空研究一下，我i以前刷入 也是记得 eth的mac 地址也是递增的