coolsnowwolf/lede #11082: R7800，外网IP无法访问内网端口，已设置端口转发

sunyink, 2023-04-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
公网ipv4，外网访问端口都是“目前无法处理此请求 502”，内网IP正常，更别说ddns了。

rt，尝试查询自排查，端口转发已设置并重启，DCHP中重定向和仅本地服务都非勾选，makeconfig也没有docker。按以前官版经验应该直接端口转发就行了，这回情况实在琢磨不出。

另外体感上似乎这个端口转发没有生效0.0？虽clash日志说发现这个生效规则...自动绕过啥的，但用站长之家测试端口连通度，BT客户端需要用upen开启的端口才能 查出“开通”，webui端口直接挂了。

### (2) 路由器型号和固件版本

A：
Netgear Nighthawk X4S R7800 (CpuMark : 13650.394776 Scores)
ARMv7 Processor rev 0 (v7l) x 2
固件版本 | OpenWrt R23.4.1 / LuCI Master (git-23.090.53841-1cae4d1)
内核版本 | 5.4.238

### (3) 详细日志

A：
etc/config：
DHCP:
`
config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option nonwildcard '1'
	option ednspacket_max '1232'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option cachesize '8192'
	option sequential_ip '1'
	option filter_aaaa '1'
	option rebind_protection '0'
	option localservice '0'
	option noresolv '0'
	option port '0'

config dhcp 'lan'
	option interface 'lan'
	option leasetime '12h'
	option start '121'
	option limit '79'
	list dhcp_option '6,192.168.1.1'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config srvhost
	option srv '_vlmcs._tcp'
	option target 'OpenWrt'
	option port '1688'
	option class '0'
	option weight '100'

config host
	option name 'Ubunt'
	option dns '1'
	option mac '98:fa:9b:01:1a:00'
	option ip '192.168.1.108'
	option leasetime 'infinite'

config host
	option name 'PC-106'
	option dns '1'
	option mac '1c:1b:0d:85:5e:00'
	option ip '192.168.1.106'
	option leasetime 'infinite'
`
firewall：
`
config defaults
	option syn_flood '1'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option fullcone '2'
	option flow_offloading_hw '0'
	option flow_offloading '0'
	option forward 'ACCEPT'

config zone
	option name 'lan'
	list network 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'

config zone
	option name 'wan'
	list network 'wan'
	list network 'wan6'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'
	option input 'ACCEPT'

config forwarding
	option src 'lan'
	option dest 'wan'

config rule
	option name 'Allow-DHCP-Renew'
	option src 'wan'
	option proto 'udp'
	option dest_port '68'
	option target 'ACCEPT'
	option family 'ipv4'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option icmp_type 'echo-request'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-IGMP'
	option src 'wan'
	option proto 'igmp'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-DHCPv6'
	option src 'wan'
	option proto 'udp'
	option src_ip 'fc00::/6'
	option dest_ip 'fc00::/6'
	option dest_port '546'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-MLD'
	option src 'wan'
	option proto 'icmp'
	option src_ip 'fe80::/10'
	list icmp_type '130/0'
	list icmp_type '131/0'
	list icmp_type '132/0'
	list icmp_type '143/0'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Input'
	option src 'wan'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	list icmp_type 'router-solicitation'
	list icmp_type 'neighbour-solicitation'
	list icmp_type 'router-advertisement'
	list icmp_type 'neighbour-advertisement'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Forward'
	option src 'wan'
	option dest '*'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-IPSec-ESP'
	option src 'wan'
	option dest 'lan'
	option proto 'esp'
	option target 'ACCEPT'

config rule
	option name 'Allow-ISAKMP'
	option src 'wan'
	option dest 'lan'
	option dest_port '500'
	option proto 'udp'
	option target 'ACCEPT'

config rule
	option name 'Support-UDP-Traceroute'
	option src 'wan'
	option dest_port '33434:33689'
	option proto 'udp'
	option family 'ipv4'
	option target 'REJECT'
	option enabled 'false'

config include
	option path '/etc/firewall.user'

config include 'zerotier'
	option type 'script'
	option path '/etc/zerotier.start'
	option reload '1'

config include 'miniupnpd'
	option type 'script'
	option path '/usr/share/miniupnpd/firewall.include'
	option family 'any'
	option reload '1'

config include 'qcanssecm'
	option type 'script'
	option path '/etc/firewall.d/qca-nss-ecm'
	option family 'any'
	option reload '1'

config include 'mia'
	option type 'script'
	option path '/etc/mia.include'
	option reload '1'

config include 'openclash'
	option type 'script'
	option path '/var/etc/openclash.include'
	option reload '1'

config include 'unblockmusic'
	option type 'script'
	option path '/var/etc/unblockmusic.include'
	option reload '1'

config redirect
	option target 'DNAT'
	option src 'wan'
	option dest 'lan'
	option proto 'tcp udp'
	option src_dport '10680-10699'
	option dest_ip '192.168.1.106'
	option dest_port '10680-10699'
	option name '106QBT'

config redirect
	option target 'DNAT'
	option src 'wan'
	option dest 'lan'
	option proto 'tcp udp'
	option src_dport '10693'
	option dest_ip '192.168.1.106'
	option dest_port '10693'
	option name '10693'

`
其他设置资料：
[image]
[image]
[image]

https://user-images.githubusercontent.com/37411580/230272876-3d446d14-5e00-4abd-8312-7faa4fdca40a.mp4

[image]
[image]
[image]

还漏了什么请告诉我。最后，谢谢。

alexyw, 2023-04-17:
把TurboACC的全锥形NAT关掉试试

sunyink, 2023-04-24:
> 外网访问 lede 是 IPv6？还是 IPv4，有相似的问题

ipv4的，很离奇。

sunyink, 2023-04-24:
我暂时换成官方源了，然后官方源转发好像有问题，500M下行经常卡成只有300M。

coolsnowwolf, 2026-09-05:
该问题基于 R23.4.1/内核 5.4 的旧 firewall3 配置，规则中还包含全锥形 NAT、Clash 等外部变量；没有提供当前源码、`ubus call firewall status`、`nft list ruleset`/连接跟踪或实际端口测试结果，无法证明是 LEDE 端口转发代码缺陷。当前无可验证代码修复，关闭。

closed 2026-09-05 by coolsnowwolf