coolsnowwolf/lede #11083: 最新版编译报错：nf_flow_table_hw.ko is missing

ricksuzade-maker, 2023-04-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
在编译x86-64版时发生错误，提示缺少“nf_flow_table_hw.ko”
### (1) 具体问题

A：
在编译x86-64版时发生错误，提示缺少“nf_flow_table_hw.ko”
### (2) 路由器型号和固件版本

A：x86-64 esxi虚拟机

### (3) 详细日志

A：
ake[3]: Leaving directory '/home/zhiliang/scripts/lede/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#1.35#0.40#1.75
make[3]: Entering directory '/home/zhiliang/scripts/lede/package/kernel/linux'
mkdir -p /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/zhiliang/scripts/lede/tmp/.root-copy.flock -c 'cp -fpR /home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
ERROR: module '/home/zhiliang/scripts/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.104/net/netfilter/nf_flow_table_hw.ko' is missing.
make[3]: *** [modules/netfilter.mk:189: /home/zhiliang/scripts/lede/bin/targets/x86/64/packages/kmod-nf-flow_5.15.104-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/zhiliang/scripts/lede/package/kernel/linux'
time: package/kernel/linux/compile#2.20#0.29#2.48
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/zhiliang/scripts/lede'
make[1]: *** [package/Makefile:110: /home/zhiliang/scripts/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhiliang/scripts/lede'
make: *** [/home/zhiliang/scripts/lede/include/toplevel.mk:231: world] Error 2

GreatMichaelLee, 2023-04-06:
同样的问题+1, 另外再报一个问题：花生壳插件phtunnel 在上游官方源码中好像被删除了，导致luci-app-phtunnel下载包编译通不过。

wuzhuang123, 2023-04-06:
重新拉重新编译，前几天更新到5.10报了一次

VIKINGYFY, 2023-04-06:
> 重新拉重新编译，前几天更新到5.10报了一次

我云编译也是报这个错，云编译应该每次都是全新环境吧？

ricksuzade-maker, 2023-04-06:
> 重新拉重新编译，前几天更新到5.10报了一次

这个我当然已经试过了，全部删除重新拉下来并且单线程编译，确认了有这个问题才来报issue的，首先要态度端正对吧，哈哈。

wuzhuang123, 2023-04-06:
> > 重新拉重新编译，前几天更新到5.10报了一次
> 
> 这个我当然已经试过了，全部删除重新拉下来并且单线程编译，确认了有这个问题才来报issue的，首先要态度端正对吧，哈哈。

这个东西很奇怪，从5.4.235升5.4.237报了一次，重新拉又好了，升5.10报一次，由重新拉也好了，这次回退就直接重新拉5.4.238，也不知道是哪个插件的依赖。反正我现在最新的可以

mahakalaya, 2023-04-06:
git reset 一下上游重新编译通过

wuqingxin1987, 2023-04-06:
破案了解决方法修改/lede/package/kernel/linux/modules/里netfilter.mk文件到lede 网站找对应文件替换掉原文就ok！猜测是l大测试MT798x QWRT 没改回来配置导致！

akihara-sam, 2023-04-07:
一样 也是编译报错，本地编译，云编译  一起报错 跟楼主相同错误

WYC-2020, 2023-04-07:
被强制覆盖了两个提交，有一个提交就是修复这玩意的
[image]

WYC-2020, 2023-04-07:
看了下 改成这样赶紧更加合理 
[image]
x86 5.4 5.10 我都测过没问题，按理5.15 和6.1 都一样懒得测试了

VIKINGYFY, 2023-04-07:
> 看了下 改成这样赶紧更加合理 [image] x86 5.4 5.10 我都测过没问题，按理5.15 和6.1 都一样懒得测试了

不用改了，重新拉源码就行了，已经修复了

WYC-2020, 2023-04-07:
> > 看了下 改成这样赶紧更加合理 [image] x86 5.4 5.10 我都测过没问题，按理5.15 和6.1 都一样懒得测试了
> 
> 不用改了，重新拉源码就行了，已经修复了

嗯 l用的CONFIG_LINUX_5_4 来限制的，我用@eq来限制

coldrook, 2023-04-07:
好像依然未修复，仍然出现nf_flow_table_ipv4.ko' is missing的提示

WYC-2020, 2023-04-08:
> 好像依然未修复，6.1.22编译，仍然出现nf_flow_table_ipv4.ko' is missing的提示

重新m配置过没？用下我那个方法试下呢，l那个方法应该也可以才对，再不行后续有时间我再看看6.1

coldrook, 2023-04-09:
> > 好像依然未修复，6.1.22编译，仍然出现nf_flow_table_ipv4.ko' is missing的提示
> 
> 重新m配置过没？用下我那个方法试下呢，l那个方法应该也可以才对，再不行后续有时间我再看看6.1

每次都是清理cache重新编译的。好的，再按你的方法再试试。

VIKINGYFY, 2023-04-09:
不清楚你们本地编译的，我云编译是没问题了

[image]

closed 2023-04-11 by aiamadeus

Y-ZD-heheka, 2023-04-24:
[image]
又出现了了

admin2012-1, 2023-05-03:
> 好像依然未修复，6.1.22编译，仍然出现nf_flow_table_ipv4.ko' is missing的提示

您问题解决了吗？，我在编译6.1的时候依然出现这个错误。