coolsnowwolf/lede #11084: pgyvpn plugin download 404

GreatMichaelLee, 2023-04-06:
梯子没问题，404，上游没有了。

make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/lang/lua-md5'
time: package/feeds/packages/lua-md5/compile#0.10#0.06#0.15
make[3]: Entering directory '/home/builder/lede_x86/feeds/packages/net/pgyvpn'
mkdir -p /home/builder/lede_x86/dl
SHELL= flock /home/builder/lede_x86/tmp/.pgyvpn_oraysl.flock -c '       /home/builder/lede_x86/scripts/download.pl "/home/builder/lede_x86/dl" "pgyvpn_oraysl" "skip" "" "https://mirrors.oray.com/orayos/packages/pgyvpn/x86_64-musl-unknown/3.1.0/bin"    '
+ curl -f --connect-timeout 20 --retry 5 --location https://mirrors.oray.com/orayos/packages/pgyvpn/x86_64-musl-unknown/3.1.0/bin/pgyvpn_oraysl
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (35) error:0A00010B:SSL routines::wrong version number
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/pgyvpn_oraysl
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   169    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.openwrt.org/pgyvpn_oraysl
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   169    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/pgyvpn_oraysl
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   153    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:64: /home/builder/lede_x86/dl/pgyvpn_oraysl] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/net/pgyvpn'
time: package/feeds/packages/pgyvpn/compile#0.34#0.09#3.59
    ERROR: package/feeds/packages/pgyvpn failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/pgyvpn/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231: world] Error 2
builder@Supernode:~/lede_x86$

GreatMichaelLee, 2023-04-06:
麻烦修复一下，除非我去掉这两个插件，否则编译不过，make download也提示这两个插件有问题

make[2] -C feeds/packages/lang/node download
 make[2] -C feeds/packages/net/nps download
 make[2] -C feeds/packages/net/ntpclient download
 make[2] -C feeds/packages/libs/oniguruma download
 make[2] -C feeds/packages/net/openssh download
 make[2] -C feeds/packages/net/openvpn download
 make[2] -C feeds/packages/net/openvpn-easy-rsa download
 make[2] -C feeds/packages/net/p910nd download
 make[2] -C feeds/packages/utils/parted download
 make[2] -C feeds/packages/utils/pciutils download
 make[2] -C feeds/packages/libs/pcre2 download
 make[2] -C feeds/packages/net/pdnsd-alt download
 make[2] -C feeds/packages/net/pgyvpn download
 make[2] -C feeds/packages/lang/php8 download
 make[2] -C feeds/packages/net/phtunnel download
 make[2] -C feeds/packages/net/pptpd download
 make[2] -C feeds/packages/net/qBittorrent-static download
 make[2] -C feeds/packages/utils/qemu download
 make[2] -C feeds/packages/net/rp-pppoe download
 make[2] -C feeds/packages/net/rpcbind download
 make[2] -C feeds/packages/utils/rpcd-mod-lxc download
 make[2] -C feeds/packages/utils/rrdtool1 download
 make[2] -C feeds/packages/lang/ruby download
 make[2] -C feeds/packages/utils/runc download
 make[2] -C feeds/packages/net/samba4 download
 make[2] -C feeds/packages/net/shadowsocks-libev download
 make[2] -C feeds/packages/net/smartdns download
    ERROR: package/feeds/packages/pgyvpn failed to build.
    ERROR: package/feeds/packages/phtunnel failed to build.
make package/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
 make[1] target/download
 make[2] -C target/linux download

aiamadeus, 2023-04-06:
你网络问题吧，https://mirrors.oray.com/orayos/packages/pgyvpn/x86_64-musl-unknown/3.1.0/bin/pgyvpn_oraysl 是打得开的。

GreatMichaelLee, 2023-04-06:
version mismatch. ok?

GreatMichaelLee, 2023-04-06:
问题找到了，是我这个build服务器上配置了xray client, xray的reality好像有些问题。。。。把mirros.oray.com放到不用代理就暂时workaround了。

closed 2023-04-06 by aiamadeus