coolsnowwolf/lede #11085: 开启openclash后网络ipv4显示无网络访问权限

hatbhtbh, 2023-04-06:
开启openclash的fake-ip模式后网络ipv4显示无网络访问权限，电脑右下角网络图标是感叹号，上外网正常，国内网络卡顿。
切换默认的redir-host兼容模式后网络ipv4显示正常接入网络，外网和国内网使用正常。

openclash的日志我看了没有关于这快问题的描述，不知道什么原因？

### (2) 路由器型号和固件版本：j4125  x86 64  固件版本今天本地编译的OpenWrt R23.4.1 / LuCI Master (git-23.095.26899-77f05f4)
内核：5.15.104
Ubuntu下用Lean源码编译openwrt
使用这个地址编译的：https://github.com/coolsnowwolf/lede/blob/master/README_EN.md

hatbhtbh, 2023-04-07:
> 开启openclash的fake-ip增强模式后网络ipv4显示无网络访问权限，电脑右下角网络图标是感叹号，上外网正常，国内网络卡顿。 切换默认的redir-host兼容模式后网络ipv4显示正常接入网络，网络图标感叹号消失，外网和国内网使用正常。
> 
> openclash的日志我看了没有关于这快问题的描述，不知道什么原因？
> 
> ### (2) 路由器型号和固件版本：j4125 x86 64 固件版本今天本地编译的OpenWrt R23.4.1 / LuCI Master (git-23.095.26899-77f05f4)
> 内核：5.15.104 Ubuntu下用Lean源码编译openwrt 使用这个地址编译的：https://github.com/coolsnowwolf/lede/blob/master/README_EN.md

closed 2023-04-07 by aiamadeus

hatbhtbh, 2023-04-07:
今天又重新本地编译了一下，在重新安装新固件，刷机后问题解决。

jikkyfu, 2023-04-28:
不知道什么问题，我是通过把网卡的dns清空，clash或者smartdns之类的转发就正常了。