coolsnowwolf/lede #11086: X86编译报错

wzh4809, 2023-04-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：原版编译也是这个提示

### (2) 路由器型号和固件版本

A：debian11

### (3) 详细日志

A：TARGET_OBJ2ELF= sh genmod.sh moddep.lst div_test.module build-grub-module-verifier div_test.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst mul_test.module build-grub-module-verifier mul_test.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst shift_test.module build-grub-module-verifier shift_test.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst cmp_test.module build-grub-module-verifier cmp_test.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst ctz_test.module build-grub-module-verifier ctz_test.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst bswap_test.module build-grub-module-verifier bswap_test.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst videotest_checksum.module build-grub-module-verifier videotest_checksum.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst gfxterm_menu.module build-grub-module-verifier gfxterm_menu.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst cmdline_cat_test.module build-grub-module-verifier cmdline_cat_test.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst bitmap.module build-grub-module-verifier bitmap.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst bitmap_scale.module build-grub-module-verifier bitmap_scale.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst efi_gop.module build-grub-module-verifier efi_gop.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst efi_uga.module build-grub-module-verifier efi_uga.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst jpeg.module build-grub-module-verifier jpeg.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst png.module build-grub-module-verifier png.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst tga.module build-grub-module-verifier tga.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst video_fb.module build-grub-module-verifier video_fb.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst video.module build-grub-module-verifier video.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst video_colors.module build-grub-module-verifier video_colors.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst datehook.module build-grub-module-verifier datehook.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst net.module build-grub-module-verifier net.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst tftp.module build-grub-module-verifier tftp.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst http.module build-grub-module-verifier http.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst efinet.module build-grub-module-verifier efinet.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst legacycfg.module build-grub-module-verifier legacycfg.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst syslinuxcfg.module build-grub-module-verifier syslinuxcfg.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst test_blockarg.module build-grub-module-verifier test_blockarg.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst xzio.module build-grub-module-verifier xzio.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst lzopio.module build-grub-module-verifier lzopio.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst testload.module build-grub-module-verifier testload.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst backtrace.module build-grub-module-verifier backtrace.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst keylayouts.module build-grub-module-verifier keylayouts.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst priority_queue.module build-grub-module-verifier priority_queue.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst time.module build-grub-module-verifier time.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst adler32.module build-grub-module-verifier adler32.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst crc64.module build-grub-module-verifier crc64.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst mpi.module build-grub-module-verifier mpi.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst all_video.module build-grub-module-verifier all_video.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst testspeed.module build-grub-module-verifier testspeed.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst tpm.module build-grub-module-verifier tpm.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst tr.module build-grub-module-verifier tr.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst progress.module build-grub-module-verifier progress.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst file.module build-grub-module-verifier file.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst rdmsr.module build-grub-module-verifier rdmsr.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst wrmsr.module build-grub-module-verifier wrmsr.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst gcry_arcfour.module build-grub-module-verifier gcry_arcfour.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst gcry_blowfish.module build-grub-module-verifier gcry_blowfish.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst gcry_camellia.module build-grub-module-verifier gcry_camellia.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst gcry_cast5.module build-grub-module-verifier gcry_cast5.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst gcry_crc.module build-grub-module-verifier gcry_crc.mod
TARGET_OBJ2ELF= sh genmod.sh moddep.lst gcry_des.module build-grub-module-verifier gcry_des.mod
[... 23 lines cut ...]
find /root/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/ipkg-x86_64/grub2-efi -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/root/lede/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/root/lede/scripts/strip-kmod.sh" PATCHELF="/root/lede/staging_dir/host/bin/patchelf" /root/lede/scripts/rstrip.sh /root/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/ipkg-x86_64/grub2-efi
(cd /root/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/ipkg-x86_64/grub2-efi/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /root/lede/bin/targets/x86/64/packages
/root/lede/staging_dir/host/bin/fakeroot /root/lede/staging_dir/host/bin/bash /root/lede/scripts/ipkg-build -m "" /root/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/ipkg-x86_64/grub2-efi /root/lede/bin/targets/x86/64/packages
Packaged contents of /root/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/ipkg-x86_64/grub2-efi into /root/lede/bin/targets/x86/64/packages/grub2-efi_2.06-6_x86_64.ipk
rm -rf /root/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/.pkgdir/grub2-efi.installed /root/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/.pkgdir/grub2-efi
mkdir -p /root/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/.pkgdir/grub2-efi
sed 's#msdos1#gpt1#g' ./files/grub-early.cfg >/root/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-early.cfg
/root/lede/staging_dir/host/bin/grub-mkimage -d /root/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-core -p /boot/grub -O "x86_64"-efi -c /root/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-early.cfg -o /root/lede/staging_dir/target-x86_64_musl/image/grub2/bootx64.efi at_keyboard boot chain configfile fat linux ls part_gpt reboot serial efi_gop efi_uga
/root/lede/staging_dir/host/bin/grub-mkimage -d /root/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-core -p /boot/grub -O "x86_64"-efi -c ./files/grub-early.cfg -o /root/lede/staging_dir/target-x86_64_musl/image/grub2/iso-bootx64.efi at_keyboard boot chain configfile fat iso9660 linux ls part_msdos part_gpt reboot serial test efi_gop efi_uga
touch /root/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/.pkgdir/grub2-efi.installed
mkdir -p /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /root/lede/tmp/.root-copy.flock -c 'cp -fpR /root/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/.pkgdir/grub2-efi/. /root/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.grub2-efi_installed
[... cut ...]
make[3]: Leaving directory '/root/lede/package/boot/grub2'
time: package/boot/grub2/efi/compile#117.21#22.87#137.41
make[2]: Leaving directory '/root/lede'
make[1]: *** [package/Makefile:110: /root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/lede'
make: *** [/root/lede/include/toplevel.mk:231: world] Error 2
----------------------------------------------------------------------------
★[2023-04-07 01:09:45] Successful
----------------------------------------------------------------------------

baodp, 2023-04-07:
一样的错误，默认配置的X86，点选了6.1内核、IPV6，加了PASSWALL2。
全新拉的库编译。

WYC-2020, 2023-04-07:
算不出来 用j1

baodp, 2023-04-08:
确实要单核编译才过了。

closed 2023-04-11 by aiamadeus