coolsnowwolf/lede #11087: 6.1.23编译失败

GreatMichaelLee, 2023-04-07:
Applying /home/builder/lede_x86/target/linux/generic/pending-6.1/732-13-net-ethernet-mtk_eth_soc-fix-flow_offload-related-re.patch using plaintext:
patching file drivers/net/ethernet/mediatek/mtk_ppe_offload.c
Hunk #2 FAILED at 569.
1 out of 2 hunks FAILED -- saving rejects to file drivers/net/ethernet/mediatek/mtk_ppe_offload.c.rej
Patch failed!  Please fix /home/builder/lede_x86/target/linux/generic/pending-6.1/732-13-net-ethernet-mtk_eth_soc-fix-flow_offload-related-re.patch!
make[4]: *** [Makefile:35: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.23/.prepared_af0589fd4f56396a3a8b70f646f5dae8] Error 1
make[4]: Leaving directory '/home/builder/lede_x86/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/target/linux'
time: target/linux/compile#10.54#6.95#27.22
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [target/Makefile:23: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231: world] Error 2
builder@Supernode:~/lede_x86$

baodp, 2023-04-07:
一样的 等大佬修复。哈哈

CYaNu, 2023-04-07:
第二条有一行错位了 直接手动改下这个patch的第二条就行了 然后把patch第二条删掉就可以了

CYaNu, 2023-04-07:
或者直接改下这个patch 把最后一个if前的一句往前提2行 记得 我改完就没记 直接编译了 大概吧

GreatMichaelLee, 2023-04-07:
> /home/builder/lede_x86/target/linux/generic/pending-6.1/732-13-net-ethernet-mtk_eth_soc-fix-flow_offload-related-re.patch

不懂，哈哈，我是菜鸟

GreatMichaelLee, 2023-04-07:
> 或者直接改下这个patch 把最后一个if前的一句往前提2行 记得 我改完就没记 直接编译了 大概吧

不懂，哈哈，我是菜鸟

ximiTech, 2023-04-07:
> 或者直接改下这个patch 把最后一个if前的一句往前提2行 记得 我改完就没记 直接编译了 大概吧

改了以后怎么编译呢？用make似乎还是在应用之前有错的patch？

CYaNu, 2023-04-07:
你改的patch还是.c 只要改对了 不会报错 我已经编译完了

ximiTech, 2023-04-07:
> 

改的 patch，重新运行make又会取原来有错的编译？

CYaNu, 2023-04-07:
不会啊 要不你就去手动改那个文件 改完把 patch第二条删了

ximiTech, 2023-04-07:
改完后执行什么命令呢？还是 make -j$(nproc) V=s ？

CYaNu, 2023-04-07:
我直接make V=s -j8

WYC-2020, 2023-04-07:
@@ -569,17 +569,23 @@ mtk_eth_setup_tc_block(struct net_device *dev, struct flow_block_offload *f)
 	switch (f->command) {
 	case FLOW_BLOCK_BIND:
 		block_cb = flow_block_cb_lookup(f->block, cb, dev);
-		if (block_cb) {
-			flow_block_cb_incref(block_cb);
-			return 0;
-		}
+		if (!block_cb) {
+			block_cb = flow_block_cb_alloc(cb, dev, dev, NULL);
+			if (IS_ERR(block_cb))
+				return PTR_ERR(block_cb);
+
+			register_block = true;
+ 		}
 		block_cb = flow_block_cb_alloc(cb, dev, dev, NULL);
 		if (IS_ERR(block_cb))
 			return PTR_ERR(block_cb);
 
 		flow_block_cb_incref(block_cb);
-		flow_block_cb_add(block_cb, f);
-		list_add_tail(&block_cb->driver_list, &block_cb_list);
+
+		if (register_block) {
+			flow_block_cb_add(block_cb, f);
+			list_add_tail(&block_cb->driver_list, &block_cb_list);
+		}
 		return 0;
 	case FLOW_BLOCK_UNBIND:
 		block_cb = flow_block_cb_lookup(f->block, cb, dev);

CYaNu, 2023-04-07:
也行 就是if前面一句的位置不对

CYaNu, 2023-04-07:
往前提2行就行

WYC-2020, 2023-04-07:
刚patch有错 用这个patch吧
[732-13-net-ethernet-mtk_eth_soc-fix-flow_offload-related-re.patch](https://github.com/coolsnowwolf/lede/files/11178109/732-13-net-ethernet-mtk_eth_soc-fix-flow_offload-related-re.patch)
直接替换不行的话就直接拷贝最后那个节点 复制到你们相同的patch 即可，因为我这边patch有可能和你们的不同

closed 2023-04-11 by aiamadeus