coolsnowwolf/lede #11092: 安装docker报错：opkg_install_cmd: Cannot install package dockerd.

coch1, 2023-04-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：安装docke时出现问题，应该是dockerd的报错。
重现：
opkg update
opkg install luci-lib-docker dockerd luci-lib-jsonc docker
docker --version（有版本号）

### (2) 路由器型号和固件版本

A：x86 j4125 ESXi 8.0 虚拟机 其他 6.x 或更高版本 Linux (64 位)
Linux version 6.1.22 (lean@lean-esxi) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r5834-89e25b5a13) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT_DYNAMIC Fri Mar 31 05:34:43 2023

### (3) 详细日志

A：root@OpenWrt:~# opkg install luci-lib-docker dockerd luci-lib-jsonc docker
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Package luci-lib-docker (v0.3.4) installed in root is up to date.
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Installing dockerd (23.0.1-1) to root...
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/x86_64/packages/dockerd_23.0.1-1_x86_64.ipk
Package luci-lib-jsonc (git-23.087.57160-8186ccc-1) installed in root is up to date.
Package docker (23.0.1-1) installed in root is up to date.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-crypto-hash
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-crypto-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-lib-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-crypto-acompress
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-lib-lzo
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-lib-zlib-inflate
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-lib-zlib-deflate
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-lib-raid6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-lib-raid6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-lib-xor
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-lib-xor found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-lib-zstd
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-lib-zstd found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-fs-btrfs
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-fs-btrfs found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nf-log
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-ipt-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nfnetlink
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nf-reject6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-reject6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nf-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nf-conntrack6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nf-nat
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nf-log6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-log6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nft-core
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-core found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nft-compat
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-compat found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-ipt-extra
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nf-ipt6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-ipt6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-ip6tables
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ip6tables found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nf-nat6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-nat6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-ipt-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-ipt-nat
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-ipt-nat6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-nat6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-br-netfilter
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-br-netfilter found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-ipt-physdev
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-physdev found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nf-ipvs
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-ipvs found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-veth
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-veth found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nft-fib
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-fib found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nf-flow
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-flow found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nft-nat
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-nat found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nft-offload
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-offload found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for dockerd:
 *      kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f)
 * opkg_install_cmd: Cannot install package dockerd.

hxlls, 2023-04-09:
docker 有内核依赖，要选对应指纹的内核组件安装后才能安装docker ，嫌麻烦就直接编译进去

WYC-2020, 2023-04-10:
如果确定自己 这些库都有，只是 版本对不上的话 可以试下强制安装

closed 2023-04-11 by aiamadeus