coolsnowwolf/lede #11093: 6.1编译报错

boyu899, 2023-04-09:
[image]

WYC-2020, 2023-04-10:
算不出来，j1日志

ShireCiel, 2023-04-10:
make -j1 V=s 重新编译后 把最后两页都贴出来。

boyu899, 2023-04-10:
> 算不出来，j1日志

好的 再试试 也参考过你在别的议题的办法 还是不行 

不过我编译5.15之前一直是成功 
新6.1是首次尝试 这几天都试过了j1单线程 也是失败

WYC-2020, 2023-04-10:
> > 算不出来，j1日志
> 
> 好的 再试试 也参考过你在别的议题的办法 还是不行
> 
> 不过我编译5.15之前一直是成功 新6.1是首次尝试 这几天都试过了j1单线程 也是失败

我意思是 你贴j1 日志  你 make -j1 V=s  出错后吧日志贴上了，多线程日志有时候错误位置你们没贴到，所以算不出来

boyu899, 2023-04-10:
github.com/jackc/pgx/v5/internal/nbconn
github.com/aws/aws-sdk-go/service/s3/s3iface
golang.org/x/text/internal
github.com/jackc/pgx/v5/pgproto3
golang.org/x/text/cases
gorm.io/driver/mysql
github.com/jackc/pgx/v5/pgtype
github.com/aws/aws-sdk-go/service/s3/s3manager
google.golang.org/protobuf/internal/filetype
github.com/alist-org/alist/v3/server/common
github.com/mattn/go-sqlite3
google.golang.org/protobuf/runtime/protoimpl
golang.org/x/text/secure/precis
google.golang.org/protobuf/types/descriptorpb
internal/profile
github.com/alist-org/alist/v3/internal/fs
github.com/alist-org/alist/v3/drivers/123
github.com/alist-org/alist/v3/drivers/alist_v2
github.com/alist-org/alist/v3/drivers/alist_v3
github.com/u2takey/ffmpeg-go
github.com/alist-org/alist/v3/drivers/mediatrack
github.com/alist-org/alist/v3/drivers/alias
github.com/alist-org/alist/v3/drivers/pikpak
github.com/alist-org/alist/v3/drivers/s3
github.com/alist-org/alist/v3/drivers/thunder
google.golang.org/protobuf/reflect/protodesc
github.com/jackc/pgx/v5/pgconn
github.com/alist-org/alist/v3/internal/message
runtime/pprof
github.com/alist-org/alist/v3/drivers/local
github.com/alist-org/alist/v3/internal/bootstrap/data
runtime/trace
github.com/boombuler/barcode
github.com/alist-org/alist/v3/drivers
github.com/boombuler/barcode/utils
github.com/pquerna/otp/internal
github.com/alist-org/alist/v3/server/middlewares
github.com/alist-org/alist/v3/server/webdav/internal/xml
net/http/pprof
github.com/golang/protobuf/proto
github.com/boombuler/barcode/qr
github.com/gin-contrib/cors
github.com/jackc/pgx/v5/internal/stmtcache
github.com/alist-org/alist/v3/server/static
encoding/csv
os/signal
github.com/alist-org/alist/v3/drivers/template
github.com/pquerna/otp
github.com/spf13/pflag
github.com/pquerna/otp/hotp
github.com/alist-org/alist/v3/pkg/gowebdav/cmd/gowebdav
github.com/blevesearch/bleve/v2/index/upsidedown
github.com/winfsp/cgofuse/fuse
github.com/pquerna/otp/totp
github.com/alist-org/alist/v3/server/webdav
# github.com/winfsp/cgofuse/fuse
/home/li/lede/dl/go-mod-cache/github.com/winfsp/cgofuse@v1.5.0/fuse/host_cgo.go:116:10: fatal error: fuse.h: No such file or directory
  116 | #include <fuse.h>
      |          ^~~~~~~~
compilation terminated.
github.com/blevesearch/bleve/v2
github.com/spf13/cobra
github.com/alist-org/alist/v3/internal/search/bleve
github.com/alist-org/alist/v3/internal/search
github.com/alist-org/alist/v3/server/handles
github.com/alist-org/alist/v3/server
github.com/jackc/pgx/v5
github.com/jackc/pgx/v5/stdlib
gorm.io/driver/postgres
gorm.io/driver/sqlite
github.com/alist-org/alist/v3/internal/bootstrap
github.com/alist-org/alist/v3/cmd
github.com/alist-org/alist/v3

make[3]: *** [Makefile:72: /home/li/lede/build_dir/target-x86_64_musl/alist-3.15.1/.built] Error 1
make[3]: Leaving directory '/home/li/lede/feeds/kiddin9/alist'
time: package/feeds/kiddin9/alist/compile#136.61#51.72#61.89
    ERROR: package/feeds/kiddin9/alist failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kiddin9/alist/compile] Error 1
make[2]: Leaving directory '/home/li/lede'
make[1]: *** [package/Makefile:110: /home/li/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/li/lede'
make: *** [/home/li/lede/include/toplevel.mk:231：world] 错误 2

make -j1 V=s    错误如下

boyu899, 2023-04-10:
取消alist 重新编 speedtestweb又出错了？ 这个之前5.15每次都能成功编译进去 [image]

WYC-2020, 2023-04-11:
等下看下

boyu899, 2023-04-11:
> alist安装sudo apt install libfuse-dev  但是我感觉应该改成依赖，先用安装的形式 编译出来看下能否运行吧
> 
> 你那个speedtest-web没找到这个玩意，只有这两个speedtest-netperf speedtest-go,看你那个是没找到upx 压缩 自己加上upx依赖即可

非常感谢 晚上开干

closed 2023-04-12 by coolsnowwolf