coolsnowwolf/lede #11094: 报错  大神看一下 哪里问题

zhiwen1987, 2023-04-11:
make[3]: Entering directory '/home/bl/lede/feeds/helloworld/gn'
ninja -C /home/bl/lede/build_dir/hostpkg/gn-2023-02-17-b25a2f8c/out
ninja: Entering directory `/home/bl/lede/build_dir/hostpkg/gn-2023-02-17-b25a2f8c/out'
[1/287] CXX src/base/files/file_util.o
FAILED: src/base/files/file_util.o 
g++ -MMD -MF src/base/files/file_util.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++17 -c ../src/base/files/file_util.cc -o src/base/files/file_util.o
g++: error: unrecognized command line option '-Wextra-semi'; did you mean '-Wextra'?
[2/287] CXX src/base/files/file_path.o
FAILED: src/base/files/file_path.o 
g++ -MMD -MF src/base/files/file_path.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++17 -c ../src/base/files/file_path.cc -o src/base/files/file_path.o
g++: error: unrecognized command line option '-Wextra-semi'; did you mean '-Wextra'?
[3/287] CXX src/base/command_line.o
FAILED: src/base/command_line.o 
g++ -MMD -MF src/base/command_line.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++17 -c ../src/base/command_line.cc -o src/base/command_line.o
g++: error: unrecognized command line option '-Wextra-semi'; did you mean '-Wextra'?
[4/287] CXX src/base/files/scoped_file.o
FAILED: src/base/files/scoped_file.o 
g++ -MMD -MF src/base/files/scoped_file.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++17 -c ../src/base/files/scoped_file.cc -o src/base/files/scoped_file.o
g++: error: unrecognized command line option '-Wextra-semi'; did you mean '-Wextra'?
[5/287] CXX src/base/json/json_parser.o
FAILED: src/base/json/json_parser.o 
g++ -MMD -MF src/base/json/json_parser.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++17 -c ../src/base/json/json_parser.cc -o src/base/json/json_parser.o
g++: error: unrecognized command line option '-Wextra-semi'; did you mean '-Wextra'?
[6/287] CXX src/base/files/scoped_temp_dir.o
FAILED: src/base/files/scoped_temp_dir.o 
g++ -MMD -MF src/base/files/scoped_temp_dir.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++17 -c ../src/base/files/scoped_temp_dir.cc -o src/base/files/scoped_temp_dir.o
g++: error: unrecognized command line option '-Wextra-semi'; did you mean '-Wextra'?
[7/287] CXX src/base/files/file.o
FAILED: src/base/files/file.o 
g++ -MMD -MF src/base/files/file.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++17 -c ../src/base/files/file.cc -o src/base/files/file.o
g++: error: unrecognized command line option '-Wextra-semi'; did you mean '-Wextra'?
[8/287] CXX src/base/environment.o
FAILED: src/base/environment.o 
g++ -MMD -MF src/base/environment.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++17 -c ../src/base/environment.cc -o src/base/environment.o
g++: error: unrecognized command line option '-Wextra-semi'; did you mean '-Wextra'?
[9/287] CXX src/base/files/file_path_constants.o
FAILED: src/base/files/file_path_constants.o 
g++ -MMD -MF src/base/files/file_path_constants.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++17 -c ../src/base/files/file_path_constants.cc -o src/base/files/file_path_constants.o
g++: error: unrecognized command line option '-Wextra-semi'; did you mean '-Wextra'?
[10/287] CXX src/base/files/file_enumerator.o
FAILED: src/base/files/file_enumerator.o 
g++ -MMD -MF src/base/files/file_enumerator.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++17 -c ../src/base/files/file_enumerator.cc -o src/base/files/file_enumerator.o
g++: error: unrecognized command line option '-Wextra-semi'; did you mean '-Wextra'?
ninja: build stopped: subcommand failed.
Makefile:60: recipe for target '/home/bl/lede/build_dir/hostpkg/gn-2023-02-17-b25a2f8c/.built' failed
make[3]: *** [/home/bl/lede/build_dir/hostpkg/gn-2023-02-17-b25a2f8c/.built] Error 1
make[3]: Leaving directory '/home/bl/lede/feeds/helloworld/gn'
time: package/feeds/helloworld/gn/host-compile#0.22#0.09#2.45
    ERROR: package/feeds/helloworld/gn [host] failed to build.
package/Makefile:114: recipe for target 'package/feeds/helloworld/gn/host/compile' failed
make[2]: *** [package/feeds/helloworld/gn/host/compile] Error 1
make[2]: Leaving directory '/home/bl/lede'
package/Makefile:110: recipe for target '/home/bl/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/bl/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bl/lede'
/home/bl/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2

AE86JAY, 2023-04-11:
我也出现这个问题，据说是helloworld下载不完整，我没有梯子，重新git也是一样。

WYC-2020, 2023-04-11:
这问题已经说过了，自己去看

zhiwen1987, 2023-04-11:
> 我也出现这个问题，据说是helloworld下载不完整，我没有梯子，重新git也是一样。可以参考这个教程升级 g++ gcc

[image]

zhiwen1987, 2023-04-11:
> 我也出现这个问题，据说是helloworld下载不完整，我没有梯子，重新git也是一样。
可以参考 这个教程  升级 g++ gcc 我也是刚升级到11 看看 会不会报错
https://www.cnblogs.com/zengjianrong/p/16624273.html

zhiwen1987, 2023-04-11:
> 
大神 厉害。

coolsnowwolf, 2026-09-05:
失败点是 2023 年 helloworld GN 源码向宿主 g++ 传入不支持的 `-Wextra-semi`，属于外部源码/宿主编译器兼容性问题，不是当前 LEDE 核心包的可验证缺陷。

closed 2026-09-05 by coolsnowwolf