coolsnowwolf/lede #11097: libgfortran.so*:No such file or directory

xingyuezhiyun, 2023-04-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
步骤
make menuconfig
base system -> libgfortran  -> [*]

make -j6 V=99
无法添加libgfortran包

### (1) 具体问题

A：无法添加libgfortran包

### (2) 路由器型号和固件版本

A：编译环境: 
Linux ubuntu 5.15.0-67-generic #74~20.04.1-Ubuntu SMP Wed Feb 22 14:52:34 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

### (3) 详细日志

A：
make[3]: Entering directory '/home/luwenle/work/lede/feeds/luci/modules/luci-base'
make[3]: Entering directory '/home/luwenle/work/lede/package/utils/lua'
make[3]: Leaving directory '/home/luwenle/work/lede/package/libs/ncurses'
time: package/libs/ncurses/host-compile#0.26#0.32#0.51
make[3]: Entering directory '/home/luwenle/work/lede/package/libs/gettext-full'
touch /home/luwenle/work/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libc_installed
mkdir -p /home/luwenle/work/lede/bin/targets/x86/64/packages /home/luwenle/work/lede/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgfortran/CONTROL /home/luwenle/work/lede/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/luwenle/work/lede/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgfortran/usr/lib
cp -fpR /home/luwenle/work/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/libgfortran.so.* /home/luwenle/work/lede/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgfortran/usr/lib/
cp: cannot stat '/home/luwenle/work/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/libgfortran.so.*': No such file or directory
make[3]: *** [Makefile:737: /home/luwenle/work/lede/bin/targets/x86/64/packages/libgfortran_11.3.0-4_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/luwenle/work/lede/package/libs/toolchain'
time: package/libs/toolchain/compile#0.21#0.36#0.72
    ERROR: package/libs/toolchain failed to build.
make[2]: *** [package/Makefile:116: package/libs/toolchain/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/home/luwenle/work/lede/package/utils/lua'
time: package/utils/lua/host-compile#0.22#0.24#0.39
make[3]: Leaving directory '/home/luwenle/work/lede/package/boot/grub2'
time: package/boot/grub2/host-compile#0.40#0.47#0.74
make[3]: Leaving directory '/home/luwenle/work/lede/package/libs/gettext-full'
time: package/libs/gettext-full/host-compile#0.30#0.56#0.79
 make[4] scripts/config/conf
. /home/luwenle/work/lede/include/shell.sh; bzcat /home/luwenle/work/lede/dl/LuaSrcDiet-0.12.1.tar.bz2 | tar -C /home/luwenle/work/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/luwenle/work/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1
touch /home/luwenle/work/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.prepared2d1a1d0fbb6a2b67147b2e0dcf3b3ee2_6664517399ebbbc92a37c5bb081b5c53
touch /home/luwenle/work/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.configured
make -C src/ clean po2lmo
make[4]: Entering directory '/home/luwenle/work/lede/feeds/luci/modules/luci-base/src'
rm -f po2lmo parser.so version.lua *.o
cc    -c -o po2lmo.o po2lmo.c
cc    -c -o template_lmo.o template_lmo.c
cc  -o po2lmo po2lmo.o template_lmo.o
make[4]: Leaving directory '/home/luwenle/work/lede/feeds/luci/modules/luci-base/src'
make -C /home/luwenle/work/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1 bin/LuaSrcDiet.lua
make[4]: Entering directory '/home/luwenle/work/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1'
make[4]: Nothing to be done for 'bin/LuaSrcDiet.lua'.
make[4]: Leaving directory '/home/luwenle/work/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1'
touch /home/luwenle/work/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.built
install -d -m0755 /home/luwenle/work/lede/staging_dir/hostpkg/bin
install -m0755 src/po2lmo /home/luwenle/work/lede/staging_dir/hostpkg/bin/po2lmo
install -m0755 /home/luwenle/work/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/bin/LuaSrcDiet.lua /home/luwenle/work/lede/staging_dir/hostpkg/bin/LuaSrcDiet
mkdir -p /home/luwenle/work/lede/staging_dir/hostpkg/stamp
touch /home/luwenle/work/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.built
touch /home/luwenle/work/lede/staging_dir/hostpkg/stamp/.luci-base_installed
make[3]: Leaving directory '/home/luwenle/work/lede/feeds/luci/modules/luci-base'
time: package/feeds/luci/luci-base/host-compile#2.05#4.38#5.99
make[2]: Leaving directory '/home/luwenle/work/lede'
make[1]: *** [package/Makefile:110: /home/luwenle/work/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/luwenle/work/lede'
make[1]: Leaving directory '/home/luwenle/work/lede'
make: *** [/home/luwenle/work/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2023-04-11:
看下是不是升级过，输出名字或者路径变了呗

xingyuezhiyun, 2023-04-12:
> 看下是不是升级过，输出名字或者路径变了呗

没有啊,我已经把要安装的项目打包到pypi了,你可以 pip install yolotest 试下看下能不能安装,我这边安装一直报依赖的错

coolsnowwolf, 2026-09-05:
日志明确是选择 `libgfortran` 后，当前工具链没有生成 `libgfortran.so.*`，属于工具链配置/构建状态不一致，而不是包文件路径的固定错误；当前安装逻辑按工具链实际产物复制，清理工具链并重新配置编译即可排查。

closed 2026-09-05 by coolsnowwolf