coolsnowwolf/lede #11098: H66K，今日云编译报错，package/boot/uboot-rockchip failed to build

ZZMISKING, 2023-04-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
X
## 2. 详细叙述

### (1) 具体问题

A：H66K，4月12日上午，云编译报错

### (2) 路由器型号和固件版本

A：H66K rk3568

### (3) 详细日志

A：make[4]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-opc-h68k-rk3568/u-boot-2023.04'
  ENVC    include/generated/env.txt
  ENVP    include/generated/env.in
  ENVT    include/generated/environment.h
make -f ./scripts/Makefile.build obj=scripts/dtc
  CC      env/common.o
  AR      env/built-in.o
  CC      lib/efi_loader/dtbdump.o
  LD      lib/efi_loader/dtbdump_efi.so
  OBJCOPY lib/efi_loader/dtbdump.efi
  CC      lib/efi_loader/initrddump.o
  LD      lib/efi_loader/initrddump_efi.so
  OBJCOPY lib/efi_loader/initrddump.efi
  LD      u-boot
  OBJCOPY u-boot.srec
  OBJCOPY u-boot-nodtb.bin
  RELOC   u-boot-nodtb.bin
  CAT     u-boot-dtb.bin
  COPY    u-boot.bin
  SYM     u-boot.sym
  MKIMAGE u-boot.img
  MKIMAGE u-boot-dtb.img
  BINMAN  .binman_stamp
binman: Node '/binman/simple-bin/fit': subnode 'images/@atf-SEQ': Failed to read ELF file: Python: No module named 'elftools'
Wrote map file './simple-bin.map' to show errors
make[4]: *** [Makefile:1108: .binman_stamp] Error 1
make[3]: *** [Makefile:334: /workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-opc-h68k-rk3568/u-boot-2023.04/.built] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-opc-h68k-rk3568/u-boot-2023.04'
make[3]: Leaving directory '/workdir/openwrt/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/opc-h68k-rk3568/compile#2.98#0.85#3.80
    ERROR: package/boot/uboot-rockchip failed to build (build variant: opc-h68k-rk3568).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

aiamadeus, 2023-04-12:
说过了，你的 action 没有更新 apt 依赖。

BenjaminX, 2023-04-12:
同一个错误，需要更新系统的那些依赖包？

ZZMISKING, 2023-04-12:
> 同一个错误，需要更新系统的那些依赖包？

更新readme 里的apt

BenjaminX, 2023-04-12:
sudo -E apt-get -qq install ack antlr3 asciidoc autoconf automake autopoint binutils bison build-essential bzip2 ccache cmake cpio curl device-tree-compiler fastjar flex gawk gettext gcc-multilib g++-multilib git gperf haveged help2man intltool libc6-dev-i386 libelf-dev libglib2.0-dev libgmp3-dev libltdl-dev libmpc-dev libmpfr-dev libncurses5-dev libncursesw5-dev libreadline-dev libssl-dev libtool lrzsz mkisofs msmtp nano ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 python3-pyelftools libpython3-dev qemu-utils rsync scons squashfs-tools subversion swig texinfo uglifyjs upx-ucl unzip vim wget xmlto xxd zlib1g-dev

更新后可以编译完成了，可以 close 了

closed 2023-04-12 by ZZMISKING