coolsnowwolf/lede #11099: 关于在活动连接、客户端多的时候，软中断变多，且CPU占用会变高，网速变慢的问题

JackHuang90, 2023-04-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：关于在活动连接、客户端多的时候，软中断变多，且CPU占用会变高，网速变慢的问题
目前连接数在5000左右，客户端在65左右，使用top命令查看占用情况，会发现ksoftirqd占用在大量使用网速时会变高。随着开机时间推移，ksoftirqd占用CPU会越来越高，导致网速变慢。
不知道是什么在影响，是不是一些软件的原因？
这个是开机3小时的截图
[image]
[image]

### (2) 路由器型号和固件版本

A：
版本 OpenWrt R23.4.1 / LuCI Master (git-23.096.61004-062aba3)，2023.04.11拉取的master分支编译的。
CPU i5 6500
内存 16GB
网卡
wan  RTL8125B 2.5G in 1G
lan CX341a 10G + 板载网卡也是螃蟹卡1G速度
实体机安装的Openwrt
### (3) 详细日志

cat /proc/softirqs
[image]

[    0.000000] Linux version 5.15.105 (flision@flision-Virtual-Machine) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r5877-c1a7461e8) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP Mon Apr 10 15:15:22 2023
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=d4f6499d-a887-1667-9646-257b28325f02 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 1616
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000b4c0efff] usable
[    0.000000] BIOS-e820: [mem 0x00000000b4c0f000-0x00000000b4c0ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000b4c10000-0x00000000b4c10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b4c11000-0x00000000b9aeafff] usable
[    0.000000] BIOS-e820: [mem 0x00000000b9aeb000-0x00000000b9e32fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b9e33000-0x00000000b9fa0fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000b9fa1000-0x00000000ba64ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000ba650000-0x00000000baea3fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000baea4000-0x00000000baefefff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000baeff000-0x00000000baefffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000baf00000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000043effffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.50 by American Megatrends
[    0.000000] efi: ACPI 2.0=0xba2a1000 ACPI=0xba2a1000 SMBIOS=0xbadcf000 SMBIOS 3.0=0xbadce000 ESRT=0xb8886718 
[... 709 lines cut ...]
[   88.086684] eth2: 0xffffc90000450000, 1c:fd:08:79:ba:f6, IRQ 18
[   91.116214] e1000e 0000:00:1f.6 eth0: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[   91.116976] br-lan: port 1(eth0) entered blocking state
[   91.117412] br-lan: port 1(eth0) entered forwarding state
[   91.117940] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   91.501630] r8125: eth2: link up
[   91.501953] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[   96.662354] pppoe-wan: renamed from ppp0
[  106.728269] squashfs: Unknown parameter 'discard'
[  106.728927] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x89e45589)
[  106.729589] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
[  106.730347] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x2e48937b)
[  106.730971] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
[  106.732160] exFAT-fs (sda128): invalid boot record signature
[  106.732655] exFAT-fs (sda128): failed to read boot sector
[  106.733107] exFAT-fs (sda128): failed to recognize exfat type
[  106.734142] fuseblk: Unknown parameter 'discard'
[  112.786471] init oaf sysctl...ok
[  113.128839] TCP: request_sock_TCPv6: Possible SYN flooding on port 53. Sending cookies.  Check SNMP counters.
[ 5935.191760] TCP: request_sock_TCPv6: Possible SYN flooding on port 53. Sending cookies.  Check SNMP counters.
[ 7024.881669] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[ 7024.882573] br-lan: port 1(eth0) entered disabled state
[ 7031.357886] mlx4_en: eth1: Link Up
[ 7031.358525] br-lan: port 2(eth1) entered blocking state
[ 7031.358978] br-lan: port 2(eth1) entered forwarding state

JackHuang90, 2023-04-12:
> 你这个要分析 为啥这个玩意那么高，这个网上搜索又遇到你这种的，看下是否是他们那种情况 https://blog.csdn.net/weixin_34526937/article/details/116838976 https://zhuanlan.zhihu.com/p/359807721 如果不行 把你非必要的软件关闭，e2g 这个你关下试试

取消了e2g、ofa的选择，重新编译了，貌似在跑speedtest的时候，还是会高，观察1天看看。目前想到可能对性能有影响的 SQM、nlbwmon、WireGuard，不知道这几个有没有可能

coolsnowwolf, 2023-04-12:
你这个明显是 oaf 的性能问题

WYC-2020, 2023-04-13:
不用重新编译 你一个个 服务给他停了 然后禁用自启动 慢慢就能排查是那个的问题了

GreatMichaelLee, 2023-04-13:
参考我提的issue,同样的问题 #9322

GreatMichaelLee, 2023-04-13:
> 参考我提的issue,同样的问题 #9322

by the way, 我没有开e2g,oaf这些东西，不过我有nlbwmon

JackHuang90, 2023-04-13:
> > 参考我提的issue,同样的问题 #9322
> 
> by the way, 我没有开e2g,oaf这些东西，不过我有nlbwmon

有看到你的issue，我这里的表现就是进过1天使用，会越来越占CPU。明天尝试只编译少量软件再试试，nlbwmon也不编译了。

fishg, 2023-04-14:
可以先netdata装上，异常的时候方便查各项指标

fishg, 2023-04-14:
Possible SYN flooding on port 53 这个可以注意下，有可能dns相关配置或者插件有问题

JackHuang90, 2023-04-18:
> Possible SYN flooding on port 53 这个可以注意下，有可能dns相关配置或者插件有问题

这个是lan的设置
[image]

这个是DHCP/DNS设置
[image]

这个是Turbo ACC 网络加速设置
[image]

貌似没啥问题

JackHuang90, 2023-04-18:
目前将大多数可能和性能有关的软件都禁用了，在运行2天后，还是会出现网速明显降低的情况。
现在重新编译一个去掉可能耗性能软件的版本试一下。

fishg, 2023-04-18:
> 目前将大多数可能和性能有关的软件都禁用了，在运行2天后，还是会出现网速明显降低的情况。
> 现在重新编译一个去掉可能耗性能软件的版本试一下。

可以先不用重装。出现问题的时候首先看连接信息，保存到文本文件以便于后边看有没有异常的连接。然后试试以下服务依次重启，每重启一个，就看看恢复正常没有，观察下哪个重启哪个以后就能恢复正常，如果可以恢复正常，范围就缩小很多了。
/etc/init.d/turboacc restart
/etc/init.d/dnsmasq restart
/etc/init.d/firewall reload
其他在用的插件。。。
最后再考虑这个
/etc/init.d/network restart

GreatMichaelLee, 2023-04-18:
我觉得跟插件没关系，内核的问题，网卡驱动在大流量中fifo queue由于某些原因(可能是bug也可能是性能问题)有堆积，导致irq异常高处理不过来，我观察过，一旦问题发生，后面必然次次重现，除非重启，且网卡计数器里有不少的drop和overrunn

JackHuang90, 2023-04-20:
> 我觉得跟插件没关系，内核的问题，网卡驱动在大流量中fifo queue由于某些原因(可能是bug也可能是性能问题)有堆积，导致irq异常高处理不过来，我观察过，一旦问题发生，后面必然次次重现，除非重启，且网卡计数器里有不少的drop和overrunn
> […](#)

初步判断是内核的原因，我换了6.1.24内核编译，情况好很多，运行了2天没掉速，满速下CPU占用也没有那么高了，软中断也少了。过段时间我再多编译几个软件看看。

JackHuang90, 2023-05-06:
6.1.24内核编译情况好很多

closed 2023-05-06 by JackHuang90

niphor, 2023-10-26:
有没有运行OpenClash,我运行了这个一般2礼拜，或者开个百度网盘下个几个G东西，第2核的软中断就会100%，不管关闭什么软件都没用，只能重启（重启network服务，只能短暂释放，立马就又100%）

GreatMichaelLee, 2023-10-30:
> 有没有运行OpenClash,我运行了这个一般2礼拜，或者开个百度网盘下个几个G东西，第2核的软中断就会100%，不管关闭什么软件都没用，只能重启（重启network服务，只能短暂释放，立马就又100%）

这个问题目前继续在困扰我，内核都升到6.1.60了，也不知道到底是什么鬼，就是跑speedtest 就 ksoftirq 100%， 重启能好，不定时就出来了。

GreatMichaelLee, 2024-07-06:
这个问题直到今天也没有任何解决方案，网上有一堆人报这个问题

JackHuang90, 2024-07-07:
> 这个问题直到今天也没有任何解决方案，网上有一堆人报这个问题

不知道解决没，后面我重新拉了新代码，重新编译之后好了，不知道现在代码有没有这个问题。

fayeyuan, 2024-08-12:
> 可以先不用重装。出现问题的时候首先看连接信息，保存到文本文件以便于后边看有没有异常的连接。然后试试以下服务依次重启，每重启一个，就看看恢复正常没有，观察下哪个重启哪个以后就能恢复正常，如果可以恢复正常，范围就缩小很多了。 /etc/init.d/turboacc restart /etc/init.d/dnsmasq restart /etc/init.d/firewall reload [...]

/etc/init.d/turboacc restart
试了这一条能迅速降低cpu占用

GreatMichaelLee, 2024-08-14:
这个turboacc里有什么tricky 会导致降速呢？我看到的降速的是有大量的softirq 耗尽了CPU

发自我的iPhone

fayeyuan, 2024-08-17:
> 这个turboacc里有什么tricky 会导致降速呢？我看到的降速的是有大量的softirq 耗尽了CPU 发自我的iPhone
> […](#)

临时解决，感觉就像是重置网络，但没有解决实际问题，openwrt一段时间后还是会崩溃自动重启。

suenley, 2024-11-12:
> 这个turboacc里有什么tricky 会导致降速呢？我看到的降速的是有大量的softirq 耗尽了CPU 发自我的iPhone

挖个坟，这个问题解决了吗?