coolsnowwolf/lede #11100: openwrt 默认编译报错！！！

adamgonglj, 2023-04-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
OpenWrt-libtool: install: warning: remember to run `libtool --finish /usr/lib'
OpenWrt-libtool: install: warning: `../lib/libsysfs.la' has not been installed in `/usr/lib'
OpenWrt-libtool: install: warning: `../lib/libsysfs.la' has not been installed in `/usr/lib'
OpenWrt-libtool: install: warning: `../lib/libsysfs.la' has not been installed in `/usr/lib'
OpenWrt-libtool: install: warning: `../lib/libsysfs.la' has not been installed in `/usr/lib'
OpenWrt-libtool: install: warning: `../lib/libsysfs.la' has not been installed in `/usr/lib'
/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: cannot find -liconv
collect2: error: ld returned 1 exit status
make[4]: *** [prog/sensors/Module.mk:43: prog/sensors/sensors] Error 1
make[3]: *** [Makefile:129: /home/ubuntu/lede/build_dir/target-x86_64_musl/lm-sensors-3-6-0/.built] Error 2
make[2]: *** [package/Makefile:116: package/feeds/packages/lm-sensors/compile] Error 1
make[1]: *** [package/Makefile:110: /home/ubuntu/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make: *** [/home/ubuntu/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2023-04-18:
万能大法，x86 这个基本不会出错

adamgonglj, 2023-04-18:
我再试一下

adamgonglj, 2023-04-21:
经过尝试x86就是会出错，默认配置都出错。

WYC-2020, 2023-04-21:
> 经过尝试x86就是会出错，默认配置都出错。

是吗我这边都不会出错，make distclean?  出错就贴错误日志，要是真有问题我想早就应该炸了？不是组合怪吧

adamgonglj, 2023-04-25:
就是报错。我是直接复制的readme的代码，config是默认的。
[image]

adamgonglj, 2023-04-25:
@coolsnowwolf 求解决

AE86JAY, 2023-05-06:
多执行几次 update 、 install 、make download