coolsnowwolf/lede #11101: RK3328 RK3568 UBOOT编译出错

VIKINGYFY, 2023-04-12:
## 1.关于你要提交的问题
云编译报错，已经按照最新依赖库修改，但依然报同样错误。

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：源码更新uboot后一直编译失败，同时编译R2S R4S R5C R66S
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_rockchip=y
CONFIG_TARGET_rockchip_armv8=y
CONFIG_TARGET_DEVICE_rockchip_armv8_DEVICE_fastrhino_r66s=y
CONFIG_TARGET_DEVICE_rockchip_armv8_DEVICE_friendlyarm_nanopi-r2s=y
CONFIG_TARGET_DEVICE_rockchip_armv8_DEVICE_friendlyarm_nanopi-r4s=y
CONFIG_TARGET_DEVICE_rockchip_armv8_DEVICE_friendlyarm_nanopi-r5c=y

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
binman: Node '/binman/simple-bin/fit': subnode 'images/@atf-SEQ': Failed to read ELF file: Python: No module named 'elftools'

lxl6125, 2023-04-12:
apt install python3-pyelftools

VIKINGYFY, 2023-04-12:
> apt install python3-pyelftools

依赖库列表已经更新过了
python相关的有
python3 python3-pip python3-pyelftools
python2.7 没装，另外 python3-pip 装不装都会报这个错

aiamadeus, 2023-04-12:
编译前 make clean，把 openwrt 编的 python 干了。

VIKINGYFY, 2023-04-12:
> 编译前 make clean，把 openwrt 编的 python 干了。

我放在make download前应该没事吧？

WYC-2020, 2023-04-13:
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev 
libz-dev patch python3 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs 
git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx 
libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf wget 
curl swig rsync aria2 ca-certificates python3-pyelftools libpython3-dev yasm python3-setuptools

@VIKINGYFY  新安装的ubuntu  用这一波依赖，我这边编译没问题

VIKINGYFY, 2023-04-13:
> sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs [...]

我发现是3328的uboot会编译失败，我把3328去掉后就正常了

aiamadeus, 2023-04-13:
不会啊，你哪个编不过日志发出来。

VIKINGYFY, 2023-04-13:
> 不会啊，你哪个编不过日志发出来。

我云编译的，大佬可以看一下
https://github.com/VIKINGYFY/OpenWRT-LEDE/actions

aiamadeus, 2023-04-13:
你就不能把错误日志复制出来。。我对你们的 action 链接没兴趣

VIKINGYFY, 2023-04-13:
> 你就不能把错误日志复制出来。。我对你们的 action 链接没兴趣

binman: Node '/binman/simple-bin/fit': subnode 'images/@atf-SEQ': Failed to read ELF file: Python: No module named 'elftools'
Wrote map file './simple-bin.map' to show errors
make[4]: *** [Makefile:1108: .binman_stamp] Error 1
make[3]: *** [Makefile:334: /home/runner/work/OpenWRT-LEDE/OpenWRT-LEDE/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r5s-rk3568/u-boot-2023.04/.built] Error 2
make[4]: Leaving directory '/home/runner/work/OpenWRT-LEDE/OpenWRT-LEDE/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r5s-rk3568/u-boot-2023.04'
make[3]: Leaving directory '/home/runner/work/OpenWRT-LEDE/OpenWRT-LEDE/lede/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/nanopi-r5s-rk3568/compile#2.68#0.69#3.32
    ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopi-r5s-rk3568).
make[2]: Leaving directory '/home/runner/work/OpenWRT-LEDE/OpenWRT-LEDE/lede'
make[2]: *** [package/Makefile:122: package/boot/uboot-rockchip/compile] Error 1
make[1]: Leaving directory '/home/runner/work/OpenWRT-LEDE/OpenWRT-LEDE/lede'
make[1]: *** [package/Makefile:110: /home/runner/work/OpenWRT-LEDE/OpenWRT-LEDE/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make: *** [/home/runner/work/OpenWRT-LEDE/OpenWRT-LEDE/lede/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

aiamadeus, 2023-04-13:
这不还是那个问题吗，你 action 缺 apt 依赖

VIKINGYFY, 2023-04-13:
> 这不还是那个问题吗，你 action 缺 apt 依赖

大佬，是这样的，在这个[提交](https://github.com/VIKINGYFY/OpenWRT-LEDE/tree/f48e4b1cc6bf6d40a28cbf05bbaee4e7f5fa2cd8)中的编译是过了的，当时的依赖是[这个](https://raw.githubusercontent.com/VIKINGYFY/OpenWRT-LEDE/f48e4b1cc6bf6d40a28cbf05bbaee4e7f5fa2cd8/Depends.txt)，当时的部分配置文件是[这个](https://raw.githubusercontent.com/VIKINGYFY/OpenWRT-LEDE/f48e4b1cc6bf6d40a28cbf05bbaee4e7f5fa2cd8/Config/Rockchip.txt)，这时候R2S的配置我是注释掉的，后来我又起了一次编译，我直接把R2S这行删掉了，但是这次失败了，依赖是按照你的提交修改的
python3-pip更换为python3-pyelftools

commit 64a79c97 (2023-04-16): uboot-rockchip: fix python link path (#11117)

Fix: #11101
closed 2023-04-16 by aiamadeus