coolsnowwolf/lede #11102: r68s r66s uboot编译不过去。。。麻烦看看

rshi0212, 2023-04-12:
time: package/boot/arm-trusted-firmware-rockchip-vendor/rk3328/compile#0.11#0.01#0.11
make[3]: Entering directory '/home/rshi0212/r68s_lede/package/boot/uboot-rockchip'
rm -f /home/rshi0212/r68s_lede/build_dir/target-aarch64_generic_musl/u-boot-r66s-rk3568/u-boot-2023.04/.built
touch /home/rshi0212/r68s_lede/build_dir/target-aarch64_generic_musl/u-boot-r66s-rk3568/u-boot-2023.04/.built_check
make  -C /home/rshi0212/r68s_lede/build_dir/target-aarch64_generic_musl/u-boot-r66s-rk3568/u-boot-2023.04 CROSS_COMPILE=aarch64-openwrt-linux-musl- DTC="/home/rshi0212/r68s_lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.104/scripts/dtc/dtc" HOSTCC="gcc" HOSTCFLAGS="-O2 -I/home/rshi0212/r68s_lede/staging_dir/host/include -I/home/rshi0212/r68s_lede/staging_dir/hostpkg/include -I/home/rshi0212/r68s_lede/staging_dir/target-aarch64_generic_musl/host/include -I/home/rshi0212/r68s_lede/staging_dir/host/include -I/home/rshi0212/r68s_lede/staging_dir/hostpkg/include -I/home/rshi0212/r68s_lede/staging_dir/target-aarch64_generic_musl/host/include -std=gnu11" HOSTLDFLAGS="-L/home/rshi0212/r68s_lede/staging_dir/host/lib -L/home/rshi0212/r68s_lede/staging_dir/hostpkg/lib -L/home/rshi0212/r68s_lede/staging_dir/target-aarch64_generic_musl/host/lib" LOCALVERSION="-OpenWrt-r5890-a4426eb5a" STAGING_PREFIX="/home/rshi0212/r68s_lede/staging_dir/host" PKG_CONFIG_PATH="/home/rshi0212/r68s_lede/staging_dir/host/lib/pkgconfig" PKG_CONFIG_LIBDIR="/home/rshi0212/r68s_lede/staging_dir/host/lib/pkgconfig" PKG_CONFIG_EXTRAARGS="--static" V='' BL31=/home/rshi0212/r68s_lede/staging_dir/target-aarch64_generic_musl/image/rk3568_bl31.elf ROCKCHIP_TPL=/home/rshi0212/r68s_lede/build_dir/target-aarch64_generic_musl/u-boot-r66s-rk3568/u-boot-2023.04/rk3568_ddr_1560MHz_v1.13.bin
make[4]: Entering directory '/home/rshi0212/r68s_lede/build_dir/target-aarch64_generic_musl/u-boot-r66s-rk3568/u-boot-2023.04'
  ENVC    include/generated/env.txt
  ENVP    include/generated/env.in
  ENVT    include/generated/environment.h
make -f ./scripts/Makefile.build obj=scripts/dtc
  CC      env/common.o
  AR      env/built-in.o
  CC      lib/efi_loader/dtbdump.o
  LD      lib/efi_loader/dtbdump_efi.so
  OBJCOPY lib/efi_loader/dtbdump.efi
  CC      lib/efi_loader/initrddump.o
  LD      lib/efi_loader/initrddump_efi.so
  OBJCOPY lib/efi_loader/initrddump.efi
  LD      u-boot
  OBJCOPY u-boot.srec
  OBJCOPY u-boot-nodtb.bin
  RELOC   u-boot-nodtb.bin
  CAT     u-boot-dtb.bin
  COPY    u-boot.bin
  SYM     u-boot.sym
  MKIMAGE u-boot.img
  MKIMAGE u-boot-dtb.img
  BINMAN  .binman_stamp
Wrote map file './simple-bin.map' to show errors
binman: Node '/binman/simple-bin/fit': subnode 'images/@atf-SEQ': Failed to read ELF file: Python: No module named 'elftools'
make[4]: *** [Makefile:1108: .binman_stamp] Error 1
make[4]: Leaving directory '/home/rshi0212/r68s_lede/build_dir/target-aarch64_generic_musl/u-boot-r66s-rk3568/u-boot-2023.04'
make[3]: *** [Makefile:334: /home/rshi0212/r68s_lede/build_dir/target-aarch64_generic_musl/u-boot-r66s-rk3568/u-boot-2023.04/.built] Error 2
make[3]: Leaving directory '/home/rshi0212/r68s_lede/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/r66s-rk3568/compile#1.47#0.17#1.69
    ERROR: package/boot/uboot-rockchip failed to build (build variant: r66s-rk3568).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/home/rshi0212/r68s_lede'
make[1]: *** [package/Makefile:110: /home/rshi0212/r68s_lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/rshi0212/r68s_lede'
make: *** [/home/rshi0212/r68s_lede/include/toplevel.mk:231: world] Error 2

之前都ok，今天git pull以后过不去了，尝试过apt install readme和rm .config tmp/，还是不行。

rshi0212, 2023-04-12:
google了一下elftoos，sudo apt-get install python-pyelftools python3-pyelftools -y后还是不行。

closed 2023-04-13 by coolsnowwolf

a786458540, 2023-05-06:
一样的问题，你解决了吗