coolsnowwolf/lede #11104: 新码死活编不过去啊

Travel2Here, 2023-04-13:
跟其他几位类似，在uboot这部分出错。
两台不同的设备，按最新指示更新了各种依赖，且make clean了依然如此。
设备如日志所示 R4S~
求解~感谢

**部分日志：**

起初:
make[3] -C package/boot/uboot-rockchip compile
    ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopi-r4s-rk3399).
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/zo/lede/include/toplevel.mk:231: world] Error 1

继续 j1一段后：

make[4]: Entering directory '/home/zo/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04'
  ENVC    include/generated/env.txt
  ENVP    include/generated/env.in
  ENVT    include/generated/environment.h
make -f ./scripts/Makefile.build obj=scripts/dtc
  CC      env/common.o
  AR      env/built-in.o
  CC      lib/efi_loader/dtbdump.o
  LD      lib/efi_loader/dtbdump_efi.so
  OBJCOPY lib/efi_loader/dtbdump.efi
  CC      lib/efi_loader/initrddump.o
  LD      lib/efi_loader/initrddump_efi.so
  OBJCOPY lib/efi_loader/initrddump.efi
  LD      u-boot
  OBJCOPY u-boot.srec
  OBJCOPY u-boot-nodtb.bin
  RELOC   u-boot-nodtb.bin
  CAT     u-boot-dtb.bin
  COPY    u-boot.bin
  SYM     u-boot.sym
  MKIMAGE u-boot.img
  MKIMAGE u-boot-dtb.img
  BINMAN  .binman_stamp
Wrote map file './simple-bin.map' to show errors
binman: Node '/binman/simple-bin/fit': subnode 'images/@atf-SEQ': Failed to read ELF file: Python: No module named 'elftools'
make[4]: *** [Makefile:1108: .binman_stamp] Error 1
make[4]: Leaving directory '/home/zo/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04'
make[3]: *** [Makefile:334: /home/zo/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/.built] Error 2
make[3]: Leaving directory '/home/zo/lede/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/nanopi-r4s-rk3399/compile#1.94#0.67#2.56
    ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopi-r4s-rk3399).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/home/zo/lede'
make[1]: *** [package/Makefile:110: /home/zo/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zo/lede'
make: *** [/home/zo/lede/include/toplevel.mk:231: world] Error 2
xx@ubuntu:~/lede$

WYC-2020, 2023-04-13:
刚我还本地编译了 我这边可以过 ，依赖我在另一个issue 说过 你可以试试

Travel2Here, 2023-04-13:
> 刚我还本地编译了 我这边可以过 ，依赖我在另一个issue 说过 你可以试试

也用过，没戏。我也是奇了怪了。随后make dirclean了，现在正重新J1开始尝试。

是指的这段吧?

sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev
libz-dev patch python3 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs
git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx
libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf wget
curl swig rsync aria2 ca-certificates python3-pyelftools libpython3-dev yasm python3-setuptools

Lionzj, 2023-04-13:
我这边R4S本地编译也失败了，报的错也是一样：
binman: Node '/binman/simple-bin/fit': subnode 'images/@atf-SEQ': Failed to read ELF file: Python: No module named 'elftools'
依赖也是更新过的，python3-pyelftools肯定是有的。就很怪。。。

Travel2Here, 2023-04-13:
> 我这边R4S本地编译也失败了，报的错也是一样： binman: Node '/binman/simple-bin/fit': subnode 'images/@atf-SEQ': Failed to read ELF file: Python: No module named 'elftools' 依赖也是更新过的，python3-pyelftools肯定是有的。就很怪。。。

不容易，经历俩小时，我单线程的成了。

[image]

kenvix, 2023-11-07:
同样遇到此问题，指出一点：如果当前处于Anaconda环境，需要为conda单独安装pyelftools