coolsnowwolf/lede #11109: 还是昨天的问题，修改无线配置会死机

decike, 2023-04-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
修改无限配置会死机

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
有 #11105 但问题并未解决

## 2. 详细叙述
首先感谢L大昨天的更新。问题跟昨天一样，昨天修复firewall的更新似乎没有起作用。但是我有了新发现，修改配置之后会死机其实不完全死机，可以连接wifi接着操作，这样就可以获得系统日志，为了确保不是本地编译环境的问题，我还是本地编译一份固件云编译一份，问题依旧，但是这次我获得了比较完整的日志文件，恕本人愚钝看不懂日志文件，故再提一次issue

A：

### (2) 路由器型号和固件版本
#11105
mt6728an
xiaom-router-4c
A：

### (3) 详细日志
以下两份文件分别是云编译和本地wsl编译的固件获得的日志
[yun.log](https://github.com/coolsnowwolf/lede/files/11227861/yun.log)
[wsl.log](https://github.com/coolsnowwolf/lede/files/11227863/wsl.log)

A：
不胜感激

decike, 2023-04-14:
下面是新固件/etc/config目录下的一些配置文件
[nlbwon.txt](https://github.com/coolsnowwolf/lede/files/11229395/nlbwon.txt)
[wireless.txt](https://github.com/coolsnowwolf/lede/files/11229396/wireless.txt)
[dhcp.txt](https://github.com/coolsnowwolf/lede/files/11229397/dhcp.txt)
[firewall.txt](https://github.com/coolsnowwolf/lede/files/11229398/firewall.txt)
[network.txt](https://github.com/coolsnowwolf/lede/files/11229399/network.txt)
出bug之后感觉是dhcp没了，因为lan口接电脑查ip地址会变成169.254.12.26，dns服务器和dhcp服务器都是空，手动设置一个同网段IP还有dns也还是无法连接路由器，不知道为什么无线连接有时候可以连上
再次感谢

decike, 2023-04-14:
对了，新编译出来的固件版本和内核版本信息如下
固件版本 OpenWrt R23.4.1 / LuCI Master (git-23.102.51528-077696d)
内核版本 5.4.240

coolsnowwolf, 2026-09-05:
该 issue 针对 2023 年小米 4C、OpenWrt 5.4.240，正文只引用外部日志附件，缺少可核对的日志内容、无线配置和当前源码复现条件；相关内核与无线栈也已变化，无法形成安全的通用修复。

closed 2026-09-05 by coolsnowwolf