coolsnowwolf/lede #11115: 编译报错

zycdy, 2023-04-15:
反馈bug/问题模板，提建议请删除

## 1.按说明编译报错 添加了kenzok8的插件
sed -i '$a src-git kenzo https://github.com/kenzok8/openwrt-packages' feeds.conf.default
sed -i '$a src-git small https://github.com/kenzok8/small' feeds.conf.default

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
arm_cortex-a53  ip60xx 平台  添加了kenzok8的插件 编译报错

### (1) 具体问题

A：每次编译到 Signing package index..   xray-core
这就报这错  make menuconfig中取消选中  xray-core 也报这错误  但有输出固件

### (2) 路由器型号和固件版本

A：GL-AX1800  

### (3) 详细日志

A：Generating index for package ./v2ray-plugin_5.4.0-1_aarch64_cortex-a53.ipk
Generating index for package ./xray-core_1.8.0-1_aarch64_cortex-a53.ipk
Signing package index...
bash: line 4: /home/u/lede/staging_dir/host/bin/usign: No such file or directory
bash: line 4: /home/u/lede/staging_dir/host/bin/usign: No such file or directory
bash: line 4: /home/u/lede/staging_dir/host/bin/usign: No such file or directory
bash: line 4: /home/u/lede/staging_dir/host/bin/usign: No such file or directory
bash: line 4: /home/u/lede/staging_dir/host/bin/usign: No such file or directory
bash: line 4: /home/u/lede/staging_dir/host/bin/usign: No such file or directory
bash: line 4: /home/u/lede/staging_dir/host/bin/usign: No such file or directory
bash: line 4: /home/u/lede/staging_dir/host/bin/usign: No such file or directory
make[2]: *** [package/Makefile:88: package/index] Error 127
make[2]: Leaving directory '/home/u/lede'
make[1]: *** [Makefile:129: world] Error 2
make[1]: Leaving directory '/home/u/lede'
make: *** [/home/u/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2023-04-16:
自己签名的这个程序没有，加上编译呗，我记得没错这玩意在host编译的

zycdy, 2023-04-16:
好的 谢谢

closed 2023-04-16 by zycdy