coolsnowwolf/lede #11119: RK系列芯片卡刷以后均无法引导启动

VIKINGYFY, 2023-04-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
RK系列芯片均无法引导启动，怀疑是这个[提交](https://github.com/coolsnowwolf/lede/commit/bb084c1d09e7b81ba4eda974ab0d7abcd6e3a114)还未撤回导致的。
从这个[提交](https://github.com/coolsnowwolf/lede/commit/e9553e8101981f7283bc34dfe0244320c44fbef7)开始，RK系列就无法编译通过，最近撤回了2个相关改动后能稳定编译通过，但还是卡刷后无法引导启动。

### (2) 路由器型号和固件版本

A：
R4S、R5C、R66S、R68S

### (3) 详细日志

A：

BenjaminX, 2023-04-17:
同样的问题

BenjaminX, 2023-04-17:
这个 patch 修改量不小，感觉需要具体适配检查花不少功夫了。

VIKINGYFY, 2023-04-17:
> 这个 patch 修改量不小，感觉需要具体适配检查花不少功夫了。

我觉得直接撤回就行了呗，还省事😂

Travel2Here, 2023-04-17:
不少人同样情况来反馈，希望共建者们给点回复，共同交流发现、解决问题~

WYC-2020, 2023-04-18:
怀疑什么直接撤回 自己编译了再试就知道，确定了撤回某个就好了 直接告知提交这个commit的人就行，或者直接pr

BenjaminX, 2023-04-19:
回滚修复了

hyy-666, 2023-04-20:
现在最新的源码应该没有问题了吧？

closed 2023-05-06 by aiamadeus