coolsnowwolf/lede #11124: 更新源码后编译的版本上不了网

mitsukileung, 2023-04-17:
IPQ807X / Ax3600

Mon Apr 17 23:44:45 2023 daemon.notice netifd: Interface 'wan' is now down
Mon Apr 17 23:44:45 2023 daemon.notice netifd: Interface 'wan' is setting up now
Mon Apr 17 23:44:45 2023 daemon.notice netifd: wan (21959): uci: Entry not found
Mon Apr 17 23:44:45 2023 daemon.notice netifd: wan (21959): sh: out of range
Mon Apr 17 23:44:45 2023 daemon.info pppd[21990]: Plugin pppoe.so loaded.
Mon Apr 17 23:44:45 2023 daemon.info pppd[21990]: PPPoE plugin from pppd 2.4.9
Mon Apr 17 23:44:45 2023 daemon.notice pppd[21990]: pppd 2.4.9 started by root, uid 0
Mon Apr 17 23:44:55 2023 user.notice shadowsocksr: ssrplus server error.restart!
Mon Apr 17 23:45:00 2023 daemon.warn pppd[21990]: Timeout waiting for PADO packets
Mon Apr 17 23:45:00 2023 daemon.err pppd[21990]: Unable to complete PPPoE Discovery
Mon Apr 17 23:45:00 2023 daemon.info pppd[21990]: Exit.

WYC-2020, 2023-04-18:
拨号都失败了 能上网才有鬼，确定网口配置 是wan口，用的虚拟机别看标注的eth1,直接ifconfig看mac 然后在虚拟机配置看 改mac是否就是wan,实体机的话 直接吧所有非必要的全部拔掉，只保留wan口 ifconfig 看有流量那个

mitsukileung, 2023-04-18:
> 拨号都失败了 能上网才有鬼，确定网口配置 是wan口，用的虚拟机别看标注的eth1,直接ifconfig看mac 然后在虚拟机配置看 改mac是否就是wan,实体机的话 直接吧所有非必要的全部拔掉，只保留wan口 ifconfig 看有流量那个

用Files大法编译，之前都是好好的。今天编译出来就是这样

potoo0, 2023-04-19:
换下网口试试？我在 四月一 编译的网口顺序是反的，界面上看到的 eth1 其实是物理的 eth0

closed 2023-04-22 by mitsukileung