coolsnowwolf/lede #11127: Hinlink H68K 编译uboot失败

boss420111, 2023-04-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
编译Hinlink H68K失败，已更新到最新的代码

### (1) 具体问题

A：代码更新到最新：
4bba40608bd2b9c05ee33cff62277a4efe7ebda7
Author: Husky <57535444+hyy-666@users.noreply.github.com>
Date:   Mon Apr 17 23:38:51 2023 +0800

    uboot-rockchip: revert use of standard boot (#11121)
空编失败。
看起来是报错一个Python.h文件找不到？是否要在Language里选个Python？

### (2) 路由器型号和固件版本

A：Hinlink H68K

### (3) 详细日志

checking for external symbol _system_configuration... config.status: creating src/Makefile
mv -f .deps/pam_unix_auth.Tpo .deps/pam_unix_auth.Plo
/bin/bash ../../libtool  --tag=CC   --mode=compile aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../..   -I/home/msroot/openwrt/lean-lede/lean-lede/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/msroot/openwrt/lean-lede/lean-lede/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/msroot/openwrt/lean-lede/lean-lede/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include  -I../../libpam/include -I../../libpamc/include -DCHKPWD_HELPER=\"/usr/sbin/unix_chkpwd\" -DUPDATE_HELPER=\"/usr/sbin/unix_update\" -W -Wall -Wbad-function-cast -Wcast-align -Wcast-align=strict -Wcast-qual -Wdeprecated -Winline -Wmain -Wmissing-declarations -Wmissing-format-attribute -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wshadow -Wstrict-prototypes -Wuninitialized -Wwrite-strings -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/msroot/openwrt/lean-lede/lean-lede/lede/build_dir/target-aarch64_generic_musl/Linux-PAM-1.5.2=Linux-PAM-1.5.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -MT pam_unix_passwd.lo -MD -MP -MF .deps/pam_unix_passwd.Tpo -c -o pam_unix_passwd.lo pam_unix_passwd.c
rstrip.sh: /home/msroot/openwrt/lean-lede/lean-lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-slhc/lib/modules/5.15.104/slhc.ko: relocatable
Checking for function "[1mmkostemp[0m" : [1;32mYES[0m 
no
checking for __fsetlocking... 
[11/119] Building C object dbus/CMakeFiles/dbus-1.dir/dbus-marshal-validate.c.o[K
[11/119] Building C object dbus/CMakeFiles/dbus-1.dir/dbus-message.c.o[KOpenWrt-libtool: compile:  aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../.. -I/home/msroot/openwrt/lean-lede/lean-lede/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/msroot/openwrt/lean-lede/lean-lede/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/msroot/openwrt/lean-lede/lean-lede/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include -I../../libpam/include -I../../libpamc/include -DCHKPWD_HELPER=\"/usr/sbin/unix_chkpwd\" -DUPDATE_HELPER=\"/usr/sbin/unix_update\" -W -Wall -Wbad-function-cast -Wcast-align -Wcast-align=strict -Wcast-qual -Wdeprecated -Winline -Wmain -Wmissing-declarations -Wmissing-format-attribute -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wshadow -Wstrict-prototypes -Wuninitialized -Wwrite-strings -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/msroot/openwrt/lean-lede/lean-lede/lede/build_dir/target-aarch64_generic_musl/Linux-PAM-1.5.2=Linux-PAM-1.5.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -MT pam_unix_passwd.lo -MD -MP -MF .deps/pam_unix_passwd.Tpo -c pam_unix_passwd.c  -fPIC -DPIC -o .libs/pam_unix_passwd.o
aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -DBUILDING_NCURSES -I../ncurses -I. -I../include -I/home/msroot/openwrt/lean-lede/lean-lede/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/msroot/openwrt/lean-lede/lean-lede/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/msroot/openwrt/lean-lede/lean-lede/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include  -D_FORTIFY_SOURCE=1 -DPIC -DNDEBUG -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/msroot/openwrt/lean-lede/lean-lede/lede/build_dir/target-aarch64_generic_musl/ncurses-6.4=ncurses-6.4 -Wformat -Werror=format-security -fstack-protector -Wl,-z,now -Wl,-z,relro -fPIC --param max-inline-insns-single=1200  -fPIC -c ../ncurses/./base/lib_pad.c -o ../obj_s/lib_pad.o
Checking if "[1mfutex(2) system call[0m" : links: [1;32mYES[0m 
mv -f .deps/ilmid.Tpo .deps/ilmid.Po
aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../.. -I./../ilmid/asn1  -I/home/msroot/openwrt/lean-lede/lean-lede/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/msroot/openwrt/lean-lede/lean-lede/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/msroot/openwrt/lean-lede/lean-lede/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include   -I../../src/include -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/msroot/openwrt/lean-lede/lean-lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-atm-2.5.2=linux-atm-2.5.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -Wall -Wshadow -Wpointer-arith -Wwrite-strings -Wstrict-prototypes -MT wait.o -MD -MP -MF .deps/wait.Tpo -c -o wait.o wait.c
[01m[Kscripts/dtc/pylibfdt/libfdt_wrap.c:157:11:[m[K [01;31m[Kfatal error: [m[KPython.h: No such file or directory
  157 | # include [01;31m[K<Python.h>[m[K
      |           [01;31m[K^~~~~~~~~~[m[K
compilation terminated.
yes
error: command 'gcc' failed with exit status 1
config.status: creating po/Makefile.in
checking for canonicalize_file_name... Checking if "[1meventfd(2) system call[0m" : links: [1;32mYES[0m 
make[6]: *** [scripts/dtc/pylibfdt/Makefile:33: rebuild] Error 1
make[5]: *** [scripts/Makefile.build:397: scripts/dtc/pylibfdt] Error 2
make[4]: *** [Makefile:2003: scripts_dtc] Error 2
make[4]: Leaving directory '/home/msroot/openwrt/lean-lede/lean-lede/lede/build_dir/target-aarch64_generic_musl/u-boot-opc-h68k-rk3568/u-boot-2023.04'
make[3]: *** [Makefile:313: /home/msroot/openwrt/lean-lede/lean-lede/lede/build_dir/target-aarch64_generic_musl/u-boot-opc-h68k-rk3568/u-boot-2023.04/.built] Error 2
make[3]: Leaving directory '/home/msroot/openwrt/lean-lede/lean-lede/lede/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/opc-h68k-rk3568/compile#13.01#6.30#23.02
    ERROR: package/boot/uboot-rockchip failed to build (build variant: opc-h68k-rk3568).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: *** Waiting for unfinished jobs....

WYC-2020, 2023-04-18:
更新基础环境的依赖

boss420111, 2023-04-19:
更新依赖后解决，感谢

closed 2023-04-19 by boss420111