coolsnowwolf/lede #11129: 编译时出错 gcc failed to build.

azwhikaru, 2023-04-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译时报错 gcc failed to build，已安装 README 和 Issues 里提到的依赖
```
make[6]: *** [Makefile:2217: i386.o] Error 1
make[6]: Leaving directory '/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/host-x86_64-openwrt-linux-musl/gcc'
make[5]: *** [Makefile:4236: all-gcc] Error 2
make[5]: Leaving directory '/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0'
make[4]: *** [Makefile:901: all] Error 2
make[4]: Leaving directory '/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0'
make[3]: *** [Makefile:226: /home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/.built] Error 2
make[3]: Leaving directory '/home/user/lede/feeds/packages/devel/gcc'
time: package/feeds/packages/gcc/compile#489.82#54.02#540.68
    ERROR: package/feeds/packages/gcc failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/gcc/compile] Error 1
make[2]: Leaving directory '/home/user/lede'
make[1]: *** [package/Makefile:110: /home/user/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/lede'
make: *** [/home/user/lede/include/toplevel.mk:231：world] 错误 2
```

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：单线程 log
``` 
from ../.././gcc/tree.c:31:
../.././gcc/tree.c: In function 'tree_node* get_file_function_name(const char*)':
../.././gcc/../include/libiberty.h:727:33: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
  727 |    char *const libiberty_nptr = (char *const) alloca (libiberty_len); \
      |                                 ^
../.././gcc/tree.c:9757:13: note: in expansion of macro 'ASTRDUP'
 9757 |     p = q = ASTRDUP (first_global_object_name);
      |             ^~~~~~~
../.././gcc/../include/libiberty.h:727:33: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
  727 |    char *const libiberty_nptr = (char *const) alloca (libiberty_len); \
      |                                 ^
../.././gcc/tree.c:9772:15: note: in expansion of macro 'ASTRDUP'
 9772 |       p = q = ASTRDUP (lbasename (file));
      |               ^~~~~~~
x86_64-openwrt-linux-musl-g++ -fno-PIE -c   -g -O2 -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../.././gcc -I../.././gcc/. -I../.././gcc/../include -I../.././gcc/../libcpp/include -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/host-x86_64-openwrt-linux-musl/gmp -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/gmp -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/host-x86_64-openwrt-linux-musl/mpfr/src -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/mpfr/src -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/mpc/src  -I../.././gcc/../libdecnumber -I../.././gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/../libbacktrace   -o typed-splay-tree.o -MT typed-splay-tree.o -MMD -MP -MF ./.deps/typed-splay-tree.TPo ../.././gcc/typed-splay-tree.c
x86_64-openwrt-linux-musl-g++ -fno-PIE -c   -g -O2 -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../.././gcc -I../.././gcc/. -I../.././gcc/../include -I../.././gcc/../libcpp/include -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/host-x86_64-openwrt-linux-musl/gmp -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/gmp -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/host-x86_64-openwrt-linux-musl/mpfr/src -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/mpfr/src -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/mpc/src  -I../.././gcc/../libdecnumber -I../.././gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/../libbacktrace   -o valtrack.o -MT valtrack.o -MMD -MP -MF ./.deps/valtrack.TPo ../.././gcc/valtrack.c
x86_64-openwrt-linux-musl-g++ -fno-PIE -c   -g -O2 -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../.././gcc -I../.././gcc/. -I../.././gcc/../include -I../.././gcc/../libcpp/include -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/host-x86_64-openwrt-linux-musl/gmp -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/gmp -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/host-x86_64-openwrt-linux-musl/mpfr/src -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/mpfr/src -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/mpc/src  -I../.././gcc/../libdecnumber -I../.././gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/../libbacktrace   -o value-prof.o -MT value-prof.o -MMD -MP -MF ./.deps/value-prof.TPo ../.././gcc/value-prof.c
x86_64-openwrt-linux-musl-g++ -fno-PIE -c   -g -O2 -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../.././gcc -I../.././gcc/. -I../.././gcc/../include -I../.././gcc/../libcpp/include -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/host-x86_64-openwrt-linux-musl/gmp -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/gmp -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/host-x86_64-openwrt-linux-musl/mpfr/src -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/mpfr/src -I/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/mpc/src  -I../.././gcc/../libdecnumber -I../.././gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/../libbacktrace   -o var-tracking.o -MT var-tracking.o -MMD -MP -MF ./.deps/var-tracking.TPo ../.././gcc/var-tracking.c
In file included from ../.././gcc/var-tracking.c:103:
../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx) const':
../.././gcc/recog.h:300:51: warning: cast between incompatible function types from 'insn_gen_fn::stored_funcptr' {aka 'rtx_insn* (*)()'} to 'insn_gen_fn::f1' {aka 'rtx_insn* (*)(rtx_def*)'} [-Wcast-function-type]
  300 |   rtx_insn * operator () (rtx a0) const { return ((f1)func) (a0); }
      |                                                   ^~~~~~~~
../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx, rtx) const':
../.././gcc/recog.h:301:59: warning: cast between incompatible function types from 'insn_gen_fn::stored_funcptr' {aka 'rtx_insn* (*)()'} to 'insn_gen_fn::f2' {aka 'rtx_insn* (*)(rtx_def*, rtx_def*)'} [-Wcast-function-type]
  301 |   rtx_insn * operator () (rtx a0, rtx a1) const { return ((f2)func) (a0, a1); }
      |                                                           ^~~~~~~~
../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx, rtx, rtx) const':
../.././gcc/recog.h:302:67: warning: cast between incompatible function types from 'insn_gen_fn::stored_funcptr' {aka 'rtx_insn* (*)()'} to 'insn_gen_fn::f3' {aka 'rtx_insn* (*)(rtx_def*, rtx_def*, rtx_def*)'} [-Wcast-function-type]
  302 |   rtx_insn * operator () (rtx a0, rtx a1, rtx a2) const { return ((f3)func) (a0, a1, a2); }
      |                                                                   ^~~~~~~~
../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx, rtx, rtx, rtx) const':
[... 225 lines cut ...]
 9625 |   bool error_p = NULL;
      |                  ^~~~
../.././gcc/config/i386/i386.c: In function 'rtx_def* ix86_expand_special_args_builtin(const builtin_description*, tree, rtx)':
../.././gcc/config/i386/i386.c:37227:7: warning: this statement may fall through [-Wimplicit-fallthrough=]
37227 |       switch (icode)
      |       ^~~~~~
../.././gcc/config/i386/i386.c:37254:5: note: here
37254 |     case V64QI_FTYPE_PCCHAR_V64QI_UDI:
      |     ^~~~
make[6]: *** [Makefile:2217: i386.o] Error 1
make[6]: Leaving directory '/home/user/lede/build_dir/target-x86_64_musl/gcc-7.4.0/host-x86_64-openwrt-linux-musl/gcc'
make[5]: *** [Makefile:4236: all-gcc] Error 2
[... cut ...]
make[3]: Leaving directory '/home/user/lede/feeds/packages/devel/gcc'
time: package/feeds/packages/gcc/compile#489.82#54.02#540.68
    ERROR: package/feeds/packages/gcc failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/gcc/compile] Error 1
make[2]: Leaving directory '/home/user/lede'
make[1]: *** [package/Makefile:110: /home/user/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/lede'
make: *** [/home/user/lede/include/toplevel.mk:231：world] 错误 2
```

WYC-2020, 2023-04-18:
指针类型不能转换成bool改成用false,而且l已经去掉了gcc 7.4你这个自己修下呗，以前我修过了

azwhikaru, 2023-04-19:
添加环境变量后成功
```shell
export GO111MODULE on
export GOPROXY https://goproxy.cn,direct
```

closed 2023-04-19 by azwhikaru

WYC-2020, 2023-04-19:
你逗我添加这个环境变量是不可能解决这个错误的 
[image]
反正你成功就行，只是说明 这个不是解决这个问题的方法，并且也不可能靠这个解决