coolsnowwolf/lede #11131: 已安装mt7921e驱动但无法使用

zzh151223, 2023-04-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最新版6.1.22已安装mt7921e驱动，但无线里面所有东西都是空的
[image]
这个是自动生成的无线配置
[image]
这个是我手动根据pci设备写入的配置
[image]
[image]
都无法使用

### (2) 路由器型号和固件版本

A：x86_6.1.24

### (3) 详细日志

A：

aiamadeus, 2023-04-19:
很明显你没有选 wpad-openssl

zzh151223, 2023-04-19:
> 很明显你没有选 wpad-openssl

新版本我不知道，老版本就算没装也能选模式、频宽什么的，也能搜索信号，只是连不上而已，现在是什么都搞不了，新版本改了？

closed 2023-04-20 by aiamadeus