coolsnowwolf/lede #11132: mwan3 helper导致dhcp失效

iam993, 2023-04-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

但凡开启mwan3 helper的情况下，端口将无法正常获取IP，在获取ip的时候关闭mwan3 helper，马上就可以获取到。

### (2) 路由器型号和固件版本

源码自编译的，x86 ，OpenWrt R23.4.1 / LuCI Master (git-23.104.57901-bf5e9ad)

### (3) 详细日志

日志没有太多相关的内容，但是看了代码，似乎和dnmasq有关，导致服务没启动

`start(){
  /etc/mwan3helper/genipset.sh cn '/etc/mwan3helper/all_cn.txt'
  /etc/mwan3helper/genipset.sh ct '/etc/mwan3helper/chinatelecom.txt'
  /etc/mwan3helper/genipset.sh cnc '/etc/mwan3helper/unicom_cnc.txt'
  /etc/mwan3helper/genipset.sh cmcc '/etc/mwan3helper/cmcc.txt'
  /etc/mwan3helper/genipset.sh crtc '/etc/mwan3helper/crtc.txt'
  /etc/mwan3helper/genipset.sh cernet '/etc/mwan3helper/cernet.txt'
  /etc/mwan3helper/genipset.sh gwbn '/etc/mwan3helper/gwbn.txt'
  /etc/mwan3helper/genipset.sh othernet '/etc/mwan3helper/othernet.txt'
  en=$(uci get mwan3helper.config.enabled 2>/dev/null)
  if [ $en -eq 1 ];  then
    start_pdnsd
    mkdir -p /tmp/etc/dnsmasq-mwan3.d
    ipset -N gfwlist hash:net 2>/dev/null
    awk '!/^$/&&!/^#/{printf("ipset=/.%s/'"gfwlist"'\n",$0)}' /etc/mwan3helper/gfw.txt > /tmp/etc/dnsmasq-mwan3.d/mwan3_gfwlist.conf
    awk '!/^$/&&!/^#/{printf("server=/.%s/'"127.0.0.1#53000"'\n",$0)}' /etc/mwan3helper/gfw.txt >> /tmp/etc/dnsmasq-mwan3.d/mwan3_gfwlist.conf
    cat > /tmp/dnsmasq.d/dnsmasq-mwan3.conf <<EOF
conf-dir=/var/etc/dnsmasq-mwan3.d
EOF
    /etc/init.d/dnsmasq restart
  fi
}`

部分日志
Wed Apr 19 18:15:13 2023 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-mwan3.d: No such file or directory
Wed Apr 19 18:15:13 2023 daemon.crit dnsmasq[1]: FAILED to start up

WYC-2020, 2023-04-21:
我这边运行正常，你还是看看为啥这个目录为啥没有创建成功吧，原则上你开启了这个
[image]
就会在/var/etc 生成dnsmasq-mwan3.d 目录的

iam993, 2023-04-22:
> 我这边运行正常，你还是看看为啥这个目录为啥没有创建成功吧，原则上你开启了这个 [image] 就会在/var/etc 生成dnsmasq-mwan3.d 目录的
目录创建成功了，但就是说找不到，我改了脚本，换成tmp目录也不行。不知道是不是用U盘引导导致的。

cskeleton, 2023-11-08:
参考这里 https://www.right.com.cn/forum/thread-8296669-1-1.html

coolsnowwolf, 2026-09-05:
日志已直接显示根因：mwan3helper 脚本写入 `/tmp/etc/dnsmasq-mwan3.d`，但生成的 dnsmasq 配置使用 `conf-dir=/var/etc/dnsmasq-mwan3.d`，目录不存在导致 dnsmasq 启动失败，进而 DHCP 失效。该 `mwan3helper` 脚本不属于当前 LEDE/OpenWrt 主线（仓库中无对应包或源码），不能提交到 LEDE 修复；应在第三方 helper 中统一目录路径（并在重启前创建目录），或移除该组件。关闭该第三方脚本问题。

closed 2026-09-05 by coolsnowwolf