coolsnowwolf/lede #11133: X86固件最近很长一段时间，一直编译不成功，报错：openssl-qat failed to build.

guoshh1978, 2023-04-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
最近很长一段时间，一直编译不成功，报错：openssl-qat failed to build.
### (2) 路由器型号和固件版本

A：
X86-64固件
### (3) 详细日志

A：
configure: error: qatlib library not installed or --with-qat_hw_dir or --enable-qat_sw not set, If you like to use QAT Hardware acceleration, set it to the path to the Intel Quickassist Technology Driver source code. e.g: --with-qat_hw_dir=/path/to/qat_driver or If you like to use QAT Software acceleration using Intel crypto multibuffer library and Intel IPSec (AES-GCM),  provide --enable-qat_sw after installing crypto_mb and IPSec_mb libraries as per README
make[3]: *** [Makefile:88: /home/guoshh/lede/build_dir/target-x86_64_musl/QAT_Engine-0.6.12/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/guoshh/lede/package/qat/libs/openssl-qat'
time: package/qat/libs/openssl-qat/compile#6.99#0.60#8.72
    ERROR: package/qat/libs/openssl-qat failed to build.
make[2]: *** [package/Makefile:116: package/qat/libs/openssl-qat/compile] Error 1
make[2]: Leaving directory '/home/guoshh/lede'
make[1]: *** [package/Makefile:110: /home/guoshh/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/guoshh/lede'
make: *** [/home/guoshh/lede/include/toplevel.mk:231: world] Error 2
guoshh@mail:~/lede$ make menuconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

WYC-2020, 2023-04-20:
等会给你看下

WYC-2020, 2023-04-20:
那是因为默认配置是没有选中硬件加速的，如果没有硬件加速sw这个选项也不能启用，所以这样改下即可，但是默认不选择任何硬件加速编译这玩意不知道有啥用，暂时这样改吧
[image]
要改默认选择改这里即可
[image]

guoshh1978, 2023-04-20:
好的，我试一下，谢谢大佬

guoshh1978, 2023-04-20:
[image]
好像不能直接禁掉硬件加速，禁掉了就直接编译失败了。请问启用的话，应该是用C2XXX还是C3XXX？谢谢！

guoshh1978, 2023-04-20:
menu "Configuration"
        depends on TARGET_x86_64

        choice
                prompt "Platform"
                **_default QAT_C2XXX_**
                #default QAT_NONE
                config QAT_NONE
                        bool "None (Default)"
                config QAT_C2XXX
                        bool "c2xxx - Rangeley"
                        depends on LINUX_4_14
                        select PACKAGE_quickassist-c2xxx
                        select PACKAGE_kmod-crypto-qat-c2xxx
                        select PACKAGE_kmod-crypto-qat-c2xxx-usdm
                config QAT_C3XXX
                        bool "c3xxx - Denverton"
                        select PACKAGE_quickassist-c3xxx
                        select PACKAGE_kmod-crypto-qat-c3xxx
                        select PACKAGE_kmod-crypto-qat-common
                        select PACKAGE_kmod-crypto-qat-usdm

修改成2XXX的3XXX貌似都不行，都是报一样的错，说CONFIG_CRYPTO_HW没有启用。

guoshh1978, 2023-04-20:
make[4]: Entering directory '/home/guoshh/lede/build_dir/target-x86_64_musl/linux-x86_64/quickassist-c3xxx-1.7/quickassist/qat'
Makefile:48: *** CONFIG_CRYPTO_HW not enabled in kernel configuration.  Stop.
make[4]: Leaving directory '/home/guoshh/lede/build_dir/target-x86_64_musl/linux-x86_64/quickassist-c3xxx-1.7/quickassist/qat'
make[3]: *** [Makefile:236: /home/guoshh/lede/build_dir/target-x86_64_musl/linux-x86_64/quickassist-c3xxx-1.7/.built] Error 2
make[3]: Leaving directory '/home/guoshh/lede/package/qat/firmware/quickassist-c3xxx'
time: package/qat/firmware/quickassist-c3xxx/compile#0.15#0.01#0.16
    ERROR: package/qat/firmware/quickassist-c3xxx failed to build.
make[2]: *** [package/Makefile:116: package/qat/firmware/quickassist-c3xxx/compile] Error 1
make[2]: Leaving directory '/home/guoshh/lede'
make[1]: *** [package/Makefile:110: /home/guoshh/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/guoshh/lede'
make: *** [/home/guoshh/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2023-04-20:
> [image] 好像不能直接禁掉硬件加速，禁掉了就直接编译失败了。请问启用的话，应该是用C2XXX还是C3XXX？谢谢！

我晕 你选择了c3xx 你倒是编译一下内核哈真是

guoshh1978, 2023-04-20:
不好意思，水限不行，让大神见笑了：（，我执行了make dirclean，以为会自动全部重新编译

closed 2023-04-21 by guoshh1978