coolsnowwolf/lede #11134: 用 0412 的 git（bb084c1d0）编译R2S固件，编译python3报错，上一个git（88989f272）正常

szdosar, 2023-04-20:
用0412 的 git：uboot-rockchip: bump to v2023.04 编译R2S固件，编译选择 Languages  --->Python  ---><*> python3结果失败；
同一天的 git：arm-trusted-firmware-rockchip: update to release 2.8 编译R2S固件，编译选择 Languages  --->Python  ---><*> python3结果正常。

## 1.我的编译环境
操作系统: Ubuntu 22.04.2 LTS  内核版本: 5.15.0  CPU架构 : amd64v4  虚拟化类型: kvm

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细编译指令
```
cd
rm -rf lede
sudo -E apt-get -qq install $(curl -fsSL raw.githubusercontent.com/szdosar/ish/master/depends-ubuntu-2204)
sudo -E apt-get -qq autoremove --purge
sudo -E apt-get -qq clean
git clone https://github.com/coolsnowwolf/lede
cd lede
git reset --hard bb084c1d09e7b81ba4eda974ab0d7abcd6e3a114
sed -i '$a src-git lienol https://github.com/Lienol/openwrt-package' feeds.conf.default
sed -i '$a src-git helloworld https://github.com/fw876/helloworld.git' feeds.conf.default
sed -i '$a src-git kenzo https://github.com/kenzok8/openwrt-packages' feeds.conf.default
sed -i '$a src-git small https://github.com/kenzok8/small' feeds.conf.default
sed -i 's/192.168.1.1/192.168.4.1/g' package/base-files/files/bin/config_generate
./scripts/feeds update -a && ./scripts/feeds install -a && make defconfig
rm .config
wget https://raw.githubusercontent.com/szdosar/Actions-OpenWrt-r2s/main/.config
./scripts/feeds update -a -f && ./scripts/feeds install -a -f && make defconfig
make download -j8 && make V=s -j$(nproc)
```

### (1) 具体报错
```
Wrote map file './simple-bin.map' to show errors
binman: Node '/binman/simple-bin/fit': subnode 'images/@atf-SEQ': Failed to read ELF file: Python: No module named 'elftools'
make[4]: *** [Makefile:1108: .binman_stamp] Error 1
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.04'
make[3]: *** [Makefile:334: /workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.04/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/nanopi-r2s-rk3328/compile#3.10#0.51#3.57
    ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopi-r2s-rk3328).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
```

### (2) 路由器型号和固件版本
友善R2S

同样环境和配置，正确编译了同一天的 git 如下：
```
git reset --hard 88989f2722124de846334028320c54c562c67e46
HEAD is now at 88989f272 arm-trusted-firmware-rockchip: update to release 2.8
```

d975xbx, 2023-04-20:
一样的错误

Wrote map file './simple-bin.map' to show errors
binman: Node '/binman/simple-bin/fit': subnode 'images/@atf-SEQ': Failed to read ELF file: Python: No module named 'elftools'
make[4]: *** [Makefile:1108: .binman_stamp] Error 1
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.04'
make[3]: *** [Makefile:334: /workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.04/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/nanopi-r2s-rk3328/compile#3.10#0.51#3.57
    ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopi-r2s-rk3328).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

hyy-666, 2023-04-20:
有没有试一下最新的源码？只要根据readme配好环境，外加我从immortalwrt偷过来的两个提交应该没有问题了才是，readme里的依赖有更新

szdosar, 2023-04-20:
> 有没有试一下最新的源码？只要根据readme配好环境，外加我从immortalwrt偷过来的两个提交应该没有问题了才是，readme里的依赖有更新

谢谢你提醒了我，我比对了一下依赖：
1.我的旧依赖中的aria2，在新依赖中被删了；
2.我的旧依赖中的python3-pip，在新依赖中被替换成了python3-pyelftools。
目前正在用最新依赖和代码测试。有结果在回来反馈。

szdosar, 2023-04-20:
> 

成功了。再次感谢@hyy-666相助。

closed 2023-04-21 by aiamadeus