coolsnowwolf/lede #11137: macOS 编译出错 已经有段时间了 以为会修复

epenli, 2023-04-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

编译失败

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

make[2]: Entering directory '/Volumes/OpenWrt/lede'
make[3]: Entering directory '/Volumes/OpenWrt/lede/package/libs/libjson-c'
MAKEFLAGS="" /Volumes/OpenWrt/lede/staging_dir/host/bin/ninja  -j1 -C /Volumes/OpenWrt/lede/build_dir/hostpkg/json-c-0.16 
ninja: Entering directory `/Volumes/OpenWrt/lede/build_dir/hostpkg/json-c-0.16'
[1/3] Building C object CMakeFiles/json-c.dir/json_util.c.o
FAILED: CMakeFiles/json-c.dir/json_util.c.o 
/Volumes/OpenWrt/lede/staging_dir/host/bin/gcc -D_GNU_SOURCE -I/Volumes/OpenWrt/lede/build_dir/hostpkg/json-c-0.16 -O2 -I/Volumes/OpenWrt/lede/staging_dir/host/include -I/Volumes/OpenWrt/lede/staging_dir/hostpkg/include -I/Volumes/OpenWrt/lede/staging_dir/target-x86_64_musl/host/include -ffunction-sections -fdata-sections -Werror -Wall -Wcast-qual -Wno-error=deprecated-declarations -Wextra -Wwrite-strings -Wno-unused-parameter -Wstrict-prototypes -DNDEBUG -O2 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk -fPIC -D_REENTRANT -MD -MT CMakeFiles/json-c.dir/json_util.c.o -MF CMakeFiles/json-c.dir/json_util.c.o.d -o CMakeFiles/json-c.dir/json_util.c.o -c /Volumes/OpenWrt/lede/build_dir/hostpkg/json-c-0.16/json_util.c
/Volumes/OpenWrt/lede/build_dir/hostpkg/json-c-0.16/json_util.c:63:35: error: a function declaration without a prototype is deprecated in all versions of C [-Werror,-Wstrict-prototypes]
const char *json_util_get_last_err()
                                  ^
                                   void
1 error generated.
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:61: /Volumes/OpenWrt/lede/build_dir/hostpkg/json-c-0.16/.built] Error 1
make[3]: Leaving directory '/Volumes/OpenWrt/lede/package/libs/libjson-c'
time: package/libs/libjson-c/host-compile#0.42#0.60#1.29
    ERROR: package/libs/libjson-c [host] failed to build.
make[2]: *** [package/Makefile:114: package/libs/libjson-c/host/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt/lede'
make[1]: *** [package/Makefile:110: /Volumes/OpenWrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/lede'
make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2023-04-21:
[image]
[image]
加上void 试下。没那么土豪用mac编译。没法修复，盲修

Travel2Here, 2023-04-21:
哥们，实在不行Mac上装虚拟机用Linux干吧，我就这样 😅

WYC-2020, 2023-04-21:
哈哈

WYC-2020, 2023-04-21:
那我在想我是不是可以虚拟机里面装mac 来试试

raphaeltao, 2023-04-21:
我更惨。。。按照教程编译环境都配不上

epenli, 2023-04-21:
> 

记得进入bash 再make  然后环境参数 要在 bashrc 里面 不是 zsh

epenli, 2023-04-21:
> 那我在想我是不是可以虚拟机里面装mac 来试试
修改后成功通过 但是又遇到了 go 编译失败.
所以编译环境有什么需求? 
make[3]: Leaving directory '/Volumes/OpenWrt/lede/package/boot/grub2'
time: package/boot/grub2/efi/compile#88.00#76.30#205.82
make[3]: Entering directory '/Volumes/OpenWrt/lede/feeds/packages/lang/golang/golang'
( cd "/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src" ;   GOHOSTARCH="amd64" GOCACHE="/Volumes/OpenWrt/lede/tmp/go-build" GOENV=off CC="gcc" CXX="g++"  bash make.bash -v --no-banner ; )
# Building C bootstrap tool.
cmd/dist

# Building compilers and Go bootstrap tool for host, darwin/amd64.
lib9
libbio
liblink
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/liblink/asm5.c:644:22: warning: variable 'times' set but not used [-Wunused-but-set-variable]
        int m, bflag, i, v, times;
                            ^
1 warning generated.
cmd/cc
cmd/gc
cmd/6l
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/cmd/6l/../ld/dwarf.c:1479:15: warning: implicit conversion from 'int' to 'char' changes value from 156 to -100 [-Wconstant-conversion]
        block[i++] = DW_OP_call_frame_cfa;
                   ~ ^~~~~~~~~~~~~~~~~~~~
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/cmd/6l/../ld/dwarf.c:1763:21: warning: implicit conversion from 'int' to 'char' changes value from 144 to -112 [-Wconstant-conversion]
        cput(DW_CFA_offset + FAKERETURNCOLUMN);  // return address
        ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/cmd/6l/../ld/lib.h:168:13: note: expanded from macro 'cput'
        { *cbp++ = c;\
                 ~ ^
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/cmd/6l/../ld/dwarf.c:2228:23: warning: a function declaration without a prototype is deprecated in all versions of C [-Wstrict-prototypes]
dwarfaddelfsectionsyms()
                      ^
                       void
3 warnings generated.
cmd/6a
cmd/6c
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/cmd/6c/txt.c:995:28: warning: shifting a negative signed value is undefined [-Wshift-negative-value]
                                f->vconst |= (vlong)~0 << 32;
                                             ~~~~~~~~~ ^
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/cmd/6c/txt.c:1045:28: warning: shifting a negative signed value is undefined [-Wshift-negative-value]
                                f->vconst |= (vlong)~0 << 32;
                                             ~~~~~~~~~ ^
2 warnings generated.
cmd/6g
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/cmd/6g/peep.c:771:13: warning: converting the enum constant to a boolean [-Wint-in-bool-context]
                if(REGEXT && v->type <= REGEXT && v->type > exregoffset)
                          ^
1 warning generated.
runtime
errors
sync/atomic
sync
io
unicode
unicode/utf8
unicode/utf16
bytes
math
strings
strconv
bufio
sort
container/heap
encoding/base64
syscall
time
os
reflect
fmt
encoding
encoding/json
flag
path/filepath
path
io/ioutil
log
regexp/syntax
regexp
go/token
go/scanner
go/ast
go/parser
os/exec
os/signal
net/url
text/template/parse
text/template
go/doc
go/build
cmd/go
fatal error: runtime: bsdthread_register error

runtime stack:
runtime.throw(0x3ec895)
        /Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/runtime/panic.go:491 +0xad fp=0x7ff7bfefe710 sp=0x7ff7bfefe6e0
runtime.goenvs()
        /Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/runtime/os_darwin.c:104 +0x70 fp=0x7ff7bfefe728 sp=0x7ff7bfefe710
runtime.schedinit()
        /Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/runtime/proc.c:141 +0x5f fp=0x7ff7bfefe750 sp=0x7ff7bfefe728
runtime.rt0_go(0x7ff7bfefe788, 0x4, 0x7ff7bfefe788, 0x0, 0x1000, 0x4, 0x7ff7bfefeb98, 0x7ff7bfefebfb, 0x7ff7bfefec01, 0x7ff7bfefec04, ...)
        /Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/runtime/asm_amd64.s:95 +0x116 fp=0x7ff7bfefe758 sp=0x7ff7bfefe750
make[3]: *** [Makefile:402: /Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.built] Error 2
make[3]: Leaving directory '/Volumes/OpenWrt/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#33.77#12.53#23.58
    ERROR: package/feeds/packages/golang [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/golang/host/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt/lede'
make[1]: *** [package/Makefile:110: /Volumes/OpenWrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/lede'
make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:231: world] Error 2

epenli, 2023-04-22:
修复上面的问题后 编译到 go 的时候 又出问题了
make[3]: Leaving directory '/Volumes/OpenWrt/lede/package/boot/grub2'
time: package/boot/grub2/efi/compile#88.00#76.30#205.82
make[3]: Entering directory '/Volumes/OpenWrt/lede/feeds/packages/lang/golang/golang'
( cd "/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src" ; GOHOSTARCH="amd64" GOCACHE="/Volumes/OpenWrt/lede/tmp/go-build" GOENV=off CC="gcc" CXX="g++" bash make.bash -v --no-banner ; )

Building C bootstrap tool.
cmd/dist

Building compilers and Go bootstrap tool for host, darwin/amd64.
lib9
libbio
liblink
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/liblink/asm5.c:644:22: warning: variable 'times' set but not used [-Wunused-but-set-variable]
int m, bflag, i, v, times;
^
1 warning generated.
cmd/cc
cmd/gc
cmd/6l
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/cmd/6l/../ld/dwarf.c:1479:15: warning: implicit conversion from 'int' to 'char' changes value from 156 to -100 [-Wconstant-conversion]
block[i++] = DW_OP_call_frame_cfa;
~ ^~~~~~~~~~~~~~~~~~~~
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/cmd/6l/../ld/dwarf.c:1763:21: warning: implicit conversion from 'int' to 'char' changes value from 144 to -112 [-Wconstant-conversion]
cput(DW_CFA_offset + FAKERETURNCOLUMN); // return address
~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/cmd/6l/../ld/lib.h:168:13: note: expanded from macro 'cput'
{ *cbp++ = c;
~ ^
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/cmd/6l/../ld/dwarf.c:2228:23: warning: a function declaration without a prototype is deprecated in all versions of C [-Wstrict-prototypes]
dwarfaddelfsectionsyms()
^
void
3 warnings generated.
cmd/6a
cmd/6c
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/cmd/6c/txt.c:995:28: warning: shifting a negative signed value is undefined [-Wshift-negative-value]
f->vconst |= (vlong)~0 << 32;
~~~~~~~~~ ^
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/cmd/6c/txt.c:1045:28: warning: shifting a negative signed value is undefined [-Wshift-negative-value]
f->vconst |= (vlong)~0 << 32;
~~~~~~~~~ ^
2 warnings generated.
cmd/6g
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/cmd/6g/peep.c:771:13: warning: converting the enum constant to a boolean [-Wint-in-bool-context]
if(REGEXT && v->type <= REGEXT && v->type > exregoffset)
^
1 warning generated.
runtime
errors
sync/atomic
sync
io
unicode
unicode/utf8
unicode/utf16
bytes
math
strings
strconv
bufio
sort
container/heap
encoding/base64
syscall
time
os
reflect
fmt
encoding
encoding/json
flag
path/filepath
path
io/ioutil
log
regexp/syntax
regexp
go/token
go/scanner
go/ast
go/parser
os/exec
os/signal
net/url
text/template/parse
text/template
go/doc
go/build
cmd/go
fatal error: runtime: bsdthread_register error

runtime stack:
runtime.throw(0x3ec895)
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/runtime/panic.go:491 +0xad fp=0x7ff7bfefe710 sp=0x7ff7bfefe6e0
runtime.goenvs()
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/runtime/os_darwin.c:104 +0x70 fp=0x7ff7bfefe728 sp=0x7ff7bfefe710
runtime.schedinit()
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/runtime/proc.c:141 +0x5f fp=0x7ff7bfefe750 sp=0x7ff7bfefe728
runtime.rt0_go(0x7ff7bfefe788, 0x4, 0x7ff7bfefe788, 0x0, 0x1000, 0x4, 0x7ff7bfefeb98, 0x7ff7bfefebfb, 0x7ff7bfefec01, 0x7ff7bfefec04, ...)
/Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.go_bootstrap/src/runtime/asm_amd64.s:95 +0x116 fp=0x7ff7bfefe758 sp=0x7ff7bfefe750
make[3]: *** [Makefile:402: /Volumes/OpenWrt/lede/build_dir/hostpkg/go-1.20.2/.built] Error 2
make[3]: Leaving directory '/Volumes/OpenWrt/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#33.77#12.53#23.58
ERROR: package/feeds/packages/golang [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/golang/host/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt/lede'
make[1]: *** [package/Makefile:110: /Volumes/OpenWrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/lede'
make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2023-04-25:
mac 问题多，我虚拟机安装macos 试了下，在10.15的macos 下 x86 默认配置基本能过，但是如果加入我自定义要编译的库的话 upx 没法过，可以修复但是不想修，修了这个可能后续还有，然后我用了13的macos 试了默认配置都有问题，一个是我上面说的那个json问题 然后samba4 也会有问题，asn一堆函数没有，问题大大的 哈哈

raphaeltao, 2023-04-25:
我这两天试了下macos比较轻量的方法就是装multipass然后用ubuntu编，这个又快又方便。但是我M1的还碰上了一堆依赖装不上的问题，不能用22的lts因为好几个包都没有arm版本，要用20的然后手动把没有的那几个版本自己去官方源找替换。还要自己装一下go，反正坑是真不少，早知道用pve开个ubuntu直接编都比折腾mac强

WYC-2020, 2023-05-05:
经过我的不断的缝缝补补 终于在macos 上面编译成功，还让我找了一台黑苹果来编译

WYC-2020, 2023-05-05:
好几个库都需要修补，才能编译过 坑不少

DoveKi, 2023-05-11:
连openwrt主线都不推荐直接mac系统下编译 问题太多 大多数包基本没人测试维护mac下编译

WYC-2020, 2023-05-15:
> 连openwrt主线都不推荐直接mac系统下编译 问题太多 大多数包基本没人测试维护mac下编译

是的，主要闲来没事，还是功夫不负有心人，修完哈哈

coolsnowwolf, 2026-09-05:
日志显示首先是 macOS/Clang 在 `json-c` 的 `-Wstrict-prototypes -Werror` 下拒绝旧式函数声明，后续 Go 又触发 macOS runtime 的 `bsdthread_register`；评论也确认 macOS 下还会遇到 samba4、asn 与其他主机工具兼容问题。LEDE/OpenWrt 的主机编译环境面向 Linux，当前没有 macOS 主机支持承诺或可泛化的代码修复，建议在 Linux/容器/虚拟机中编译，关闭 issue。

closed 2026-09-05 by coolsnowwolf