coolsnowwolf/lede #11138: 玩客云  AT91-sama5-ama5d3编译失败

yifang5775, 2023-04-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败

### (2) 路由器型号和固件版本

A：玩客云  AT91-sama5-ama5d3

### (3) 详细日志

A：ld FLAGS
========
-nostartfiles -Map=/home/fred/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/binaries/sama5d3_xplained-nandflashboot-uboot-3.9.3.map --cref -static -T elf32-littlearm.lds --gc-sections -Ttext 0x300000

  LD        sama5d3_xplained-nandflashboot-uboot-3.9.3.elf
arm-openwrt-linux-muslgnueabi-ld: Error: unable to disambiguate: -nostartfiles (did you mean --nostartfiles ?)
Makefile:326: recipe for target '/home/fred/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/binaries/sama5d3_xplained-nandflashboot-uboot-3.9.3.bin' failed
make[4]: *** [/home/fred/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/binaries/sama5d3_xplained-nandflashboot-uboot-3.9.3.bin] Error 1
make[4]: Leaving directory '/home/fred/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3'
Makefile:142: recipe for target '/home/fred/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/.built' failed
make[3]: *** [/home/fred/lede/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/at91bootstrap-sama5d3_xplainednf_uboot/at91bootstrap-v3.9.3/.built] Error 2
make[3]: Leaving directory '/home/fred/lede/package/boot/at91bootstrap'
time: package/boot/at91bootstrap/sama5d3_xplainednf_uboot/compile#0.30#0.11#0.39
    ERROR: package/boot/at91bootstrap failed to build (build variant: sama5d3_xplainednf_uboot).
package/Makefile:114: recipe for target 'package/boot/at91bootstrap/compile' failed
make[2]: *** [package/boot/at91bootstrap/compile] Error 1
make[2]: Leaving directory '/home/fred/lede'
package/Makefile:110: recipe for target '/home/fred/lede/staging_dir/target-arm_cortex-a5+vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/fred/lede/staging_dir/target-arm_cortex-a5+vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/fred/lede'
/home/fred/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2

WYC-2020, 2023-04-21:
[image]

aiamadeus, 2023-04-21:
玩客云是 meson 啊，编个锤子 at91

yifang5775, 2023-04-21:
> 玩客云是介子啊，编个锤子at91

https://www.right.com.cn/forum/thread-7718577-1-1.html 晕晕，按恩山上的，准备用在docker上

aiamadeus, 2023-04-21:
恩山那些垃圾教程少看，都有原生支持了。

closed 2023-05-15 by aiamadeus