coolsnowwolf/lede #11142: nanopi r5s编译不成功

JackyLeeGM, 2023-04-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述
ubuntu物理机系统环境，编译nanopi r5s一直失败，最简化操作是只选择了r5s，其他luci插件默认没选也是编译失败。倒是在虚拟机debian下编译成功了几次。

### (1) 具体问题

A：

ubuntu物理机系统环境，编译nanopi r5s一直失败，最简化操作是只选择了r5s，其他luci插件默认没选也是编译失败

### (2) 路由器型号和固件版本

A：nanopi r5s    5.15.105（ lean大佬github下最新源码拉取）

### (3) 详细日志

A：
Traceback (most recent call last):
  File "/home/jackylee/nanopi/r5s/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r5s-rk3568/u-boot-2023.04/scripts/dtc/pylibfdt/setup.py", line 23, in <module>
    from setuptools import setup, Extension
ModuleNotFoundError: No module named 'setuptools'
make[6]: *** [scripts/dtc/pylibfdt/Makefile:33: rebuild] Error 1
make[5]: *** [scripts/Makefile.build:397: scripts/dtc/pylibfdt] Error 2
make[4]: *** [Makefile:2003: scripts_dtc] Error 2
make[4]: Leaving directory '/home/jackylee/nanopi/r5s/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r5s-rk3568/u-boot-2023.04'
make[3]: *** [Makefile:325: /home/jackylee/nanopi/r5s/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r5s-rk3568/u-boot-2023.04/.built] Error 2
make[3]: Leaving directory '/home/jackylee/nanopi/r5s/lede/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/nanopi-r5s-rk3568/compile#8.10#1.32#8.83
    ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopi-r5s-rk3568).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/home/jackylee/nanopi/r5s/lede'
make[1]: *** [package/Makefile:110: /home/jackylee/nanopi/r5s/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jackylee/nanopi/r5s/lede'
make: *** [/home/jackylee/nanopi/r5s/lede/include/toplevel.mk:231：world] 错误 2

hyy-666, 2023-04-22:
你的环境有没有按照现在新的readme重新部署？readme近期有更新

JackyLeeGM, 2023-04-23:
> 你的环境有没有按照现在新的readme重新部署？readme近期有更新

额，没做这个更新！大佬，这个在哪部分更新，有链接让我学习下吗？

XiaoBinin, 2023-04-24:
make clean再尝试

JackyLeeGM, 2023-04-25:
> make clean再尝试

试了，没成功，make dirclean也试了，也是失败

XiaoBinin, 2023-04-25:
> > make clean再尝试
> 
> 试了，没成功，make dirclean也试了，也是失败

git pull拉取最新源码看看

JackyLeeGM, 2023-04-25:
> > > make clean再尝试
> > 
> > 
> > 试了，没成功，make dirclean也试了，也是失败
> 
> git pull拉取最新源码看看

感谢大佬帮忙，用chatgpt解决了，再次感谢大佬指点

popeyema, 2023-04-25:
楼主能告知，怎么解决的吗？

> > > > make clean 再试
> > > 
> > > 
> > > 试了，没成功，make dirclean也试了，也是失败
> > 
> > 
> > git pull拉取最新源码看
> 
> 感谢大佬帮忙，用chatgpt解决了，再次感谢大佬指点

楼主能告知，如何解决的吗？

JackyLeeGM, 2023-04-26:
> 楼主能告知，怎么解决的吗？
> 
> > > > > make clean 再试
> > > > 
> > > > 
> > > > 试了，没成功，make dirclean也试了，也是失败
> > > 
> > > 
> > > git pull拉取最新源码看
> > 
> > 
> > 感谢大佬帮忙，用chatgpt解决了，再次感谢大佬指点
> 
> 楼主能告知，如何解决的吗？

sudo apt-get install python3-pip
pip install setuptools

gpt告诉我这个，我试了下，就成了

popeyema, 2023-04-26:
> > 楼主能告知，怎么解决的吗？
> > > > > > make clean 再试
> > > > > 
> > > > > 
> > > > > 试了，没成功，make dirclean也试了，也是失败
> > > > 
> > > > 
> > > > git pull拉取最新源码看
> > > 
> > > 
> > > 感谢大佬帮忙，用chatgpt解决了，再次感谢大佬指点
> > 
> > 
> > 楼主能告知，如何解决的吗？
> 
> sudo apt-get install python3-pip pip install setuptools
> 
> gpt告诉我这个，我试了下，就成了

感谢!

popeyema, 2023-04-26:
> sudo apt-get install python3-setupt

我用的云编译，如何解决这个问题

closed 2023-04-30 by aiamadeus