coolsnowwolf/lede #11148: 最新代码编译R4S失败

liuyanantj, 2023-04-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

用最新代码编译R4S  单线程编译也是失败  有梯子

详细日志如下
 COPY    u-boot.dtb
  MKIMAGE u-boot-dtb.img
  BINMAN  .binman_stamp
Wrote map file './simple-bin.map' to show errors
binman: Node '/binman/simple-bin/fit': subnode 'images/@atf-SEQ': Failed to read ELF file: Python: No module named 'elftools'
make[4]: *** [Makefile:1108: .binman_stamp] Error 1
make[4]: Leaving directory '/home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04'
make[3]: *** [Makefile:325: /home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/.built] Error 2
make[3]: Leaving directory '/home/tc/lede/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/nanopi-r4s-rk3399/compile#50.70#10.25#58.52
    ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopi-r4s-rk3399).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/home/tc/lede'
make[1]: *** [package/Makefile:110: /home/tc/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tc/lede'
make: *** [/home/tc/lede/include/toplevel.mk:231：world] 错误 2

编译环境Ubuntu 20和22都试过了  都不行  请大神帮忙看看

baiywt, 2023-04-23:
把openwrt编译的python3删除掉
rm staging_dir/hostpkg/bin/python3*

liuyanantj, 2023-04-23:
> 把openwrt编译的python3删除掉 rm staging_dir/hostpkg/bin/python3*

Packaged contents of /home/tc/lede/build_dir/target-aarch64_generic_musl/elfutils-0.188/ipkg-aarch64_generic/libelf into /home/tc/lede/bin/packages/aarch64_generic/base/libelf1_0.188-1_aarch64_generic.ipk
echo "libelf" >> /home/tc/lede/staging_dir/target-aarch64_generic_musl/pkginfo/elfutils.default.install
rm -rf /home/tc/lede/tmp/stage-elfutils
mkdir -p /home/tc/lede/tmp/stage-elfutils/host /home/tc/lede/staging_dir/target-aarch64_generic_musl/packages
install -d -m0755 /home/tc/lede/tmp/stage-elfutils/usr/include
cp -fpR /home/tc/lede/build_dir/target-aarch64_generic_musl/elfutils-0.188/ipkg-install/usr/include/* /home/tc/lede/tmp/stage-elfutils/usr/include/
install -d -m0755 /home/tc/lede/tmp/stage-elfutils/usr/lib
cp -fpR /home/tc/lede/build_dir/target-aarch64_generic_musl/elfutils-0.188/ipkg-install/usr/lib/libasm*.{a,so*} /home/tc/lede/tmp/stage-elfutils/usr/lib/
cp -fpR /home/tc/lede/build_dir/target-aarch64_generic_musl/elfutils-0.188/ipkg-install/usr/lib/libdw*.{a,so*} /home/tc/lede/tmp/stage-elfutils/usr/lib/
cp -fpR /home/tc/lede/build_dir/target-aarch64_generic_musl/elfutils-0.188/ipkg-install/usr/lib/libelf*.{a,so*} /home/tc/lede/tmp/stage-elfutils/usr/lib/
install -d -m0755 /home/tc/lede/tmp/stage-elfutils/usr/lib/pkgconfig
cp -fpR /home/tc/lede/build_dir/target-aarch64_generic_musl/elfutils-0.188/ipkg-install/usr/lib/pkgconfig/libelf.pc /home/tc/lede/tmp/stage-elfutils/usr/lib/pkgconfig/
cp -fpR /home/tc/lede/build_dir/target-aarch64_generic_musl/elfutils-0.188/ipkg-install/usr/lib/pkgconfig/libdw.pc /home/tc/lede/tmp/stage-elfutils/usr/lib/pkgconfig/
find /home/tc/lede/tmp/stage-elfutils -name '*.la' | xargs -r rm -f; 
if [ -f /home/tc/lede/staging_dir/target-aarch64_generic_musl/packages/elfutils.list ]; then /home/tc/lede/scripts/clean-package.sh "/home/tc/lede/staging_dir/target-aarch64_generic_musl/packages/elfutils.list" "/home/tc/lede/staging_dir/target-aarch64_generic_musl"; fi
if [ -d /home/tc/lede/tmp/stage-elfutils ]; then (cd /home/tc/lede/tmp/stage-elfutils; find ./ > /home/tc/lede/tmp/stage-elfutils.files); 	SHELL= flock /home/tc/lede/tmp/.staging-dir.flock -c ' mv /home/tc/lede/tmp/stage-elfutils.files /home/tc/lede/staging_dir/target-aarch64_generic_musl/packages/elfutils.list && cp -fpR /home/tc/lede/tmp/stage-elfutils/* /home/tc/lede/staging_dir/target-aarch64_generic_musl/; '; fi
rm -rf /home/tc/lede/tmp/stage-elfutils
touch /home/tc/lede/staging_dir/target-aarch64_generic_musl/stamp/.elfutils_installed
make[3]: Leaving directory '/home/tc/lede/package/libs/elfutils'
time: package/libs/elfutils/compile#49.75#13.14#67.14
make[2]: Leaving directory '/home/tc/lede'
make[1]: *** [package/Makefile:110: /home/tc/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tc/lede'
make: *** [/home/tc/lede/include/toplevel.mk:231：world] 错误 2
多线程编译 还是报错

liuyanantj, 2023-04-23:
> 把openwrt编译的python3删除掉 rm staging_dir/hostpkg/bin/python3*
@baiywt 
touch /home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/.built_check
make  -C /home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04 CROSS_COMPILE=aarch64-openwrt-linux-musl- DTC="/home/tc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.105/scripts/dtc/dtc" HOSTCC="gcc" HOSTCFLAGS="-O2 -I/home/tc/lede/staging_dir/host/include -I/home/tc/lede/staging_dir/hostpkg/include -I/home/tc/lede/staging_dir/target-aarch64_generic_musl/host/include -I/home/tc/lede/staging_dir/host/include -I/home/tc/lede/staging_dir/hostpkg/include -I/home/tc/lede/staging_dir/target-aarch64_generic_musl/host/include -std=gnu11" HOSTLDFLAGS="-L/home/tc/lede/staging_dir/host/lib -L/home/tc/lede/staging_dir/hostpkg/lib -L/home/tc/lede/staging_dir/target-aarch64_generic_musl/host/lib" LOCALVERSION="-OpenWrt-r5911-702087ab3" STAGING_PREFIX="/home/tc/lede/staging_dir/host" PKG_CONFIG_PATH="/home/tc/lede/staging_dir/host/lib/pkgconfig" PKG_CONFIG_LIBDIR="/home/tc/lede/staging_dir/host/lib/pkgconfig" PKG_CONFIG_EXTRAARGS="--static" V='' PATH=/home/tc/lede/staging_dir/host/bin:/home/tc/lede/staging_dir/target-aarch64_generic_musl/host/bin:/home/tc/lede/staging_dir/hostpkg/bin:/home/tc/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/bin:/home/tc/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/bin:/home/tc/lede/staging_dir/host/bin:/home/tc/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/bin:/home/tc/lede/staging_dir/host/bin:/home/tc/lede/staging_dir/host/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin BL31=/home/tc/lede/staging_dir/target-aarch64_generic_musl/image/rk3399_bl31_v1.35.elf
make[4]: Entering directory '/home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04'
  ENVC    include/generated/env.txt
  ENVP    include/generated/env.in
  ENVT    include/generated/environment.h
make -f ./scripts/Makefile.build obj=scripts/dtc
  CC      env/common.o
  AR      env/built-in.o
  CC      lib/efi_loader/dtbdump.o
  LD      lib/efi_loader/dtbdump_efi.so
  OBJCOPY lib/efi_loader/dtbdump.efi
  CC      lib/efi_loader/initrddump.o
  LD      lib/efi_loader/initrddump_efi.so
  OBJCOPY lib/efi_loader/initrddump.efi
  LD      u-boot
  OBJCOPY u-boot.srec
  OBJCOPY u-boot-nodtb.bin
  RELOC   u-boot-nodtb.bin
  CAT     u-boot-dtb.bin
  COPY    u-boot.bin
  SYM     u-boot.sym
  MKIMAGE u-boot.img
  MKIMAGE u-boot-dtb.img
  BINMAN  .binman_stamp
Wrote map file './simple-bin.map' to show errors
binman: Node '/binman/simple-bin/fit': subnode 'images/@atf-SEQ': Failed to read ELF file: Python: No module named 'elftools'
make[4]: *** [Makefile:1108: .binman_stamp] Error 1
make[4]: Leaving directory '/home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04'
make[3]: *** [Makefile:325: /home/tc/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/.built] Error 2
make[3]: Leaving directory '/home/tc/lede/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/nanopi-r4s-rk3399/compile#1.52#0.95#2.61
    ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopi-r4s-rk3399).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/home/tc/lede'
make[1]: *** [package/Makefile:110: /home/tc/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tc/lede'
make: *** [/home/tc/lede/include/toplevel.mk:231：world] 错误 2
单线程日志

youzipp, 2023-04-23:
我也出现了这样做的问题

youzipp, 2023-04-23:
> 我也出现了这样做的问题

[image]

youzipp, 2023-04-23:
[image]

baiywt, 2023-04-23:
除了删除python3
rm staging_dir/hostpkg/bin/python3*
还要删除uboot，这样应该就可以了
rm -rf build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04

liuyanantj, 2023-04-23:
> 除了删除python3 rm staging_dir/hostpkg/bin/python3* 还要删除uboot，这样应该就可以了 rm -rf build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04

还是编译不过 @baiywt

baiywt, 2023-04-23:
现在报什么错？

baiywt, 2023-04-23:
sudo apt install python3-pyelftools
这个装了吗？

youzipp, 2023-04-23:
> 除了删除python3> rm staging_dir/hostpkg/bin/python3*> 还要删除uboot，这样应该就可以了> rm -rf build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04zh> 除了删除python3> rm staging_dir/hostpkg/bin/python3*> 还要删除uboot，这样应该就可以了> rm -rf build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04dalao
大佬晚上好 我执行了您命令其中
rm staging_dir/hostpkg/bin/python3*
rm: cannot remove 'staging_dir/hostpkg/bin/python3*': No such file or directory

rm -rf build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04
删除了

sudo apt install python3-pyelftools
已经安装了
但是编译还是会出现错误！以下是j-1输出内容
make[3]: Leaving directory '/home/youzip/lede/package/utils/jsonfilter'
time: package/utils/jsonfilter/compile#0.11#0.02#0.12
make[3]: Entering directory '/home/youzip/lede/package/system/openwrt-keyring'
mkdir -p /home/youzip/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
SHELL= flock /home/youzip/lede/tmp/.root-copy.flock -c 'cp -fpR /home/youzip/lede/build_dir/target-aarch64_generic_musl/openwrt-keyring-2022-03-25-62471e69/.pkgdir/openwrt-keyring/. /home/youzip/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/youzip/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.openwrt-keyring_installed
echo "openwrt-keyring" >> /home/youzip/lede/staging_dir/target-aarch64_generic_musl/pkginfo/openwrt-keyring.default.install
make[3]: Leaving directory '/home/youzip/lede/package/system/openwrt-keyring'
time: package/system/openwrt-keyring/compile#0.10#0.02#0.11
make[3]: Entering directory '/home/youzip/lede/package/system/usign'
mkdir -p /home/youzip/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
SHELL= flock /home/youzip/lede/tmp/.root-copy.flock -c 'cp -fpR /home/youzip/lede/build_dir/target-aarch64_generic_musl/usign-2020-05-23-f1f65026/.pkgdir/usign/. /home/youzip/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/youzip/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.usign_installed
echo "usign" >> /home/youzip/lede/staging_dir/target-aarch64_generic_musl/pkginfo/usign.default.install
make[3]: Leaving directory '/home/youzip/lede/package/system/usign'
time: package/system/usign/compile#0.14#0.03#0.15
make[3]: Entering directory '/home/youzip/lede/package/base-files'
mkdir -p /home/youzip/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
SHELL= flock /home/youzip/lede/tmp/.root-copy.flock -c 'cp -fpR /home/youzip/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/base-files/.pkgdir/base-files/. /home/youzip/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/youzip/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.base-files_installed
echo "base-files" >> /home/youzip/lede/staging_dir/target-aarch64_generic_musl/pkginfo/base-files.default.install
make[3]: Leaving directory '/home/youzip/lede/package/base-files'
time: package/base-files/compile#0.13#0.05#0.17
make[3]: Entering directory '/home/youzip/lede/package/boot/arm-trusted-firmware-rockchip-vendor'
Makefile:79: WARNING: skipping arm-trusted-firmware-rk3399 -- package has no install section
make[3]: Leaving directory '/home/youzip/lede/package/boot/arm-trusted-firmware-rockchip-vendor'
time: package/boot/arm-trusted-firmware-rockchip-vendor/rk3399/compile#0.14#0.02#0.15
make[3]: Entering directory '/home/youzip/lede/package/boot/uboot-envtools'
mkdir -p /home/youzip/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
SHELL= flock /home/youzip/lede/tmp/.root-copy.flock -c 'cp -fpR /home/youzip/lede/build_dir/target-aarch64_generic_musl/u-boot-2023.01/.pkgdir/uboot-envtools/. /home/youzip/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/youzip/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.uboot-envtools_installed
echo "uboot-envtools" >> /home/youzip/lede/staging_dir/target-aarch64_generic_musl/pkginfo/uboot-envtools.default.install
make[3]: Leaving directory '/home/youzip/lede/package/boot/uboot-envtools'
time: package/boot/uboot-envtools/compile#0.19#0.10#0.23
make[3]: Entering directory '/home/youzip/lede/package/boot/uboot-rockchip'
rm -rf /home/youzip/lede/tmp/stage-uboot-rockchip
mkdir -p /home/youzip/lede/tmp/stage-uboot-rockchip/host /home/youzip/lede/staging_dir/target-aarch64_generic_musl/packages
install -d -m0755 /home/youzip/lede/staging_dir/target-aarch64_generic_musl/image
/home/youzip/lede/staging_dir/target-aarch64_generic_musl/image/loaderimage --pack --uboot /home/youzip/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/u-boot-dtb.bin /home/youzip/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/uboot.img 0x200000
bash: line 1: /home/youzip/lede/staging_dir/target-aarch64_generic_musl/image/loaderimage: cannot execute binary file: Exec format error
make[3]: *** [Makefile:326: /home/youzip/lede/staging_dir/target-aarch64_generic_musl/stamp/.uboot-rockchip.nanopi-r4s-rk3399_installed] Error 126
make[3]: Leaving directory '/home/youzip/lede/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/nanopi-r4s-rk3399/compile#0.22#0.11#0.26
    ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopi-r4s-rk3399).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/home/youzip/lede'
make[1]: *** [package/Makefile:110: /home/youzip/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/youzip/lede'
make: *** [/home/youzip/lede/include/toplevel.mk:231: world] Error 2

Travel2Here, 2023-04-23:
同R4S昨天编的没问题，绝招：make dirclean 解千愁

baiywt, 2023-04-23:
@youzipp 我不是大佬。。
你这个报错不一样
[/home/youzip/lede/staging_dir/target-aarch64_generic_musl/image/loaderimage](bash: line 1: /home/youzip/lede/staging_dir/target-aarch64_generic_musl/image/loaderimage: cannot execute binary file: Exec format error)

你执行下 file /home/youzip/lede/staging_dir/target-aarch64_generic_musl/image/loaderimage 看下

baiywt, 2023-04-23:
@Travel2Here make distclean后应该是需要单线程才能编译过的。至于这个报错的原因是因为在编译uboot的时候，python环境切换到了openwrt自身编译的python3（这时就缺 python3-pyelftools，要调用系统的python3才有 python3-pyelftools），所以可以删除openwrt的python3应该就能解决，单线程编译能过应该是编译uboot的时候，刚好调用的还是系统的python3

Travel2Here, 2023-04-23:
> @Travel2Here make distclean后应该是需要单线程才能编译过的。至于这个报错的原因是因为在编译uboot的时候，python环境切换到了openwrt自身编译的python3（这时就缺 python3-pyelftools，要调用系统的python3才有 python3-pyelftools），所以可以删除openwrt的python3应该就能解决，单线程编译能过应该是编译uboot的时候，刚好调用的还是系统的python3

解决了就好 我是没仔细排查 所以干脆清干净 也就是多花一次更长的时间 😂

liuyanantj, 2023-04-24:
回家再试试看吧  多线程好像还是不行  意思是要make dirclean了？

youzipp, 2023-04-24:
> @youzipp 我不是大佬。。 你这个报错不一样 [/home/youzip/lede/staging_dir/target-aarch64_generic_musl/image/loaderimage](bash: line 1: /home/youzip/lede/staging_dir/target-aarch64_generic_musl/image/loaderimage: [...]

执行完显示
file /home/youzip/lede/staging_dir/target-aarch64_generic_musl/image/loaderimage
/home/youzip/lede/staging_dir/target-aarch64_generic_musl/image/loaderimage: ELF 64-bit LSB pie executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 3.2.0, BuildID[sha1]=ca9b8738f2e60990038cadfc2e1b090cc93876e9, not stripped

XiaoBinin, 2023-04-24:
先更新环境，再make clean尝试

liuyanantj, 2023-04-24:
前面所有的命令跑了一遍 然后更新了最新代码  最后make dirclean 然后单线程跑的 编译成功了