coolsnowwolf/lede #11150: r8125不定时出现断线

suhdd, 2023-04-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
 有一个很早的案例，
#6447 
其中的解决方案已尝试，未解决

## 2. 详细叙述

### (1) 具体问题

A：r8125出现不定期频率不固定的断连，已确定以下情况和做以下操作
1、已关闭SFE
2、机器BIOS已关闭所有PCIE设备的ASPM
3、固件驱动版本最新，详情如下
root@OpenWrt:~# ethtool -i eth0
driver: r8125
version: 9.010.01-NAPI-RSS
firmware-version: 
expansion-rom-version: 
bus-info: 0000:00:10.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: no
4、问题出现时段没有大流量大连接数活动
5、看了下爱快的日志，没有出现断连，猜测与软路由机器可能无关

### (2) 路由器型号和固件版本

A：
J4125+8125B机器
非自编译固件。PVE虚拟openwrt做旁路由，爱快主路由
固件来源和版本：EzOpWrt Ipv6-Mini-Sirpdboy-R2023.04.18 by Sirpdboy
内核版本：5.15.104

### (3) 详细日志

系统日志如下：

A：Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.269815] ------------[ cut here ]------------
Sun Apr 23 00:08:38 2023 kern.info kernel: [93382.270624] NETDEV WATCHDOG: eth0 (r8125): transmit queue 0 timed out
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.271745] WARNING: CPU: 0 PID: 0 at dev_watchdog+0x29b/0x2a0
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.272622] Modules linked in: pppoe ppp_async option l2tp_ppp cdc_mbim usb_wwan qmi_wwan pptp pppox ppp_mppe ppp_generic lzo ipt_REJECT huawei_cdc_ncm cdc_ncm cdc_ether ax88179_178a aqc111 amdgpu xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_bpf xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY via_velocity usbserial usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_fsm ts_bm tcp_bbr slhc rtl8150 r8168 r8152 r8125 pcnet32 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.272730]  nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount mlx5_core mlx4_en mlx4_core macvlan lzo_decompress lzo_compress iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables igc iavf i915 i40e gpu_sched forcedeth e1000e crc_ccitt cdc_wdm cdc_acm bnx2x bnx2 asn1_decoder alx 8139too 8139cp fuse sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit act_ipt act_csum libcrc32c sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact evdev i2c_dev cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.285671]  ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsv3 nfs bonding ip_gre gre ixgbe igbvf e1000 amd_xgbe mdio_devres ifb sit mdio l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel oid_registry tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag tun lockd sunrpc grace dns_resolver nls_utf8 ena crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_ssse3 sha1_generic authenc arc4 crypto_acompress nls_iso8859_1 nls_cp437 vfat fat uas sdhci_pltfm dwc3 dwc2 roles igb exfat button_hotplug tg3 mii libphy
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.309115] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.15.104 #0
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.310083] Hardware name: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.16.1-0-g3208b098f51a-prebuilt.qemu.org 04/01/2014
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.311826] RIP: 0010:dev_watchdog+0x29b/0x2a0
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.312538] Code: ff ff 48 8b 5d d0 c6 05 a5 18 b9 00 01 48 89 df e8 ea 2e fb ff 44 89 e9 48 89 de 48 c7 c7 08 79 06 82 48 89 c2 e8 84 00 14 00 <0f> 0b eb af 90 55 48 89 e5 41 57 49 89 d7 41 56 4d 89 c6 41 55 49
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.315382] RSP: 0018:ffffc90000003e88 EFLAGS: 00010292
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.316175] RAX: 0000000000000039 RBX: ffff888036ee0000 RCX: 0000000000000027
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.317270] RDX: ffff88807dc1c588 RSI: 0000000000000001 RDI: ffff88807dc1c580
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.318398] RBP: ffffc90000003ec0 R08: 0000000000000000 R09: ffffc90000003c90
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.319500] R10: ffffc90000003c88 R11: ffffffff822adfc8 R12: ffff888036ee0480
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.320580] R13: 0000000000000000 R14: ffff8880031e8c40 R15: 0000000000000000
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.321657] FS:  0000000000000000(0000) GS:ffff88807dc00000(0000) knlGS:0000000000000000
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.322905] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.323804] CR2: 00007f9b7f3f8b7c CR3: 0000000038026000 CR4: 0000000000350ef0
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.324904] Call Trace:
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.325300]  <IRQ>
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.325637]  ? pfifo_fast_enqueue+0x150/0x150
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.326327]  call_timer_fn.constprop.0+0x13/0x80
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.327055]  run_timer_softirq+0x289/0x2e0
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.327703]  ? lapic_next_deadline+0x27/0x30
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.328372]  ? clockevents_program_event+0x8d/0xf0
Sun Apr 23 00:08:38 2023 kern.warn kernel: [93382.329115]  ? tick_program_event+0x3e/0x70
[... 424 lines cut ...]
Sun Apr 23 07:12:29 2023 daemon.notice netifd: Network device 'eth0' link is up
Sun Apr 23 07:12:29 2023 daemon.notice netifd: bridge 'br-lan' link is up
Sun Apr 23 07:12:29 2023 daemon.notice netifd: Interface 'lan' has link connectivity
Sun Apr 23 07:13:19 2023 kern.info kernel: [118861.857310] r8125: eth0: link down
Sun Apr 23 07:13:19 2023 daemon.notice netifd: Network device 'eth0' link is down
Sun Apr 23 07:13:19 2023 kern.info kernel: [118861.863710] br-lan: port 1(eth0) entered disabled state
Sun Apr 23 07:13:20 2023 daemon.notice netifd: bridge 'br-lan' link is down
Sun Apr 23 07:13:20 2023 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sun Apr 23 07:13:22 2023 kern.info kernel: [118864.939984] r8125: eth0: link up
Sun Apr 23 07:13:22 2023 kern.info kernel: [118864.940570] br-lan: port 1(eth0) entered blocking state
Sun Apr 23 07:13:22 2023 kern.info kernel: [118864.941431] br-lan: port 1(eth0) entered forwarding state
Sun Apr 23 07:13:22 2023 daemon.notice netifd: Network device 'eth0' link is up
Sun Apr 23 07:13:22 2023 daemon.notice netifd: bridge 'br-lan' link is up
Sun Apr 23 07:13:22 2023 daemon.notice netifd: Interface 'lan' has link connectivity
Sun Apr 23 07:39:34 2023 daemon.notice netifd: Network device 'eth0' link is down
Sun Apr 23 07:39:34 2023 kern.info kernel: [120435.981552] br-lan: port 1(eth0) entered disabled state
Sun Apr 23 07:39:34 2023 kern.info kernel: [120435.982673] r8125: eth0: link down
Sun Apr 23 07:39:35 2023 daemon.notice netifd: bridge 'br-lan' link is down
Sun Apr 23 07:39:35 2023 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sun Apr 23 07:39:37 2023 kern.info kernel: [120439.897422] r8125: eth0: link up
Sun Apr 23 07:39:37 2023 kern.info kernel: [120439.898021] br-lan: port 1(eth0) entered blocking state
Sun Apr 23 07:39:37 2023 kern.info kernel: [120439.898854] br-lan: port 1(eth0) entered forwarding state
Sun Apr 23 07:39:37 2023 daemon.notice netifd: Network device 'eth0' link is up
Sun Apr 23 07:39:37 2023 daemon.notice netifd: bridge 'br-lan' link is up
Sun Apr 23 07:39:37 2023 daemon.notice netifd: Interface 'lan' has link connectivity

rrkai, 2023-04-24:
检查下网线.之前我也遇到..网线老化了.

suhdd, 2023-04-25:
> 检查下网线.之前我也遇到..网线老化了.

好的我换个条线试下

coolsnowwolf, 2026-09-05:
该问题使用第三方 EzOpWrt 2023.04.18、PVE 旁路由和旧版 r8125，日志是 2023 年 Linux 5.15 的 TX queue watchdog，未提供 LEDE 当前源码、配置、硬件 PCI 信息或最小复现条件。旁路由/虚拟化链路和网线也可能导致该现象，当前没有可验证的 LEDE 代码修复，关闭。

closed 2026-09-05 by coolsnowwolf