coolsnowwolf/lede #11151: 编译友善nanopi r4s出现报错 求指点！！

youzipp, 2023-04-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
编译环境：M2 Mac mini 使用ARM ubuntu 22.04.2  虚拟机环境编译

### (1) 具体问题

A：同样的环境另外一台虚拟机 编译X86的固件是完全没有问题的。编译r4s时出错求教！！
最近一直纠结这个问题！已经一个礼拜了 都没有解决！
### (2) 路由器型号和固件版本

A：

### (3) 详细日志
make[4]: Leaving directory '/home/youzip/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04'
touch /home/youzip/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/.built
rm -rf /home/youzip/lede/tmp/stage-uboot-rockchip
mkdir -p /home/youzip/lede/tmp/stage-uboot-rockchip/host /home/youzip/lede/staging_dir/target-aarch64_generic_musl/packages
install -d -m0755 /home/youzip/lede/staging_dir/target-aarch64_generic_musl/image
/home/youzip/lede/staging_dir/target-aarch64_generic_musl/image/loaderimage --pack --uboot /home/youzip/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/u-boot-dtb.bin /home/youzip/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/uboot.img 0x200000
bash: line 1: /home/youzip/lede/staging_dir/target-aarch64_generic_musl/image/loaderimage: cannot execute binary file: Exec format error
make[3]: *** [Makefile:326: /home/youzip/lede/staging_dir/target-aarch64_generic_musl/stamp/.uboot-rockchip.nanopi-r4s-rk3399_installed] Error 126
make[3]: Leaving directory '/home/youzip/lede/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/nanopi-r4s-rk3399/compile#44.40#8.73#50.56
    ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopi-r4s-rk3399).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/home/youzip/lede'
make[1]: *** [package/Makefile:110: /home/youzip/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/youzip/lede'
make: *** [/home/youzip/lede/include/toplevel.mk:231: world] Error 2
A：

youzipp, 2023-04-24:
贴出执行make -j1 V=s
[image]

aiamadeus, 2023-04-24:
rk 的闭源二进制工具只支持 x86

Travel2Here, 2023-04-27:
同R4S没问题，建议楼主pull了最新后，make dirclean再J1。稳妥 肯定行。

raiden2007, 2023-04-27:
我最新源码换了23.04ubuntu才编译成功.22.04跟20.04编译都失败

youzipp, 2023-04-27:
> 同R4S没问题，建议楼主pull了最新后，make dirclean再J1。稳妥 肯定行。

请问您也是同样的环境吗？

Travel2Here, 2023-04-27:
> > 同R4S没问题，建议楼主pull了最新后，make dirclean再J1。稳妥 肯定行。
> 
> 
> 
> 请问您也是同样的环境吗？

我Ubuntu22 前几天也遇到你同样问题 主线也回滚修复了问题，所以你清了dir就行，好几个issue探讨过

closed 2023-05-06 by aiamadeus