coolsnowwolf/lede #11155: 无法单独编译luci-base 请指点

boyu899, 2023-04-25:
[image]

流程都是参考readme 编译lede前想先编译luci-base  错误

WYC-2020, 2023-04-25:
你这参考一点都不靠谱，乱搞，你怕是链工具链都没编译 就直接想搞luci 建议去看点教程

boyu899, 2023-04-25:
> 你这参考一点都不靠谱，乱搞，你怕是链工具链都没编译 就直接想搞luci 建议去看点教程

再请教下大神 纯净lede 按流程走  无任何第三方插件 可以编译成功  

然后单单添加个feed插件库
src-git openclash https://github.com/vernesong/OpenClash

 luci里勾选了openclash这个动作  就编译失败 

年前都是成功的 也用了新的readme新依赖

WYC-2020, 2023-04-26:
> > 你这参考一点都不靠谱，乱搞，你怕是链工具链都没编译 就直接想搞luci 建议去看点教程
> 
> 再请教下大神 纯净lede 按流程走 无任何第三方插件 可以编译成功
> 
> 然后单单添加个feed插件库 src-git openclash https://github.com/vernesong/OpenClash
> 
> luci里勾选了openclash这个动作 就编译失败
> 
> 年前都是成功的 也用了新的readme新依赖

我一直都是成功的，再说 你说失败 有不贴日志，并且选了这些库 最好编译一下内核 这个玩意有的会依赖内核的一些模块

boyu899, 2023-04-26:
> > > 你这参考一点都不靠谱，乱搞，你怕是链工具链都没编译 就直接想搞luci 建议去看点教程
> 
> > 
> 
> > 再请教下大神 纯净lede 按流程走 无任何第三方插件 可以编译成功
> 
> > 
> 
> > 然后单单添加个feed插件库 src-git openclash https://github.com/vernesong/OpenClash
> 
> > 
> 
> > luci里勾选了openclash这个动作 就编译失败
> 
> > 
> 
> > 年前都是成功的 也用了新的readme新依赖
> 
> 
> 
> 我一直都是成功的，再说 你说失败 有不贴日志，并且选了这些库 最好编译一下内核 这个玩意有的会依赖内核的一些模块

好 学习一下先编译内核再编译