coolsnowwolf/lede #11156: Ubuntu20.04,lede的tool版本要求太高了。

tudouershi, 2023-04-25:
[image]

Q：是否搜索了issue (使用 "x" 选择)
* [× ] 没有类似的issue

## 2. 详细叙述
使用make V=s -j1编译出错。自查发现是软件版本不匹配。我的Ubutun20.04不能安装3.25.2的cmake。

   @##能不能更能调整其高版本改低是怎么弄吗？大佬们😀

baodp, 2023-04-25:
Ubuntu 23.04 没有问题

hyy-666, 2023-04-26:
我的22.04.1完全可以啊
[image]

hyy-666, 2023-04-26:
如果理解没错的话，openwrt的工具链是自己有一套，不会调用系统的吧？你是不是魔改了什么？

WYC-2020, 2023-04-26:
可以很大胆的说 这玩意用ubuntu 编译不会错，最多也就是初始化 安装系统依赖的时候 有的高版本的ubuntu 名字变了，现在编译都是用的ubuntu 像这种基本的tools或者工具链都编译失败，基本都是自己问题，要不然早就炸了，还有像那些常见的平台基本的库都出错的 也是，想比如x86 这玩意编译，有些基本的必须的库 自己编译都失败的基本都是自己问题

tudouershi, 2023-04-26:
> 如果理解没错的话，openwrt的工具链是自己有一套，不会调用系统的吧？你是不是魔改了什么？

没有啊，我把原来的卸载了重新执行仍然这样。今天又下了UBUNTU23.04

tudouershi, 2023-04-26:
就是有时候安装慢，我会来回切换代理

WYC-2020, 2023-04-26:
> 就是有时候安装慢，我会来回切换代理

如果肯定 那个速度没问题 那就是 软件有问题，要真全局

hcym, 2023-04-26:
实体安装的20和22都完全肯定没有问题，都不知道怎么乱搞的再一本正经的乱问，

xiehuayi75, 2023-04-26:
九成是代理网络不稳定

WYC-2020, 2023-04-27:
是吗？因为我看云编译 都是用的ubuntu22 不应该出错才对，我这边没验证，因为我一直用的18.04  也没那个时间 再去搞一台 22或者23的ubuntu ,这个后面有时间在测试看看，有可能有什么库升级导致

adamgonglj, 2023-04-27:
> 是吗？因为我看云编译 都是用的ubuntu22 不应该出错才对，我这边没验证，因为我一直用的18.04 也没那个时间 再去搞一台 22或者23的ubuntu ,这个后面有时间在测试看看，有可能有什么库升级导致

感谢WYC-2020，非常感谢。

adamgonglj, 2023-04-27:
> 是吗？因为我看云编译 都是用的ubuntu22 不应该出错才对，我这边没验证，因为我一直用的18.04 也没那个时间 再去搞一台 22或者23的ubuntu ,这个后面有时间在测试看看，有可能有什么库升级导致

再次报错。

[image]

tudouershi, 2023-04-27:
Ubuntu23.04编成了，遇到错误多执行几遍。不能来回切换代理软件，主要还是外网不稳定。

adamgonglj, 2023-04-27:
我是Ubuntu22.04

szdosar, 2023-04-27:
我咋觉得是依赖没更新导致的呢？试试先执行这行更新依赖：
`sudo -E apt-get -qq install $(curl -fsSL raw.githubusercontent.com/szdosar/ish/master/depends-ubuntu-2204)
sudo -E apt-get -qq autoremove --purge
sudo -E apt-get -qq clean
`

adamgonglj, 2023-04-27:
我执行上面的代码，没有任何提示。
我再重新编译试一下

adamgonglj, 2023-04-27:
> 我咋觉得是依赖没更新导致的呢？试试先执行这行更新依赖： `sudo -E apt-get -qq install $(curl -fsSL raw.githubusercontent.com/szdosar/ish/master/depends-ubuntu-2204)` `sudo -E apt-get -qq autoremove --purge` `sudo -E apt-get -qq clean`

执行这段命令后，仍然报错。
[image]

AE86JAY, 2023-04-28:
我云编译用22的也是出错，用20.04正常

VIKINGYFY, 2023-04-29:
一直都是云编译，用的最新22版ubuntu，没有遇到什么问题

https://github.com/VIKINGYFY/OpenWRT-CI

closed 2023-05-03 by aiamadeus

WYC-2020, 2023-05-05:
@adamgonglj   亲测了，没问题，看下是否你的依赖没有安装全，cmake是完全能编译过的