coolsnowwolf/lede #11157: target/linux failed to build

very20101, 2023-04-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：target/linux failed to build 

### (2) 路由器型号和固件版本

A：phicomm N1

### (3) 详细日志

A：make[2] -C target/linux compile 

ERROR: target/linux failed to build. 

make -r target/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on 

make: *** [/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/include/toplevel.mk:231: target/compile] Error 1 

make[2]: Entering directory '/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/scripts/config' 

make[2]: 'conf' is up to date. 

make[2]: Leaving directory '/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/scripts/config' 

make[1]: Entering directory '/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt' 

make[2]: Entering directory '/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/target/linux' 

make[3]: Entering directory '/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/target/linux/armvirt' 

rm -f /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/localversion 

/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/scripts/kconfig.pl + + /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/target/linux/generic/config-5.15 /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/target/linux/armvirt/config-5.15 /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/target/linux/armvirt/64/config-5.15 > /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.target 

awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/.config >> /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.target 

echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.target 

echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.target 

echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.target 

/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/scripts/package-metadata.pl kconfig /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/tmp/.packageinfo /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/.config 5.15 > /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.override 

/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/scripts/kconfig.pl 'm+' '+' /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.target /dev/null /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.override > /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.set 

mv /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.set /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.old 

grep -v INITRAMFS /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.old > /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.set 

echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.set 

echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.set 

rm -rf /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/modules 

cmp -s /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.set /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.prev || { cp /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.set /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config; cp /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.set /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.prev; } 

export MAKEFLAGS= ; [ -d /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/user_headers ] || make -C /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105 KCFLAGS="-fmacro-prefix-map=/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/staging_dir/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Apr 25 15:56:21 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V='' cmd_syscalls= KERNELRELEASE=5.15.105 CC="aarch64-openwrt-linux-musl-gcc" INSTALL_HDR_PATH=/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/user_headers headers_install 

grep '=[ym]' /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.config.set | LC_ALL=C sort | /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/staging_dir/host/bin/mkhash md5 > /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.vermagic 
[... 13 lines cut ...]
make[3]: Leaving directory '/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/target/linux/armvirt' 

make[5]: *** [scripts/kconfig/Makefile:77: olddefconfig] Error 1 

make[4]: *** [Makefile:627: olddefconfig] Error 2 

make[3]: *** [Makefile:19: /home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.105/.modules] Error 2 

make[2]: *** [Makefile:11: compile] Error 2 

make[2]: Leaving directory '/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/target/linux' 

time: target/linux/compile#0.33#0.06#0.39 

ERROR: target/linux failed to build. 

make[1]: *** [target/Makefile:30: target/linux/compile] Error 1 

make[1]: Leaving directory '/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt' 

make: *** [/home/runner/work/openwrt_N1_test/openwrt_N1_test/workspace/openwrt/include/toplevel.mk:231: target/compile] Error 2 

yes: standard output: Broken pipe 

Error: Process completed with exit code 2.

closed 2023-04-26 by very20101

fichenx, 2023-04-27:
这几天我也遇到这个错误，大佬怎么解决的？

edgist, 2023-05-15:
`scripts/Kconfig.include:39: compiler 'aarch64-openwrt-linux-musl-gcc' not found`

应该是编译流程中拉取了不匹配的 Cache 导致的，不同架构需要用 [mixkey](https://github.com/stupidloud/cachewrtbuild#mixkey) 区分